Koenkk/zigbee2mqtt #2488: Hass.IO add-on zigbe2mqtt  stop communicate devises after restart.

DJTerentjev, 2019-12-08:
# Bug Report
## What happened
After new instalation zigbe2mqtt as HassIO plugin and paired two gledopto bulbs. They was working fine. After that I restarted RPI3 and found that HA can't send the commands any more. After troubleshooting I found out that it's enough to restart zigbee2mqtt plug-in to lost communication. Same time I can remove  and re-pair bulbs and it will work, but only until next restart. I also tried another zigbee stick and dev version of zigbee2mqtt with same result. Actually zigbee2mqt can send command to devise and physically it is changing the status, but can't get state back and status in HA stays the same. Could you please advise, how can I fix it.
**Log after re-instalation of  zigbee2mqt (initial pairing of  bulb):**
https://pastebin.com/30TYdywQ
**Log after restart (re-pair one of the bulbs):**
https://pastebin.com/FiMJf4wJ
## What did you expect to happen
I expect it to continue working after restart HA.
## How to reproduce it (minimal and precise)
For me it was just new install of current version of software (zigbee2mqtt plugin) and firmware, pair the devise, restart zigbee2mqtt plug-in, and try to use paired devise.
## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: 20190608.

Koenkk, 2019-12-10:
Could you sniff the traffic when trying to control the bulbs? It seems that zigbee2mqtt successfully sends the command but the acknowledge from the bulb is missing. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

DJTerentjev, 2019-12-10:
Let me try. Strange, before restart it works perfect.

DJTerentjev, 2019-12-10:
Ok. I capture it, but I don't see much difference. Or maybe I don't understand. Please check. 
Firs two before restart zigbee2mqtt. Working fine both light and HA status. 
[image]
[image]

And last one after restart. Light went off, but status in HA didn't change.
[image]

And one more think. If I try to publish MQTT messages manually, I can operate the light. But subsequent messages are going with delay to wireshark, and once it is there light will go off or on.  Same time error in zigbee2mqtt log - timeout. Status in HA stays the same.

Koenkk, 2019-12-11:
Can you provide the wireshark `.pcapng` file of a working and non-working case so I can investigate it further?

DJTerentjev, 2019-12-11:
Sure, how I can send it?

DJTerentjev, 2019-12-11:
Please see file attached. The way I did it. First off all I uninstalled zigbee2mqtt add-on and delete all files. Then I installed it again. I started recording before start zigbee2mqt. Then I paired one bulb and several times switched it on and off from HA interface. All was working fine. Before restart zigbee2mqtt I put light off. At line 131 I restart zigbee2mqtt. While restarting light went on by itself, but slider in HA stays in off position. Then several time I tried to move slider, but didn't work. After all I send MQTT to put light off and light went off (line 213).

[Chanel 13.zip](https://github.com/Koenkk/zigbee2mqtt/files/3952532/Chanel.13.zip)

DJTerentjev, 2019-12-11:
And second file. Just several times I tried to move HA slider in on position. On first attempt light physically went on, but slider stays in off position.  

[Chanel 13_2.zip](https://github.com/Koenkk/zigbee2mqtt/files/3952585/Chanel.13_2.zip)

Koenkk, 2019-12-12:
Could you provide the log with herdsman debug logging enabled when things fail?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

DJTerentjev, 2019-12-12:
Please see file attached zip. It's contain .pcapng, two logs from HA and two log files from zigbee2mqtt folder. I restarted zigbee2mqtt add-on on the line 137.
[Chanel_14.zip](https://github.com/Koenkk/zigbee2mqtt/files/3958037/Chanel_14.zip)

Koenkk, 2019-12-13:
I don't see the herdsman debug logging here. Note that this is only logged to STDOUT and not to the log files.

DJTerentjev, 2019-12-14:
Could you please check my [ configuration](https://pastebin.com/JceegBUy). I used **"zigbee_herdsman_debug": true**, but nothing from zigbee_herdsman in the log. I use Portainer , but nothing in  addon_7ad98f9c_zigbee2mqtt container log also.

Koenkk, 2019-12-14:
Sorry, I think it only works with the latest edge addon (not stable), please remove the stable one and switch to edge.

DJTerentjev, 2019-12-14:
Good. Now I got the correct log. Please see attached. 
[Chanel 15.zip](https://github.com/Koenkk/zigbee2mqtt/files/3963927/Chanel.15.zip)

Koenkk, 2019-12-14:
It seems that after the restart zigbee-herdsman does not receive any events anymore from the CC2538. Therefore it appears to zigbee-herdsman that the acknowledge is not received. Can you still pair other devices after this?

Could you also try if this occurs with a CC2531? Perhaps the problem is in the CC2538 firmware.

DJTerentjev, 2019-12-14:
Yes I can pair other devices after. I just tried xiaomi plug. And I can remove bulb with MQTT message, pair it again and it will work until next restart.  
I just tried with CC2531 (in this case I can't sniff ) .  The way I did it: uninstall add-on and delete  zigbee2mqtt folder, flush CC2531 with coordinator firmware, install add-on. After add-on start I paired xiaomi plug, then gledopto bulb, all was working fine. Then I restarted add-on - xiaomi plug was working, gledopto bulb - no. Please see the log.
[Chanel 15.zip](https://github.com/Koenkk/zigbee2mqtt/files/3964143/Chanel.15.zip)

Koenkk, 2019-12-15:
Probably the issue is in the firmware of the bulb (have seen this more for gledopto devices), added a workaround in the latest zigbee2mqtt dev.

DJTerentjev, 2019-12-15:
I totally agree,  something wrong with those bulbs, but last year I was able to use them after add [converter](https://github.com/Koenkk/zigbee2mqtt/issues/682).
> added a workaround in the latest zigbee2mqtt dev

Thank you. Do you mean I can try it?

Koenkk, 2019-12-15:
yes

DJTerentjev, 2019-12-16:
Sorry, I took me awhile to check. I can confirm it is working now. And restart does not affect any more. As I understood there is no request to get initial state of the bulb after restart, but is fine for me. **Thank you very much!!!** Could you please to keep it this way in next stable version.

Koenkk, 2019-12-16:
@DJTerentjev even better news for you, it's already in last day released zigbee2mqtt 1.8.0 release :smile:

closed 2019-12-16 by Koenkk