Koenkk/zigbee2mqtt #2490: Switches controlling all the lights

LucaKaufmann, 2019-12-09:
# Bug Report
## What happened
I discovered a weird bug that happened with a few different switches, mainly the Osram Smart+ Mini switch but occasionally also the Tradfri remote. Somehow the switch takes control of all the lights, when pressing down on the Osram switch all the lights turn off, when pressing up or the circle all the lights turn on. I checked the Zigbee2mqtt output and zigbee2mqtt itself isn't sending any commands or status updates over MQTT, so it seems like the switch directly talks to the light bulbs. I tried re-pairing multiple times, restarting zigbee2mqtt, calling zigbee2mqtt/bridge/group/remove_all but nothing helped

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: 20190619 source_routing

Log output when pressing the circle button on the osram switch:

```
Dec 09 08:13:01 raspberrypi npm[1354]: zigbee2mqtt:debug 2019-12-09T08:13:01: Received Zigbee message from 'entrance_switch', type 'commandMoveToLevelWithOnOff', cluster 'genLevelCtrl', data '{"level":254,"transtime":10}' from endpoint 3 with groupID 0
Dec 09 08:13:01 raspberrypi npm[1354]: zigbee2mqtt:info  2019-12-09T08:13:01: MQTT publish: topic 'zigbee2mqtt/entrance_switch', payload '{"linkquality":39,"battery":"100.00","voltage":3100,"action":"circle_click"}'
Dec 09 08:13:01 raspberrypi npm[1354]: zigbee2mqtt:info  2019-12-09T08:13:01: MQTT publish: topic 'zigbee2mqtt/entrance_switch', payload '{"linkquality":39,"battery":"100.00","voltage":3100,"action":""}'
Dec 09 08:13:01 raspberrypi npm[1354]: zigbee2mqtt:debug 2019-12-09T08:13:01: Received Zigbee message from 'entrance_switch', type 'commandMoveToLevelWithOnOff', cluster 'genLevelCtrl', data '{"level":254,"transtime":10}' from endpoint 3 with groupID 0
Dec 09 08:13:01 raspberrypi npm[1354]: zigbee2mqtt:info  2019-12-09T08:13:01: MQTT publish: topic 'zigbee2mqtt/entrance_switch', payload '{"linkquality":39,"battery":"100.00","voltage":3100,"action":"circle_click"}'
Dec 09 08:13:01 raspberrypi npm[1354]: zigbee2mqtt:info  2019-12-09T08:13:01: MQTT publish: topic 'zigbee2mqtt/entrance_switch', payload '{"linkquality":39,"battery":"100.00","voltage":3100,"action":""}'
Dec 09 08:13:01 raspberrypi npm[1354]: zigbee2mqtt:debug 2019-12-09T08:13:01: Received Zigbee message from 'entrance_switch', type 'commandMoveToLevelWithOnOff', cluster 'genLevelCtrl', data '{"level":254,"transtime":10}' from endpoint 3 with groupID 0
Dec 09 08:13:01 raspberrypi npm[1354]: zigbee2mqtt:info  2019-12-09T08:13:01: MQTT publish: topic 'zigbee2mqtt/entrance_switch', payload '{"linkquality":39,"battery":"100.00","voltage":3100,"action":"circle_click"}'
Dec 09 08:13:01 raspberrypi npm[1354]: zigbee2mqtt:info  2019-12-09T08:13:01: MQTT publish: topic 'zigbee2mqtt/entrance_switch', payload '{"linkquality":39,"battery":"100.00","voltage":3100,"action":""}'
Dec 09 08:13:01 raspberrypi npm[1354]: zigbee2mqtt:debug 2019-12-09T08:13:01: Received Zigbee message from 'entrance_switch', type 'commandMoveToLevelWithOnOff', cluster 'genLevelCtrl', data '{"level":254,"transtime":10}' from endpoint 3 with groupID 0
Dec 09 08:13:01 raspberrypi npm[1354]: zigbee2mqtt:info  2019-12-09T08:13:01: MQTT publish: topic 'zigbee2mqtt/entrance_switch', payload '{"linkquality":39,"battery":"100.00","voltage":3100,"action":"circle_click"}'
Dec 09 08:13:01 raspberrypi npm[1354]: zigbee2mqtt:info  2019-12-09T08:13:01: MQTT publish: topic 'zigbee2mqtt/entrance_switch', payload '{"linkquality":39,"battery":"100.00","voltage":3100,"action":""}'
Dec 09 08:13:02 raspberrypi npm[1354]: zigbee2mqtt:debug 2019-12-09T08:13:02: Received Zigbee message from 'entrance_switch', type 'commandMoveToColorTemp', cluster 'lightingColorCtrl', data '{"colortemp":370,"transtime":10}' from endpoint 3 with groupID 0
Dec 09 08:13:02 raspberrypi npm[1354]: zigbee2mqtt:debug 2019-12-09T08:13:02: Received Zigbee message from 'entrance_switch', type 'commandMoveToColorTemp', cluster 'lightingColorCtrl', data '{"colortemp":370,"transtime":10}' from endpoint 3 with groupID 0
Dec 09 08:13:02 raspberrypi npm[1354]: zigbee2mqtt:debug 2019-12-09T08:13:02: Received Zigbee message from 'entrance_switch', type 'commandMoveToColorTemp', cluster 'lightingColorCtrl', data '{"colortemp":370,"transtime":10}' from endpoint 3 with groupID 0
Dec 09 08:13:02 raspberrypi npm[1354]: zigbee2mqtt:debug 2019-12-09T08:13:02: Received Zigbee message from 'entrance_switch', type 'commandMoveToColorTemp', cluster 'lightingColorCtrl', data '{"colortemp":370,"transtime":10}' from endpoint 3 with groupID 0
```

I'm able to reproduce the issue pretty consistently now, so if there's any other info that would be helpful or anything I can try let me know.

Koenkk, 2019-12-10:
Could you:

- update to the latest dev
- remove the remote from the network
- pair it again (be sure that you see a sucesfully configured... For this device in the log moments after pairing it.

LucaKaufmann, 2019-12-11:
@Koenkk that seems to have fixed it, what I did was update to dev, force_remove the switch, factory reset the switch and then pairing it again

Koenkk, 2019-12-11:
Great, thanks for reporting back.

closed 2019-12-11 by Koenkk