Koenkk/zigbee2mqtt #2493: No power usage information supported Heiman HS2SK devices.

WaRSTeaM, 2019-12-09:
# Bug Report
## What happened
I've added a device: Heiman Smart Metering Plug (HS2SK) (4 in total)
The device is supported (https://www.zigbee2mqtt.io/devices/HS2SK.html) and mqtt discovery is used as is the manual configuration but to no avail.
For some kind of reason i don't get the power reading of all my Devices. I've installed the latest version of zigbee2mqtt in hass.io, the mosquitto broker and all works fine for all my other "smart things" except the Heiman Smart Metering Plugs.

If i send a get command to the device I get the following reading:
zigbee2mqtt:info  2019-12-09T22:15:27: MQTT publish: topic 'zigbee2mqtt/0x000d6f0011087355', payload '{"state":"ON","linkquality":31,"device":{"friendlyName":"0x000d6f0011087355","ieeeAddr":"0x000d6f0011087355","networkAddress":90,"type":"Router","manufacturerID":4619,"manufacturerName":"Heiman","powerSource":"Mains (single phase)","applicationVersion":17,"zclVersion":2,"hardwareVersion":16,"dateCode":"2018.1.11"}}'

I am missing the complete power payload information.
I know for sure the powermeter functionality of the devices are working correct as the are meters that were used with a smartthings setup where the readings were available.

## What did you expect to happen
i would expect a "power usage" payload information value. 
Or as i get from the devices.js from zigbee-herdsman-converters payload information for
- Voltage
- Current
- activePower

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: CC2531_DEFAULT_20190608 (installed 4 weeks ago, fresh download)

##Extra information
Join output:
zigbee2mqtt:info  2019-12-09T22:37:14: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2019-12-09T22:37:14: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.7.1","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}'
zigbee2mqtt:info  2019-12-09T22:37:19: Device '0x000d6f0011087355' joined
zigbee2mqtt:info  2019-12-09T22:37:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000d6f0011087355"}}'
zigbee2mqtt:info  2019-12-09T22:37:19: Starting interview of '0x000d6f0011087355'
zigbee2mqtt:info  2019-12-09T22:37:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000d6f0011087355"}}'
zigbee2mqtt:info  2019-12-09T22:37:20: Successfully interviewed '0x000d6f0011087355', device has successfully been paired
zigbee2mqtt:info  2019-12-09T22:37:20: Device '0x000d6f0011087355' is supported, identified as: HEIMAN Smart metering plug (HS2SK)
zigbee2mqtt:info  2019-12-09T22:37:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x000d6f0011087355","model":"HS2SK","vendor":"HEIMAN","description":"Smart metering plug","supported":true}}'
zigbee2mqtt:info  2019-12-09T22:37:20: Configuring '0x000d6f0011087355'
zigbee2mqtt:info  2019-12-09T22:37:21: Succesfully configured '0x000d6f0011087355'

Koenkk, 2019-12-10:
Please try https://github.com/Koenkk/zigbee2mqtt/issues/2490#issuecomment-564109227 (1.7.1 has a bug regarding this)

WaRSTeaM, 2019-12-11:
**- update to the latest dev**
  Currently running latest available Dev version (1.7.1)

**- remove the remote from the network**
  Done.
zigbee2mqtt:warn  2019-12-11T18:20:29: Device '0x000d6f0011087355' left the network
zigbee2mqtt:info  2019-12-11T18:20:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x000d6f0011087355"}}'
zigbee2mqtt:warn  2019-12-11T18:20:29: Device '0x000d6f0011087355' left the network
zigbee2mqtt:info  2019-12-11T18:20:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x000d6f0011087355"}}'
zigbee2mqtt:warn  2019-12-11T18:20:30: Device '0x000d6f0011087355' left the network
zigbee2mqtt:info  2019-12-11T18:20:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x000d6f0011087355"}}'

**- pair it again (be sure that you see a successfully configured... For this device in the log moments after pairing it.**
Done
zigbee2mqtt:info  2019-12-11T18:22:03: Device '0x000d6f0011087355' joined
zigbee2mqtt:info  2019-12-11T18:22:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000d6f0011087355"}}'
zigbee2mqtt:info  2019-12-11T18:22:03: Starting interview of '0x000d6f0011087355'
zigbee2mqtt:info  2019-12-11T18:22:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000d6f0011087355"}}'
zigbee2mqtt:info  2019-12-11T18:22:04: Successfully interviewed '0x000d6f0011087355', device has successfully been paired
zigbee2mqtt:info  2019-12-11T18:22:04: Device '0x000d6f0011087355' is supported, identified as: HEIMAN Smart metering plug (HS2SK)
zigbee2mqtt:info  2019-12-11T18:22:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x000d6f0011087355","model":"HS2SK","vendor":"HEIMAN","description":"Smart metering plug","supported":true}}'
zigbee2mqtt:info  2019-12-11T18:22:04: Configuring '0x000d6f0011087355'
zigbee2mqtt:info  2019-12-11T18:22:04: **Succesfully configured** '0x000d6f0011087355'

**- Extra Steps Taken**
Removed the device again. This time also removed the entry in the Devices.yaml file. 
The device is newly added to the file (Last Entry)
Added the device again, played with the connected electronic device sucking power from the socket so to generate metrics, powered it on and off a couple of times and 30 min later checked the state.json file:
    "0x000d6f0011087355": {
        "state": "ON",
        "linkquality": 39

Still missing my powerusage metrics.
If needed i can try to delete the complete device (mqtt autodiscovery added device) or in the worst case grab a spare Pi and install the software from scratch and adding only the power socket as device to verify it's not an issue residing within my specific setup.

WaRSTeaM, 2019-12-12:
Upgrading to Dev works indeed like a charm. Did not even have to remove the other 3 Powerplugs to get the payloads available from them.
I'll witch back now to the Stable version and wait for the next stable release.
Much appreciated!

closed 2019-12-12 by WaRSTeaM

answer35, 2021-03-27:
Hello, I am starting to use Zigbee2MQTT (1.18.1-2) with Home Assistant. I also use that model and I also have that issue with not receiving all datas from the plug, only state on/off, all the rest stays at N/A. The issue is from 2019 so I am curious if I am missing something or not. Thank you.

corneyl, 2022-04-03:
@answer35 is this still a problem for you? Then see #12000