Koenkk/zigbee2mqtt #2495: latency on single click on WXKG02LM

bartplessers, 2019-12-10:
#1787 Bug Report
## What happened
- connected a XIAOMI Aqara double key wireless wall switch to zigbee2mqtt
- single click events are registered, but...
- ... event in mqtt is registered after "a while" (although only fraction of seconds)

## What did you expect to happen
- compared to other devices like [IKEA E1524/E1810
](https://www.zigbee2mqtt.io/devices/E1524_E1810.html), the response is slow. 

I assume this is due to the fact that Z2M has to"recognise" the other possible events (click_double, click_long, etc...) but I'm wondering why Z2M can handle events from a device like IKEA [IKEA E1524/E1810](https://www.zigbee2mqtt.io/devices/E1524_E1810.html) so well.... (it's much faster and responsive). Maybe there is another way to make the XIAOMI device as responsive as the IKEA device?

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

Koenkk, 2019-12-11:
Zigbee2mqtt doesn't recognise the other possible events, the device actually does this.

bartplessers, 2019-12-11:
okay. something to be aware of when buying hardware. thanx for response.

closed 2019-12-11 by bartplessers