Koenkk/zigbee2mqtt #2496: Possible: bind closuresWindowCovering from Up/Down Remote to Fyrtur Blind

jdtsmith, 2019-12-10:
I've successfully paired `z2m` with  Fyrtur blinds (E1757) and the associated "open/close" remote (E1766).  By putting the blinds in a group, and binding:

```
zigbee2mqtt/bridge/bind/ikea_updown_remote_1" -m "ikea_blind_group"
```

I can even control the blinds with the remote (with a small delay).  But sadly, the remote can only fully open or close.  It cannot interrupt any opening or closing in progress.  This is unlike MQTT, and unlike the remote when paired directly to the blinds via the signal repeater.  

It appears the bind occurs via `genLevelCtrl`.  I tried switching that out for `closuresWindowCovering`, and the bound remote still works and is bound:

```
zigbee2mqtt:info  2019-12-10 04:27:26: Successfully bound cluster 'closuresWindowCovering' from 'ikea_updown_remote_1' to 'ikea_blind_group'
zigbee2mqtt:info  2019-12-10 04:27:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_bind","message":{"from":"ikea_updown_remote_1","to":"ikea_blind_group","cluster":"closuresWindowCovering"}}'
```

but remote operations are still non-interruptible.  Any possibility to give the binding the same interruption powers?

Thanks for z2m!

## Debug Info
zigbee2mqtt version: 1.7.1+dev (commit #60627ab)
CC253X firmware version: CC2531_DEFAULT_20190608

Koenkk, 2019-12-11:
This is probably because the `closuresWindowCovering` was not bound, it is now in the latest dev branch, please update and rebind the device.

jdtsmith, 2019-12-11:
OK, I've updated to dev head now, and re-bound remote to blinds group.  I am able now to see [all 3 binds complete](https://pastebin.com/Q3wFWNka) — `genOnOff`, `genLevelCtrl`, and `closuresWindowCovering`.   However, the situation is still that up/down moves are uninterruptible via the remote.  The good news is that remote works even when `z2m` isn't running.  And here's the truly interesting bit: the **remote can interrupt moves, stop, reverse direction, etc. _as long as z2m isn't running_**.  

I.e. if the controller is powered and remote bound to blinds group, but `z2m` isn't running, remote works as expected (with a 1-2sec delay).  With `z2m` running, only up/down (without interruption) works with remote.   

Let me know if I can provide more debug logs.  Feels like we are very close to having full remote operation + full MQTT control!  Thanks for your work.

Koenkk, 2019-12-13:
Can you provide the zigbee2mqtt debug log when controlling the blinds using the controller and it is not interuptable?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

jdtsmith, 2019-12-13:
It's [here](https://pastebin.com/raw/zqaSJ0yS).  Interestingly the `brightness` attribute for the group has leaked over here as well.  I tried unbinding/binding with no change.  Thanks.

kiall, 2019-12-14:
> OK, I've updated to dev head now, and re-bound remote to blinds group. I am able now to see all 3 binds complete — genOnOff, genLevelCtrl, and closuresWindowCovering. However, the situation is still that up/down moves are uninterruptible via the remote.

I've just managed to bind my up/down remotes to my blinds group. I am not seeing this behavior.

I can press a button on the remote, let them start moving, and press a button again on the remote to have them stop.

> Interestingly the brightness attribute for the group has leaked over here as well.

Yep, I see that too - it seems to be harmless though?

My groups YAML looks like this, which is slightly different to yours - though I don't believe it actually matters:

    '45589':
      friendly_name: master_bedroom_blinds_group
      devices:
        - 0xd0cf5efffe6d842f/1
        - 0xd0cf5efffeea8190/1
        - 0x000d6ffffe2b7ca7/1

jdtsmith, 2019-12-14:
Interesting.  Can you share your steps for binding, just in case they differ? BTW, what kind of delay do you get?

kiall, 2019-12-14:
I removed/unpaired the remote, paired the remote again, and just sent a MQTT message to bind the remote (source) to the blinds group (target). After that, it just worked.

I also made sure to have pressed the button just before sending the bind command, so it would definitely be awake.

Re delays - the blinds have a delay it seems, I'd say a half to 1 second? I've not timed it!

Re repeater - yes, I have the repeater paired to Z2M as well (I have 1 repeater, 3 blinds, and 2 remotes and a bulb in the same room).

The remotes / repeater / blinds are not paired to each other (as IKEAs instructions say to do..) instead, they all all paired to my Z2M controller.

jdtsmith, 2019-12-15:
Interesting, quite similar to what I've done. So the repeater is not in the group, yes? I'll give another try when I'm back to the blinds in about a week.

Thanks

kiall, 2019-12-15:
> So the repeater is not in the group

Correct, the repeater is not in the group.

sergiigladchuk, 2019-12-15:
Hi guys, it looks like you are experts in grouping and binding IKEA blinds. I have very basic question - with latest 1.8.0 Z2M - I still cannot add two blinds to one group. Here are my steps:
1. I create group in configuration (groups.yaml):
```
'7':
  friendly_name: blinds_living
  devices:
    - 0x000d6ffffea4d47a/1
    - 0x086bd7fffe55e80d/1
```
2. Then restart Z2M
3. Send MQTT commands to add devices to group `zigbee2mqtt/bridge/group/blinds_living/add` with payload of friendly names. I can see that debug log says:
```
zigbee2mqtt:info  2019-12-15 23:27:07: Adding 'blinds_dark_IKEA_2' to 'blinds_living'
zigbee2mqtt:info  2019-12-15 23:27:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload {"type":"device_group_add","message":{"friendly_name":"blinds_dark_IKEA_2","group":"blinds_living"}}'
```
And then nothing.
If I send any MQTT comands to `zigbee2mqtt/blinds_living/set` nothing happens.
If I send direct commands to each blind they work as desired, and show message with groupID 0 :( . e.g.:
```
Received Zigbee message from 'blinds_dark_IKEA_2', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":26}' from endpoint 1 with groupID 0
```

What am I doing wrong? - why aren't they added to the group? I have tested both variants without entering blinds IDs into group.yaml file, just running MQTT `add` commands - same effect, group id is not changed.

jdtsmith, 2019-12-15:
You don't need both a config-version and MQTT group-making, for me it works just with a config group.  I'm not sure this made it yet into a release since it's still under investigation.  So you need to be on the latest head version of the **dev** branch.

sergiigladchuk, 2019-12-18:
Ok, I think I understood how to add these blinds to one group. I have pressed both buttons to activate pairing mode on them, and then send MQTT command to add to the group. Now at least I receive error:
```
zigbee2mqtt:error 2019-12-18 23:57:53: Failed to call 'Groups' 'onZigbeeStarted' (Error: Timeout - 64202 - 1 - 2 - 4 - 0 after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))
```
Same error I receive when I try to add open/close remove switch to the group.
I tried both master and dev branches (and I made sure I have latest commits of both) - same error. Is that something you encountered or should I open new issue?

I also checked your log (@jdtsmith)  you have [here](https://pastebin.com/raw/zqaSJ0yS) - it  says that all your blinds have `groupID 0` not `1`. Can you control your entire group 1 via MQTT, not bonded control?

jdtsmith, 2019-12-20:
>- it says that all your blinds have `groupID 0` not `1`. Can you control your entire group 1 via MQTT, not bonded control?

I feel like the group number becomes just an ordered list starting at 0.  I can control blinds with both MQTT and `bind`ed remote.   Still trying to get remote to stop a move once initiated in this configuration, but it sounds like others have not had that problem.

sergiigladchuk, 2019-12-20:
@kiall do you also have `groupID 0` for your grouped blinds in your debug messages? 
Since everything works fine for you - can you describe your grouping process as well? Have you activated blinds (pressed two buttons up and down briefly) so blinds led glows for 2 minutes and then only  sent MQTT commands for grouping?
I will go forward and try binding too - I just wanted to have MQTT control over my group first.
The thing is that I can easily add bulb to the group and it has `groupID 7` - and it can be controlled with MQTT command to sent the group. I am pretty sure there is some bug in grouping the blinds.

jdtsmith, 2019-12-21:
> I've just managed to bind my up/down remotes to my blinds group. I am not seeing this behavior.
>
> I can press a button on the remote, let them start moving, and press a button again on the remote to [...]

I'm still unable to achieve this.  Blinds will start but not stop or reverse direction with remote.  Do you include the repeater in the group as well?

kiall, 2019-12-25:
@jdtsmith sorry, away from the house - can't check.

Also, I'm not sure I'm using the group via MQTT commands, I can't remember if I updated Home Assistant to use the ZigBee group or not, I might have left it with a HA "group" after getting the binding working ..

jdtsmith, 2019-12-26:
> @jdtsmith sorry, away from the house - can't check.

If you don't mind checking when you return.  A good sign would be whether they all start moving at one, or HomeAssistant controls them one by one.  Grouping them within HA, Homebridge, etc. is obviously a backup, that might start to suffer for very large groups. Thanks!

jdtsmith, 2020-01-05:
@kiall did you have a chance to see if your group command is really working (with start/stop/reverse)?

kiall, 2020-01-05:
Nope, still away for another while!

jdtsmith, 2020-03-03:
I managed to update the firmware of both my blinds (using #2921), reset them (7s press both buttons) and allowed repairing to z2m.  Then I regrouped them, and again bound my up/down remote to the group.  

Now the remote works as intended: I can interrupt and reverse movements, and press-and-hold then release to stop works as well. It's hard to synchronize movement, but it works well enough.  I still can't control the group from `zigbee2mqtt/blinds_group/set`, but this is progress...

closed 2020-03-03 by jdtsmith

jdtsmith, 2020-04-19:
Sadly my victory was short-lived... group & remote functionality is now again not working, and in fact attempts to bind the remote lead to the blinds dropping off the network within a day.  Not sure where to go from here...

King-Cole, 2022-02-10:
For anyone having trouble with binding their remote to the blind group directly, I have some advice. It took me a while, but finally got it. I have 3 blinds (thus 3 remotes), and I did a bad thing on the first remote that screwed me up for a while: upgrading via OTA. The second two remotes I did not do this, and they are still able to command groups just fine. Up, down, stopping partway, the works.

So, in short:

1. For firmwares: blinds to newest (2.3.088), button to old (shipped with 2.2.010)
2. Create group via z2m UI and add all your blinds. FYI, I have never gotten scenes or group commands to work here. I've tried setting the group to "optimistic" and every kinda MQTT payload imaginable, just doesn't work.
3. Via the z2m UI, go to the "Bind" page under the button device.
4. Unbind the default group (default_bind_group), press either up or down on the button to ensure it receives the setting.
5. Bind your newly created blinds group with only the "Closures" cluster. Again, press either up or down on the button.
6. With any luck (trust me you need some in this game), you can now control all your blinds!