Koenkk/zigbee2mqtt #2499: zigbee model : 'undefined' in version 1.7.1 but was working in version 1.6

redmuchroom, 2019-12-10:
I successfuly installed the CC2531on the rpi and zigbee2mqtt (1.6 at that time). 
everything was working quite fine with a climax prm5-zbs (didn't pull the config yet as I just installed and didn't work with it...)
few months later (i.e. now :) ) I installed zigbee2mqtt (1.7.1) on a new rpi with a new CC2531 (revision:20190608) .
there I cannot use the climax prm5 because it says (see log of pairing process below) that the zigbee model is undefined...
https://pastebin.com/eU2y7d7k
indeed in the database,  I don't see any information from the switch 

database line with 1.7.1:
`{"id":2,"type":"EndDevice","ieeeAddr":"0x00124b001bfbaf65","nwkAddr":7251,"manufId":0,"epList":[1,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,1794],"outClusterList":[0],"clusters":{},"binds":[]},"4":{"profId":260,"epId":4,"devId":81,"inClusterList":[],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{}}},"binds":[]}},"interviewCompleted":true,"meta":{}}`

VS database line with 1.6:
`{"id":2,"type":"EndDevice","ieeeAddr":"0x00124b001bfba4d8","nwkAddr":2443,"manufId":0,"manufName":"ClimaxTechnology","powerSource":"Mains (single phase)","modelId":"PSMP5_00.00.03.19TC","epList":[1,4],"zclVersion":1,"appVersion":3,"hwVersion":0,"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,1794],"outClusterList":[0],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{"zclVersion":1,"appVersion":3,"hwVersion":0,"manufacturerName":"ClimaxTechnology","modelId":"PSMP5_00.00.03.19TC","dateCode":"20180201","powerSource":1,"locationDesc":" ","physicalEnv":0,"deviceEnabled":1}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":1}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":128}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}},"seMetering":{"dir":{"value":1},"attrs":{"currentSummDelivered":{"0":0,"1":0},"currentSummReceived":{"0":0,"1":0},"status":0,"unitOfMeasure":0,"multiplier":1,"divisor":10000,"summaFormatting":249,"demandFormatting":148,"meteringDeviceType":0,"instantaneousDemand":0}}}},"4":{"profId":260,"epId":4,"devId":81,"inClusterList":[],"outClusterList":[25],"clusters":{"genOta":{"dir":{"value":2},"attrs":{"70":4294967295,"upgradeServerId":"0xffffffffffffffff","fileOffset":4294967295,"currentFileVersion":319,"currentZigbeeStackVersion":2,"downloadedFileVersion":4294967295,"downloadedZigbeeStackVersion":65535,"imageUpgradeStatus":0,"manufacturerId":0,"imageTypeId":0,"minimumBlockReqDelay":0,"imageStamp":0}}}}},"dateCode":"20180201","_id":"ZEwm9UXZnGYC94Lh"}`

I tried to pair several time but I get the same answer all the time...
Any idea on what could be the reason?

thanks!

Koenkk, 2019-12-11:
Can you try replacing `node_modules/zigbee-herdsman/dist/controller/model/devices.js` with the following https://gist.github.com/Koenkk/788ca3fc0d57a077582ee9fd0121d053, pair the device and provide the log again?

redmuchroom, 2019-12-12:
Hi @Koenkk ,

thanks for the reply.
here is the log : https://pastebin.com/VSXia5ks

hope this helps,
thanks for the support

Koenkk, 2019-12-12:
Thanks, this doesnt reveal any interesting which means we have to go deeper.

Could you sniff the traffic and provide the debug logs when pairing with both zigbee2mqtt 1.6 and 1.7 (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

redmuchroom, 2019-12-12:
it is what I was thinking :)
I don't have a cc2531 right in hand but should get one by next week...
I will let you know ...
thanks!

redmuchroom, 2019-12-19:
Hi,
finally got the sniffing log... 
dor some reasons (well rpi dead...), I had to reinstall all...
so the logs below are for version 1.8.0 (same issue however)

I hope this will help,
thanks,

[log_wireshark_climax.zip](https://github.com/Koenkk/zigbee2mqtt/files/3982023/log_wireshark_climax.zip)

Koenkk, 2019-12-19:
Thanks, should be fixed in the latest dev, can you confirm?

buzzdev, 2020-01-03:
I can confirm the same error:
`Device '0x00124b00096cd8f8' with Zigbee model 'undefined' is NOT supported...`
Now i've updated herdsman to version 0.12.26 and it is working fine again:
`Device '0x00124b000*****f8' with Zigbee model 'PSSP5_00.00.03.06TC' is NOT supported`
`Device '0x00124b000/////9d' with Zigbee model 'PSMP5_00.00.03.05TC' is NOT supported`

Koenkk, 2020-01-05:
Thanks for confirming!

closed 2020-01-05 by Koenkk