Koenkk/zigbee2mqtt #2500: 'MAC transaction expired' (240)'

prankousky, 2019-12-10:
# Bug Report
## What happened

When I try to **set** temperature on `SPZB0001` thermostat, this will show `zigbee2mqtt:error 2019-12-10T13:59:39: Publish 'set' 'current_heating_setpoint' to 'Jonna_Heizung' failed: 'Error: Data request failed with error: 'MAC transaction expired' (240)'`; however, when I *manually* set temperature on the thermostat, it will update and show the actual value (only when I try to change it via zigbee2mqtt (through Home Assistant) it will **not** work).

## What did you expect to happen

Set temperature in Home Assistant, have thermostat set the value accordingly. This works on **all my other SPZB0001 devices**, *only* this one keeps giving that error.

## How to reproduce it (minimal and precise)

I have tried restarting `zigbee2mqtt` as well as `Home Assistant`, I have re-paired the thermostat (then it'll work for a day or so until this error happens again), I have tried taking out the batteries and inserting them again **without resetting**, which works for all my other SPZB0001 devices when they have an issue like this.

## Debug Info
zigbee2mqtt version: `1.7.1`
CC253X firmware version: `{'type': 'zStack3x0', 'meta': {'transportrev': 2, 'product': 1, 'majorrel': 2, 'minorrel': 7, 'maintrel': 1, 'revision': 20191106}}`
device: `CC2652R`

How can this be fixed? It is kinda strange  that all my other thermostats (same brand, same model) work, except this one. Everything I tried fails except for deleting the device from `configuration.yaml` and re-pairing it. However, when I re-pair it, it will work for a couple of days until this error comes up again.

Thank you for your help.

prankousky, 2019-12-11:
I just deleted my `database.db` and `state.json` files. This resulted in my having to re-pair all my devices (I thought as long as the `configuration.yaml` was there I would *not* have to do this, but hey, learned something 😂).

After then re-pairing everything, it *seems* like all thermostats can be controlled now. However, I'd like to mention this in case somebody else encounters this problem!

If you use the `CC65x` in combination with `SPZB0001` thermostats, it seems like you **need to** have `rtscts: true`. I had tried adding two more coordinators (1x `CC2530`, 2x `CC2531`) to my network because I thought perhaps the SPZB0001 did not work due to the range, but then had to `rtscts: false` for the `CC2530` coordinator. That made ***every single** SPZB0001 **not** be controllable any longer!

I kicked the extra coordinators, removed rtscts, and eventually it worked! I'll leave this issue open because hopefully somebody can tell me how to fix this problem **without having to re-pair my entire fleet of devices**, so if you know, please let me know :)

CodeFinder2, 2020-02-09:
> I had tried adding two more coordinators (1x CC2530, 2x CC2531) to my network

Really? You added *coordinators*, not routers? If yes, that might also be a source of errors because zigbee networks can only have 1 coordinator AFAIK. `database.db` is the file where Z2M manages all paired devices so, yes, deleting it caues Z2M to "forget" those pairing infos (again: AFAIK :laughing: ).

I am not sure what the `rtscts` flag exactly does... :thinking: (would be great if anyone can explain it :see_no_evil: ). I also tried setting this flag (previously false, now true) but it didn't change anything. Still had the same issue with 1 thermostat (4 others working fine), very annoying...

I then tried deleting the device which failed. Force deleting however worked. I then reset the device (by pressing all 3 button for 10s) in order to repair it and it is also working now (let's see, how long...).

prankousky, 2020-02-09:
> Really? You added coordinators, not routers? 
Nope, my mistake. Obviously I added routers, not coordinators. 

> If yes, that might also be a source of errors because zigbee networks can only have 1 coordinator AFAIK.
I might be wrong here, but if I understand correctly, I could have 100 coordinators. They would simply not connect to my existing network. 

Either way, this problem has been solved. I **only** use *one* coordinator now, **no** routers. Most devices work just fine with it, only the `SPZB0001` still throw fits. Some work just fine (I change temperature in `Home Assistant`, a second or two later, the thermostat will be set accordingly), but some act weird (I change temperature, do **not** get any updates about it being actually set, but the thermostat *will positively* adapt to the set temperature; however, when I reload Home Assistant, instead of displaying the actual temperature it has been set to -and adapted to on the thermostat as well-, it will show whatever temperature had been set previously!).

This is really bad. Most downstairs thermostats are supposed to be at (currently) 15°C between 10pm and 8.30am. When I check from upstairs, they will be displayed as whatever they had been set to. So I'll walk downstairs to manually adjust them, then realize they had actually been set to 15°C, but it did not register via mqtt. I have done this while running `mosquitto_sub` in the command line as well as looking at my Home Assistant web interface. Neither will show anything about the "new" temperature (usually when changing them, there is a long json message will all kinds of details). However, this is missing. 

When I

* force delete (remove device entirely from `/opt/zigbee2mqtt/data/configuration.yaml`)
* restart z2m
* completely reset device
* enable pairing
* re-pair device
* properly name device in configuration
* restart z2m

Then it will work for a while, except for **one** thermostat, that will still not report back it's current temperature **all the sudden**. It had previously sometimes reported, sometimes not. Now it won't report at all.

CodeFinder2, 2020-02-09:
Sh*t...mine is already not working anymore ... :-( (code 240)

sti0, 2020-02-09:
Did you update to the latest z2m release? There are several changes after 1.7.1.

Please provide the firmware version of the devices by publishing a get command (https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigdevicesget).

sti0, 2020-02-09:
@prankousky you could test this enhancement mentioned in the other issue 
https://github.com/Koenkk/zigbee-herdsman-converters/issues/715#issuecomment-582074844

It should be a way better performance but even not perfect yet.

Koenkk, 2020-02-12:
Let's keep this in https://github.com/Koenkk/zigbee-herdsman-converters/issues/715

closed 2020-02-12 by Koenkk