Koenkk/zigbee2mqtt #2501: Sinope Thermostat

yannickasselin, 2019-12-10:
Hi,

I bought 2 Sinope thermostats because they are shown in the supported devices.
I was able to pair them. But that's all. I am not able to get any value from them or send any commands to them. Do I need to send a refresh command or something like that?
I found this on the internet which seems to be the Smarthings driver for these thermostat:

https://support.sinopetech.com/en/wp-content/uploads/sites/4/2019/08/Sinope-Technologies-Zigbee-Thermostat-V.1.0.0-SVN-547.txt

Please help!

I am using Node-Red

yannickasselin, 2019-12-11:
I figured out that the model was not supported in devices.js so I copied the sinope TH1123ZB and changed the model number to TH1400ZB.

Now I get the thermostat status every hour:
[image]

But I still cannot change the heating setpoint.
I tried sending a json payload like this:
[image]
But the heating setpoint still does not change.

Any idea what I need to send?

Koenkk, 2019-12-12:
That should be the correct payload, what do you see in the zigbee2mqtt log when sending it?

yannickasselin, 2019-12-12:
Nothing appears in the terminal when I send the command.

But I noticed that I get this warning once in a while:

zigbee2mqtt:warn  2019-12-12T21:33:16: No converter available for 'TH1400ZB' with cluster 'msTemperatureMeasurement' and type 'attributeReport' and data '{"measuredValue":2141}'

Koenkk, 2019-12-13:
Even with debug logging enabled?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

yannickasselin, 2019-12-14:
Nothing appears in the terminal. Even with "log_level: debug".

I noticed something else. I also have an Ikea bulb. If I send a command like this:
Topic: maison/lumiere/set
Payload (string): ON
Everything works fine

But if I send a command like this:
Topic: maison/lumiere
Payload (json): {"set": "ON"}
It does not work. And nothing shows up in the terminal. Same as with the thermostat. I also have an Ikea outlet, same behavior.

Works:
[image]

Does not work:
[image]

I must be doing something wrong. I also tried with MQTT Explorer and I cannot make the outlet and the light work. But I can with Node-Red!!!

yannickasselin, 2019-12-14:
I made some progress. I figured that I still need the "set" topic when sending a json payload:

This works for the light:
[image]

So now back to the thermostat. Do I need to add something in my topic when sending the payload?

yannickasselin, 2019-12-14:
I got it working by adding "set" into the topic.
Now this works, I can change the heating setpoint:
[image]

Do I need to use "set" for every command?
Where can I see what are all the different commands I can send to different device and what are the different values I can send?

yannickasselin, 2019-12-14:
I also figured out how to get the status back from the thermostat by sending this command ("get"):
[image]

Then I get this back:
[image]

I wonder how someone is supposed to figure it all out by himself or is it all written somewhere and I just cannot find it?

Thank you

Koenkk, 2019-12-15:
Yes you always need to use the `/set` postfix when sending commands to the devices. Most of the times people use Home Assistant, therefore it is not needed to figure out the commands manually. 

I agree that is is not properly documented, to figure it out find the used `toZigbee` converter in https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/toZigbee.js#L488, the key is the value you should put in the JSON.

yannickasselin, 2019-12-17:
Thank you, I will look at this file.

I also would like to make these device work with zigbee2mqtt. How hard would it be?

https://www.sinopetech.com/en/product/smart-water-valve-%c2%be-in-wi-fi-zigbee/?setLang=en

https://www.sinopetech.com/en/product/smart-electrical-load-controller-50-a-zigbee/?setLang=en

Thank you.

Koenkk, 2019-12-17:
@yannickasselin probably not easy as they are not really a common type of device (like a bulb e.g.). Therefore it may require sniffing the traffic with the original gateway.

closed 2020-02-22 by stale[bot]

frenchzed, 2020-02-29:
Don't know if that can help, but SmartThings drivers source code is available, and seem to show some zigbee communications: https://github.com/SmartThingsCommunity/SmartThingsPublic/tree/master/devicetypes/sinope-technologies