Koenkk/zigbee2mqtt #2503: (new device) Nordtronic Box Dimmer 2.0 - don't show up in HA

AThomsen, 2019-12-10:
Device: https://nordtronic.dk/shop/box-dimmer-2-712p.html
Unknown to zigbee2mqtt. Trying to add it using the "unknown device" guidelines.

I believe the device is equal to some of the other devices that looks exactly the same except for logos.

I added to devices.js:
```
  // Nordtronic
    {
        zigbeeModel: ['BoxDIM2 98425031'],
        model: 'Box Dimmer 2.0 98425031',
        vendor: 'Nordtronic2',
        description: 'ZigBee AC dimmer',
        supports: 'on/off, brightness',
        fromZigbee: [fz.brightness, fz.on_off, fz.ignore_light_brightness_report],
        toZigbee: [tz.light_onoff_brightness, tz.ignore_transition],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
            await configureReporting.onOff(endpoint);
        },
    },
```
After that and paring the log shows
```
zigbee2mqtt:info  2019-12-10T19:02:13: Device '0x680ae2fffe3a4000' joined
zigbee2mqtt:info  2019-12-10T19:02:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x680ae2fffe3a4000"}}'
zigbee2mqtt:info  2019-12-10T19:02:13: Starting interview of '0x680ae2fffe3a4000'
zigbee2mqtt:info  2019-12-10T19:02:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x680ae2fffe3a4000"}}'
zigbee2mqtt:info  2019-12-10T19:02:14: Successfully interviewed '0x680ae2fffe3a4000', device has successfully been paired
zigbee2mqtt:info  2019-12-10T19:02:14: Device '0x680ae2fffe3a4000' is supported, identified as: Nordtronic2 ZigBee AC dimmer (Box Dimmer 2.0 98425031)
zigbee2mqtt:info  2019-12-10T19:02:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x680ae2fffe3a4000","model":"Box Dimmer 2.0 98425031","vendor":"Nordtronic2","description":"ZigBee AC dimmer","supported":true}}'
zigbee2mqtt:info  2019-12-10T19:02:14: Configuring '0x680ae2fffe3a4000'
zigbee2mqtt:info  2019-12-10T19:02:15: Succesfully configured '0x680ae2fffe3a4000'
```

But the devices does not show up in Devices or Entities in HA.

Then I added to `/zigbee2mqtt-1.7.1/lib/extension/homeassistant.js` (in the running container, followed by a docker restart ...):
```
'BoxDIM2 98425031': [cfg.light_brightness],
```

Still nothing. 
Discovery works as I have no problem adding other devices (Ikea).

I installed zigbee2mqttAssistant. I can see the device but it reports:
> ***NO COMPONENTS** Maybe this device is not (yet) supported by Zigbee2Mqtt?*

## Debug Info
zigbee2mqtt version 1.7.1 
HASS.IO. All latest versions.

Koenkk, 2019-12-11:
Instead of `'BoxDIM2 98425031': [cfg.light_brightness],` add `'Box Dimmer 2.0 98425031': [cfg.light_brightness],` (the `model` should be used, not `zigbeeModel`).

AThomsen, 2019-12-13:
That's it! Tried everything but this :-)

Will create a PR ASAP. Should there be a PR for homeassistant.js as well or just devices.js?

Koenkk, 2019-12-14:
For both

closed 2019-12-14 by AThomsen