Koenkk/zigbee2mqtt #2504: Xiaomi Wall Switch's relay randomly disconnects in decoupled mode

ArkadiuszNiemiec, 2019-12-10:
# Bug Report
## What happened

I have few Xiaomi Wall Switches (no neutral, single and double versions), configured in zigbee2mqtt, working in decoupled mode (relay disconnected from switch). I want the relay to be always shorted but sometimes the relay just disconnects.

I have a Home Assistant automation that turns it back on but it's not a solution as the light turns on whet the relay is reconnected.

The same thing does not happen when the switch is in wired mode. I have also not observed the same behavior on Deconz firmware.

## What did you expect to happen

The relay should stay disconnected if I not order to connect it.

## How to reproduce it (minimal and precise)

Have a Xiaomi Wall switch set in decouple mode and wait few hours/days until the relay disconnects and the lamp goes off.

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: 20190608

Logs from the time this happened:
```
zigbee2mqtt:debug 2019-12-09T22:38:47: Received Zigbee message from 'corridor_wall_switch', type 'attributeReport', cluster 'genOnOff', data '{"61440":39699968,"onOff":0}' from endpoint 2 with groupID 0
zigbee2mqtt:info  2019-12-09T22:38:47: MQTT publish: topic 'zigbee2mqtt/corridor_wall_switch', payload '{"state":"OFF","linkquality":57}'
zigbee2mqtt:debug 2019-12-09T22:38:48: Received Zigbee message from 'corridor_wall_switch', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3300,"3":0,"5":94,"8":9750,"10":0,"100":0,"101":0,"110":0,"111":255,"153":0,"155":0}}' from endpoint 1 with groupID 0
```

closed 2020-02-15 by stale[bot]

ArkadiuszNiemiec, 2020-02-16:
I believe this is a problem with low load on the switch. I will install additional capacitors and it should fix it.

juzis28, 2020-03-29:
Exactly same problem here. Did you manage to solve this random relay disconnection?

ArkadiuszNiemiec, 2020-03-29:
@juzis28 I have tried a capacitor and a bypass and it did not solve the problem completely. Now it turns off but I can immediately turn it back on using a Home Assistant automation.