Koenkk/zigbee2mqtt #2505: Group functionality

kisseler, 2019-12-11:
I'm trying to create a group with several heating thermostates (Eurotronic SPZB0001).
```
...
devices:
  '0x00158d0001922cfd':
    friendly_name: Single_Thermostate_1
    retain: false
  '0x00158d000192d200':
    friendly_name: Single_Thermostate_2
    retain: false
  '0x00158d0001922e03':
    friendly_name: Single_Thermostate_3
    retain: false
  '0x00158d0001922d77':
    friendly_name: Single_Thermostate_4
    retain: false
...
groups:
...
  '2':
    friendly_name: My_Thermostate_Group
    retain: true
    devices:
      - '0x00158d0001922d77'
      - '0x00158d0001922e03'
      - '0x00158d000192d200'
      - '0x00158d0001922cfd'
...
```
Each single member I can address without any issue by my OH Item (e. g. `>[mqtt:zigbee/Single_Thermostate_1/set/current_heating_setpoint:command:*:default]`)
But sending the same command to my group (`>[mqtt:zigbee/My_Thermostate_Group/set/current_heating_setpoint:command:*:default]`), nothing happens:
```
2019-12-11 12:59:44 AM - debug: Received MQTT message on 'zigbee/My_Thermostate_Group/set/current_heating_setpoint' with data '20.0'
2019-12-11 12:59:44 AM - error: No converter available for 'current_heating_setpoint' (20.0)
```
Is the group functionality depending on the device abilities or just not there, or am I missing anything about how groups are working?

Thanks!!!

jdtsmith, 2019-12-12:
I get the same issues for groups of Fyrtur blinds:

```
Dec 12 01:40:14 testpi npm[11513]: zigbee2mqtt:debug 2019-12-12 01:40:14: Received MQTT message on 'zigbee2mqtt/ikea_blind_group/set' with data '{"position": 50}'
Dec 12 01:40:14 testpi npm[11513]: zigbee2mqtt:error 2019-12-12 01:40:14: No converter available for 'position' (50)
```

It also seems to give the group entities attributes like `state:ON/OFF` and `brightness: 0`, so perhaps groups of lights are the only thing currently supported by groups for group commands?

sjorge, 2019-12-12:
https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/groups.js#L78 probably needs te also include `position`, `current_heating_setpoint`, ...

Koenkk, 2019-12-13:
Added both heating setpoint and cover control via group so this is possible now.

jdtsmith, 2019-12-13:
Thanks.  There is no converter error anymore, but blind groups still don't respond and the group properties include `brightness` despite only blinds being in the group. 

```
Dec 13 18:19:16 testpi npm[1124]: zigbee2mqtt:info  2019-12-13 18:19:16: Connected to MQTT server
Dec 13 18:19:16 testpi npm[1124]: zigbee2mqtt:info  2019-12-13 18:19:16: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Dec 13 18:19:16 testpi npm[1124]: zigbee2mqtt:info  2019-12-13 18:19:16: MQTT publish: topic 'zigbee2mqtt/ikea_updown_remote_1', payload '{"linkquality":78,"battery":60}'
Dec 13 18:19:16 testpi npm[1124]: zigbee2mqtt:info  2019-12-13 18:19:16: MQTT publish: topic 'zigbee2mqtt/ikea_repeater_1', payload '{"linkquality":194}'
Dec 13 18:19:16 testpi npm[1124]: zigbee2mqtt:info  2019-12-13 18:19:16: MQTT publish: topic 'zigbee2mqtt/ikea_blind_group', payload '{"state":"OFF","brightness":0}'
Dec 13 18:19:16 testpi npm[1124]: zigbee2mqtt:info  2019-12-13 18:19:16: MQTT publish: topic 'zigbee2mqtt/ikea_blind_famroom_30in_N', payload '{"position":0,"linkquality":76,"battery":100,"state":"OFF","brightness":0}'
Dec 13 18:19:16 testpi npm[1124]: zigbee2mqtt:info  2019-12-13 18:19:16: MQTT publish: topic 'zigbee2mqtt/ikea_blind_group', payload '{"state":"OFF","brightness":0}'
Dec 13 18:19:16 testpi npm[1124]: zigbee2mqtt:info  2019-12-13 18:19:16: MQTT publish: topic 'zigbee2mqtt/ikea_blind_famroom_30in_S', payload '{"state":"OFF","brightness":0,"position":0,"linkquality":63,"battery":100}'
Dec 13 18:19:17 testpi npm[1124]: zigbee2mqtt:info  2019-12-13 18:19:17: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.7.1+dev","commit":"087d1b8","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
Dec 13 18:19:30 testpi npm[1124]: zigbee2mqtt:debug 2019-12-13 18:19:30: Received MQTT message on 'zigbee2mqtt/ikea_blind_group/set/position' with data '50'
Dec 13 18:19:30 testpi npm[1124]: zigbee2mqtt:debug 2019-12-13 18:19:30: Publishing 'set' 'position' to 'ikea_blind_group'
Dec 13 18:20:18 testpi npm[1124]: zigbee2mqtt:debug 2019-12-13 18:20:18: Received MQTT message on 'zigbee2mqtt/ikea_blind_group/set/position' with data '50'
Dec 13 18:20:18 testpi npm[1124]: zigbee2mqtt:debug 2019-12-13 18:20:18: Publishing 'set' 'position' to 'ikea_blind_group'
```

Koenkk, 2019-12-14:
Could you sniff the traffic? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

jdtsmith, 2019-12-14:
Away from the blinds for about a week, will look at it then. @kisseler how'd it work for your thermostat(s)?

kisseler, 2019-12-15:
I just updated and tried out: Still error:
`error 2019-12-15 22:36:01: No converter available for 'current_heating_setpoint' (18)`
and no group function for this topic.
@Koenkk you added the line
`zigbeeHerdsmanConverters.toZigbeeConverters.thermostat_occupied_heating_setpoint,`
My necessary topic is `current_heating_setpoint` (and in case of the Bitron Wall Themostat `occupied_heating_setpoint`). So I tried adding a new line
`    zigbeeHerdsmanConverters.toZigbeeConverters.current_heating_setpoint,`
which led to error
```
debug 2019-12-15 22:56:41: Received MQTT message on 'zigbee/OG_KuecheWohnzimmer_Heizkoerperthermostate/set/current_heating_setpoint' with data '18'
error 2019-12-15 22:56:41: Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'key' of undefined
    at converters.find (/opt/zigbee2mqtt/lib/extension/entityPublish.js:146:56)
    at Array.find (<anonymous>)
    at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/entityPublish.js:146:42)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:322:44)
    at Controller.onMQTTMessage (/opt/zigbee2mqtt/lib/controller.js:237:14)
    at MQTT.emit (events.js:198:13)
    at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.js:86:14)
    at MqttClient.emit (events.js:198:13)
    at MqttClient._handlePublish (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:1162:12)
    at MqttClient._handlePacket (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:351:12))
```

jdtsmith, 2019-12-21:
So with the latest dev I'm still unable to make group commands work for blinds groups.  There is no converter error any longer, but the commands silently fail, e.g.:

```
Dec 21 21:53:36 testpi npm[5901]: zigbee2mqtt:debug 2019-12-21 21:53:36: Received MQTT message on 'zigbee2mqtt/ikea_blind_group/set/position' with data '70'
Dec 21 21:53:36 testpi npm[5901]: zigbee2mqtt:debug 2019-12-21 21:53:36: Publishing 'set' 'position' to 'ikea_blind_group'
```

Do you expect a response for individual endpoints in the group here?

Koenkk, 2019-12-23:
@jdtsmith could you sniff the traffic when the command is being send out?https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start

jdtsmith, 2019-12-23:
I only have one stick. Can it sniff and control at the same time?  The page wasn’t clear on that.

Koenkk, 2019-12-26:
Unfortunately not

jdtsmith, 2019-12-26:
Hmm... any other ideas to try?  It's interesting that `bind`ing to a group of blinds does allow a remote to control the group (though not perfectly: see #2496).  So group command does work at some level.  Just not via MQTT.

Koenkk, 2019-12-31:
@jdtsmith I did some sniffing but it looks that the command is being send out correctly, I guess the only thing you can do is compare the packet send by zigbee2mqtt (group) vs dimmer.

jdtsmith, 2020-01-05:
Did you mean vs. the remote which is bound to the group?  Are the command packets expected to be the same?  How can I inspect them without a sniffer?  Thanks.

Koenkk, 2020-01-05:
Yes, unfortunately you will need a sniffer to do so.

jdtsmith, 2020-01-08:
> Yes, unfortunately you will need a sniffer to do so.

Here's my confusion.  If I simply `systemctl stop zigbee2mqtt`, the previously bound remote starts fully functioning immediately to control the group: start, stop, and reverse direction all work, just as with direct binding (but no repeater in use!).  I can even hold the remote button up or down and then release it to stop.  _But only if z2m is not running._  Once I start z2m, the remote can start motion, but not stop or interrupt it. 

So somehow z2m is interfering with the remote's commands, so it must be that it has some log record, yes?  Unfortunately I don't see anything relevant.  [Here is a cleaned-up simplified log](https://pastebin.com/raw/hkE3upg8).  You can see I start the blinds opening with a up-press on the remote at 00:17:13, and both blinds report a decreasing `currentPositionLiftPercentage` (which maps to an _increasing_ `position` reported over MQTT).    At 00:17:24 I hit the remote the other direction (`commandDownClose`).  But the blinds keep on moving up.   If I had not had z2m running, this would have worked.  

Is there more intensive logging than `log_level: debug` I should try?  Thanks.

Koenkk, 2020-01-09:
@jdtsmith yes it's really strange, you can try enabling the herdsman debug logging to see if there is anything suspicious there, if not you need to sniff.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

jdtsmith, 2020-01-26:
Not really sure if there is anything abnormal, but [here's some herdsman logging](https://pastebin.com/raw/rmvg84gV). Another interesting thing is that as **soon** as you stop z2m while the blinds are moving, if there was a pending button press that had been ignored (even for several seconds), the blinds will stop at that point.

jdtsmith, 2020-01-29:
An [interesting note](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1121#issuecomment-577385291) from the deconz people, who confirm they have up/down/stop/interrupt working with remotes bound to Fyrtur blinds:

> Damn, I didn’t realise this earlier. The blind supports groups (that’s how the open/close remote communicates with it), but it doesn’t understand On or Off. The REST API plugin translates state.on to Open and Close for the blind (or any window covering device), but to On and Off for groups (and obviously for lights). It doesn’t know that this is a group of window covering devices.

I wonder if we're in the same situation: sending on/off to the group as if it's a group of switches/lights/etc.?

Koenkk, 2020-01-29:
@jdtsmith this is only the case when controlling blinds through a group via zigbee2mqtt, not via the remote. When controlling through zigbee2mqtt you can already do this with the `position` (not `state`, maybe this should be added??)

Koenkk, 2020-01-29:
@jdtsmith one thing that came to my mind about the remote not working fully when zigbee2mqtt is running; do you have any automations that respond to the remotes button clicks? (E.g. in Home Assistant) Can you try disabling these and check if the remote now works with zigbee2mqtt running?

jdtsmith, 2020-01-29:
Good though.  I don't have any automations attached to the remote clicks.  So Z2M posts MQTT messages for it, but they fall on deaf ears.  

One other thing the deconz people said: when the blinds are _properly_ bound to the remote, the coordinator can be _OFF_ (as in _unplugged from the wall_) and it still works, because the remote uses a broadcast message to the blinds' group.  This is not the case for me.  The coordinator must be ON, with Z2M not running for full (aside from 1-2sec delay) remote capability.  With the coordinator ON and Z2M running the remote _half-works_ (no interruption or press to go release to stop).  So I'm thinking maybe I'm not actually getting the remote correctly bound to the blinds group at all.  I'm just doing:

```
% mosquitto_pub -t "zigbee2mqtt/bridge/bind/ikea_updown_remote_1" -m "ikea_blind_group"
% mosquitto_pub -t "zigbee2mqtt/bridge/group/ikea_blind_group/add" -m "ikea_blind_famroom_30in_S"
% mosquitto_pub -t "zigbee2mqtt/bridge/group/ikea_blind_group/add" -m "ikea_blind_famroom_30in_M"
% mosquitto_pub -t "zigbee2mqtt/ikea_blind_group/set/position" -m 70
```

The 1st three commands all seem to succeed in the logs (though I must hold the remote button down for the 1st).  The last command does nothing.  And then remote behaves in the way described above. 

Thanks for your continued help.

Koenkk, 2020-01-30:
It's very strange that the last command doesn't work indeed, it should be the same as sending it directly.

I'm not sure how I can help in continuing debugging this as I don't have the required information to. (don't have the device or sniffer logs).

jdtsmith, 2020-01-30:
Would sniffer logs from the deconz side be helpful?  I think they've used those to debug and get these blinds working with remote/etc.

Koenkk, 2020-01-30:
That will probably contain lots of noise so very hard to debug, might also not contain the zigbee2mqtt specific issues. Hopefully someone else can pick this up.

closed 2020-04-07 by stale[bot]

kisseler, 2020-09-09:
Sorry to bother again: Heating season beginns and I give the ZigBee-Thermostates a new try. Since there are a few rooms/zones, that have 2 or more thermostates, group functionality is very substantial I think.

I have in total 11 Eurotronic SPZB0001 thermostates. Creating a group and sending MQTT message to it still leads to converter error `No converter available for 'current_heating_setpoint' (10)`.

I have two questions:

1. Is group functionality working for devices of the same kind supposed to work without configuration/developing? Or do I have to kindly request to add functionality to the developer @Koenkk ?

2. If the system does not support that "out of the box" would it be possible to add devices yourself as a user?

Thank you for your help!!!

Koenkk, 2020-09-09:
@kisseler is this on Zigbee2MQTT 1.14.4 (it should work in that version)

kisseler, 2020-09-09:
Thanks for the hint, I updated from 1.14.3-dev to 1.14.4-dev. Unfortunately same error message: 
`error 2020-09-09 18:17:06: No converter available for 'current_heating_setpoint' (11.5)`
My MQTT message pattern for the group has the same structure as the single thermostates. Do I have to address the group in a different way than the single items?

Group item (OpenHAB):
`Number OG_KuecheWohnzimmer_Thermostat_Setpoint "Raumtemperatur Soll [%.1f °C]" <heating0> (gSollIst_OG_KuecheWohnzimmer, Temperature) {mqtt=">[mqtt:zigbee2mqtt/OG_KuecheWohnzimmer_Heizkoerperthermostate/set/current_heating_setpoint:command:*:default]"}`

Group member items (OpenHAB):
`Number OG_Wohnzimmer_Thermostat_L_Setpoint "Raumtemperatur Soll WZ L [%.1f °C]" <heating0> (gSoll_OG_KuecheWohnzimmer, Temperature) {mqtt=">[mqtt:zigbee2mqtt/OG_Wohnzimmer_Heizkoerperthermostat_L/set/current_heating_setpoint:command:*:default],<[mqtt:zigbee2mqtt/OG_Wohnzimmer_Heizkoerperthermostat_L:state:JSONPATH($.current_heating_setpoint)]"}`

`Number OG_Wohnzimmer_Thermostat_R_Setpoint "Raumtemperatur Soll WZ R [%.1f °C]" <heating0> (gSoll_OG_KuecheWohnzimmer, Temperature) {mqtt=">[mqtt:zigbee2mqtt/OG_Wohnzimmer_Heizkoerperthermostat_R/set/current_heating_setpoint:command:*:default],<[mqtt:zigbee2mqtt/OG_Wohnzimmer_Heizkoerperthermostat_R:state:JSONPATH($.current_heating_setpoint)]"}`

`Number OG_Kueche_Thermostat_L_Setpoint "Raumtemperatur Soll KÜ L [%.1f °C]" <heating0> (gSoll_OG_KuecheWohnzimmer, Temperature) {mqtt=">[mqtt:zigbee2mqtt/OG_Kueche_Heizkoerperthermostat_L/set/current_heating_setpoint:command:*:default],<[mqtt:zigbee2mqtt/OG_Kueche_Heizkoerperthermostat_L:state:JSONPATH($.current_heating_setpoint)]"}`

`Number OG_Kueche_Thermostat_R_Setpoint "Raumtemperatur Soll KÜ R [%.1f °C]" <heating0> (gSoll_OG_KuecheWohnzimmer, Temperature) {mqtt=">[mqtt:zigbee2mqtt/OG_Kueche_Heizkoerperthermostat_R/set/current_heating_setpoint:command:*:default],<[mqtt:zigbee2mqtt/OG_Kueche_Heizkoerperthermostat_R:state:JSONPATH($.current_heating_setpoint)]"}`

Koenkk, 2020-09-09:
Can you provide your `data/database.db`?

kisseler, 2020-09-09:
Is there a way to sanitize/regenerate the database? Resp. isn't it enough to set the groups in the configuration file?

Koenkk, 2020-09-09:
@kisseler when looking at the group where the thermostat should be in, do you see this thermostat in the `members` of the `data/databse.db` entry of this group?

kisseler, 2020-09-09:
Mhh... checking my `database.de` I cannot even find my group name or members. I also tried to manually add the devices (beside writing it directly into `configuration.yaml`.

I attached my database.db 

[database.db.txt](https://github.com/Koenkk/zigbee2mqtt/files/5197610/database.db.txt)

I appreciate your help very much!

Koenkk, 2020-09-10:
@kisseler could you recreate the group? Older Zigbee2MQTT versions did not store this info.

kisseler, 2020-09-16:
Sorry, it took me a while. I reset up the whole system. Now I tried to create groups. With my thermostates it leads to the following error

```
debug 2020-09-16 15:39:38: Received MQTT message on 'zigbee2mqtt/bridge/group/OG_KuecheWohnzimmer_Heizkoerperthermostate/add' with data 'OG_Kueche_Heizkoerperthermostat_R'
info  2020-09-16 15:39:38: Adding 'OG_Kueche_Heizkoerperthermostat_R' to 'OG_KuecheWohnzimmer_Heizkoerperthermostate'
error 2020-09-16 15:40:01: Failed to add from group (Command 0x00158d0003c46fbc/1 genGroups.add({"groupid":2,"groupname":""}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 63844 - 1 - 18 - 4 - 0 after 10000ms))
```
Might that be an indication that those thermostates are not compatible to be run in groups at all?

Koenkk, 2020-09-16:
@kisseler I think that is the case indeed.

kisseler, 2020-09-17:
OK, but in thanks for your effort!!!