Koenkk/zigbee2mqtt #2507: Coordintor firmware changelog and when to upgrade

kiwinol, 2019-12-11:
CC2531 firmware version: 20190223

I would like to know how people decide when to upgrade the coordinator firmware.
I am most interested in bug fixes and stability.

Is there a changelog I can read for the coodinator firmware?

pepp86, 2019-12-11:
Hi, 

you can find the Changelog in the github archive:
https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/CHANGELOG.md

I'm upgrading always to the latest release, as I am also running dev version of z2m.

If everything is stable for you, just stay where you are, if not, upgrade to latest firmware.

kiwinol, 2019-12-11:
@pepp86 thanks, that is exactly what I was looking for :)_

closed 2019-12-11 by kiwinol