Koenkk/zigbee2mqtt #2508: Gledopto GL-C-007 discovered as 008

mrrioes, 2019-12-11:
Hi there,

I tried to pair my Gledopto GL-C-007 (RGB-W) and it was discovered as GL-C-008 (RGB-CCT).
This means I´m able to change the RGB Values but not the white channel. 
When it was paired with my hue bridge (which I want to get rid off) it was added as two different Lamps (one RGB and one White) and I exspected this from z2m as well..
I´m working with the latest version of z2m and Domoticz as frontend and I´m not so confident in dealing with all the db stuff inside of zigbee2mqtt.
Is this a bug or am I doing something wrong?

Koenkk, 2019-12-12:
It is known that older gledopto devices announce themselves as 008 instead 007, this is a bug in the gledopto firmware which cannot be fixed in zigbee2mqtt. As a workaround you can edit the `database.db` and manually change the `modelId` of this device to `GL-C-007`

pixeldoc2000, 2019-12-20:
@mrrioes I have a Gledopto GL-C-007 (recently bought) and it sends the correct Device ID.

What Fireware Version do you have? -> "swBuildId":"2.0.0"

my database.db:
`{"id":34,"type":"Router","ieeeAddr":"0xXXXXXXXXXXXXXXX","nwkAddr":27043,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-007","epList":[11,13],"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attributes":{"zclVersion":1}},"genIdentify":{"dir":{"value":1},"attributes":{}},"genGroups":{"dir":{"value":1},"attributes":{}},"genScenes":{"dir":{"value":1},"attributes":{}},"genOnOff":{"dir":{"value":1},"attributes":{"onOff":1}},"genLevelCtrl":{"dir":{"value":1},"attributes":{"currentLevel":191}},"lightingColorCtrl":{"dir":{"value":1},"attributes":{"currentX":45940,"currentY":19595,"colorTemperature":500}}},"binds":[]},"13":{"profId":49246,"epId":13,"devId":528,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attributes":{}}},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"2.0.0","zclVersion":1,"meta":{}}`

closed 2020-02-25 by stale[bot]