Koenkk/zigbee2mqtt #2509: Linkquality above 100??

z4rn0x, 2019-12-11:
# Bug Report
## What happened
Hello
I noticed since several days that one of my Aqara thermometer has a linkquality above 100...
Here is the MQTT message:
{"temperature":21.04,"linkquality":115,"humidity":37.73,"battery":74,"voltage":2955}

## What did you expect to happen
Is it a bug? Or something that I can fix?

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: 20190608

Kryzek, 2019-12-12:
I am under impression that linkquality is a value from 0 to 255 and 255 is the best quality.

Koenkk, 2019-12-12:
This is indeed a value between 0 and 255.

closed 2019-12-12 by Koenkk

CodeFinder2, 2019-12-21:
Does it make sense to map this to [0,100] (percentage) for improved readability?

Together with a '%' at the end, it would not allow for any misunderstanding.

Koenkk, 2019-12-23:
Yes, but that will be a breaking change so can not be done easily.