Koenkk/zigbee2mqtt #2511: Installation zigbee2mqtt documentation

DMike92, 2019-12-12:
Thanks for your great work!
# Bug Report
## What happened
after 'nmp start' I have
...
zigbee2mqtt:error 2019-12-12T14:02:59: Not connected to MQTT server!
...
## What did you expect to happen
zigbee2mqtt:info  2019-12-12T15:44:05: Connected to MQTT server
## How to reproduce it (minimal and precise)
This error can easily be obtained when mosquitto is not installed.
## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version:

IMO, the documentation on https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html lacks to tell about the necessity of installing Mosquitto. (also in https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html)

Koenkk, 2019-12-12:
Added, thanks!

closed 2019-12-12 by Koenkk

DMike92, 2019-12-12:
> Added, thanks!

Thanks to YOU!