Koenkk/zigbee2mqtt #2512: Crash while starting zigbee-herdsman

StefanSchmelz, 2019-12-12:
I found nothing that fits to the log i got and it happens on raspbian buster. 
Sadly I'm not that good with nodejs, so I paste the npm output starting with the first error and the contents of the logfile. I'm using a Raspberry Pi 3b with raspbian Buster and bought a cc2531 stick which has a zigbee2mqtt firmware already programmed. I contacted the seller and he could not tell me if it is the latest firmware.
I worked through everything i could find of troulbleshooting.

`zigbee2mqtt:error 2019-12-12T17:57:44: Error while starting zigbee-herdsman
zigbee2mqtt:error 2019-12-12T17:57:44: Failed to start zigbee
zigbee2mqtt:error 2019-12-12T17:57:44: Exiting...
zigbee2mqtt:error 2019-12-12T17:57:44: Error: SRSP - SYS - version after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.7.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.7.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-12-12T17_57_44_651Z-debug.log
`

I attached the logfile.
[2019-12-12T17_57_44_651Z-debug.log](https://github.com/Koenkk/zigbee2mqtt/files/3957235/2019-12-12T17_57_44_651Z-debug.log)

Koenkk, 2019-12-13:
Did you check: https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start ?

snakuzzo, 2019-12-13:
Hi,
I've got the same problem.
I'm on Raspbian Buster on an RPI3B and I'm using latest zigbee2mqtt version and an ESP2866 ESPEasy + CC2530 with firmware 1.2 (followed [THIS GUIDE](https://www.zigbee2mqtt.io/information/connecting_cc2530.html)

This is the service log...

```
Dec 13 23:16:33 snakeberry systemd[1]: Started zigbee2mqtt.
Dec 13 23:16:34 snakeberry npm[1361]: > zigbee2mqtt@1.7.1 start /opt/zigbee2mqtt
Dec 13 23:16:34 snakeberry npm[1361]: > node index.js
Dec 13 23:16:37 snakeberry npm[1361]: zigbee2mqtt:info  2019-12-13T22:16:37: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-12-13.23-16-36'
Dec 13 23:16:37 snakeberry npm[1361]: zigbee2mqtt:info  2019-12-13T22:16:37: Starting zigbee2mqtt version 1.7.1 (commit #b459c35)
Dec 13 23:16:37 snakeberry npm[1361]: zigbee2mqtt:info  2019-12-13T22:16:37: Starting zigbee-herdsman...
Dec 13 23:16:44 snakeberry npm[1361]: zigbee2mqtt:error 2019-12-13T22:16:44: Error while starting zigbee-herdsman
Dec 13 23:16:44 snakeberry npm[1361]: zigbee2mqtt:error 2019-12-13T22:16:44: Failed to start zigbee
Dec 13 23:16:44 snakeberry npm[1361]: zigbee2mqtt:error 2019-12-13T22:16:44: Exiting...
Dec 13 23:16:44 snakeberry npm[1361]: zigbee2mqtt:error 2019-12-13T22:16:44: Error: SRSP - SYS - version after 6000ms
Dec 13 23:16:44 snakeberry npm[1361]:     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Dec 13 23:16:44 snakeberry npm[1361]:     at ontimeout (timers.js:436:11)
Dec 13 23:16:44 snakeberry npm[1361]:     at tryOnTimeout (timers.js:300:5)
Dec 13 23:16:44 snakeberry npm[1361]:     at listOnTimeout (timers.js:263:5)
Dec 13 23:16:44 snakeberry npm[1361]:     at Timer.processTimers (timers.js:223:10)
Dec 13 23:16:44 snakeberry npm[1361]: npm ERR! code ELIFECYCLE
Dec 13 23:16:44 snakeberry npm[1361]: npm ERR! errno 1
Dec 13 23:16:44 snakeberry npm[1361]: npm ERR! zigbee2mqtt@1.7.1 start: `node index.js`
Dec 13 23:16:44 snakeberry npm[1361]: npm ERR! Exit status 1
Dec 13 23:16:44 snakeberry npm[1361]: npm ERR!
Dec 13 23:16:44 snakeberry npm[1361]: npm ERR! Failed at the zigbee2mqtt@1.7.1 start script.
Dec 13 23:16:44 snakeberry npm[1361]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Dec 13 23:16:44 snakeberry npm[1361]: npm ERR! A complete log of this run can be found in:
Dec 13 23:16:44 snakeberry npm[1361]: npm ERR!     /home/pi/.npm/_logs/2019-12-13T22_16_44_727Z-debug.log
Dec 13 23:16:44 snakeberry systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Dec 13 23:16:44 snakeberry systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Dec 13 23:16:44 snakeberry systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Dec 13 23:16:44 snakeberry systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 1.
Dec 13 23:16:44 snakeberry systemd[1]: Stopped zigbee2mqtt.
```

I tested the write permission on port /opt/zigbee2mqtt/vusb/zigbee_cc2530 following [FAQ](https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start )...

```
pi@snakeberry:/opt/zigbee2mqtt $ test -w /opt/zigbee2mqtt/vusb/zigbee_cc2530 && echo success || echo failure
success
```

and this is generated logfile content...

```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.11.3
3 info using node@v10.17.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.7.1~prestart: zigbee2mqtt@1.7.1
6 info lifecycle zigbee2mqtt@1.7.1~start: zigbee2mqtt@1.7.1
7 verbose lifecycle zigbee2mqtt@1.7.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.7.1~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.7.1~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.7.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.7.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.7.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.7.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:198:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:198:13)
13 verbose stack     at maybeClose (internal/child_process.js:982:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.7.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.19.75-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.17.0
19 verbose npm  v6.11.3
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.7.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.7.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

I tried also changing pan_id (eg. 6755)...same result

Koenkk, 2019-12-14:
Does it work when connecting it directly to the pi?

snakuzzo, 2019-12-14:
I don't know...
I had a CC2531 connected directly to RPI, and it worked fine, but know I need to move the coordinator in other place and without RPI. I've read about this solution ESP8266 + CC2530 and I'm trying...but I have this problem

snakuzzo, 2019-12-14:
Just another input...
on my ESPEasy configuration tab "Devices" there is an option not mentioned in the guide.

I have `TX Enable Pin` to set...and I leaved `None`, is it correct?

snakuzzo, 2019-12-14:
UPDATE: 

I just tried connecting CC2530 on RPI using CP2102 adapter...same results! :(

spripe, 2020-01-14:
Any update with this issue? I've the same problem

snakuzzo, 2020-01-14:
Read my previous update. Problem was wrong cc2530 firmware. You need v.1.2

ohminy, 2020-01-20:
I have same h/w setting " esp8266 with espeasy + cc2530 "

How would I set seral & port setting??

in the tutorial
serial:
    port: 'tcp://192.168.2.13:20108'

in your post above
serial:
  port: /opt/zigbee2mqtt/vusb/zigbee_cc2530

I guess..
serial:
  port: tcp://192.168.0.41:1775

plz help

snakuzzo, 2020-01-20:
It's changed....You don't need to use socat-usb, so you can directly connect to ESP8266 using TCP

ohminy, 2020-01-21:
> It's changed....You don't need to use socat-usb, so you can directly connect to ESP8266 using TCP

so... what address & port can I use ?
192.168.0.41 is my wemos d1 mini address in my router
1775 is espeasy setting value follow [this](https://www.zigbee2mqtt.io/information/connecting_cc2530.html#setting-up-the-esp8266)
then.. "port: tcp://192.168.0.41:1775" is right setting?

https://github.com/Koenkk/zigbee2mqtt/issues/2794#issue-552646588

snakuzzo, 2020-01-21:
`port: 'tcp://192.168.0.41:1775'`

StefanSchmelz, 2020-01-21:
> Did you check: https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start ?

Jup, I did and I found no Problems.
I also run a RPI3B with Buster. For now I'll wait until my debugger arrives and try flashing the newest firmware. 
I also run the newest version of zigbee2mqtt and connected the cc2531 directly to my raspberry.

closed 2020-03-28 by stale[bot]

herik63, 2020-07-18:
Hallo all, I'm trying same situation but 2 issues: how can i test the serial communication between the module and net? I'm using a mac with Catalina, maybe telnet?
And second issue: is this working in docker environment for zigbee2mqtt?