Koenkk/zigbee2mqtt #2518: Hue switch stopped working (after floating from CC2530 to Bulb router)

davekr8, 2019-12-13:
#1787 Bug Report
## What happened
Hue Switch seems to be move in the mesh from CC2530 router to Hue Bulb and stopped working.

## What did you expect to happen
Not getting tired from removing/repairing devices.

## How to reproduce it (minimal and precise)
Hard to reproduce as the are no indications, when a device moves in the mesh. Maybe this would be a major improvement to log (output via mqtt), when an end device floats in the mesh. 

## Debug Info
zigbee2mqtt version: zigbee2mqtt@1.6.0
CC2530 firmware version:  
 * 20190619 + source routing (coordinator)
 * two times CC2530 router from http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112

Maybe you have some answers to the following questions:

* The last seen flag was updated when I soft-reseted the Switch but no functionality given. How does this fit together?
* Why are such mesh-floatings killing the functionality of the end-device?
* How can I repair such issues without removing an end device and repairing?
* Is there a way to avoid devices floating in the mesh? From my perspective after repairing a lot, that most router-devices are not compliant with the zigbee2mqtt sorftware and break the system by either sleeping or violating timing constraints... this is really a pain using z2m.
* Would it be possible to report end-device movements, modifications in the mesh via mqtt. This would lower the time to get noticed of any issues that are cause by structural changes in the mesh. Some buttons I only hit 2-3 times a week. 

Here are some log (as I tried to reset the switch: hard via pin, soft via 4-buttons down):
This does not give me any hint how to proceed in the right way. I try to stop removing/repairing devices again and again ... 
```
zigbee2mqtt:debug 2019-12-13 9:45:21 AM Received zigbee message of type 'endDeviceAnnce' with data '"0x0017880103e58ad2"' of device 'RWL021' (0x0017880103e58ad2) of endpoint 1
zigbee2mqtt:error 2019-12-13 9:45:25 AM Cannot get the Node Descriptor of the Device: 0x0017880103e58ad2 (Error: Timed out after 10000 ms)
zigbee2mqtt:error 2019-12-13 9:45:45 AM Cannot get the Node Descriptor of the Device: 0x0017880103e58ad2 (Error: Timed out after 10000 ms)
zigbee2mqtt:debug 2019-12-13 9:45:50 AM Received zigbee message of type 'endDeviceAnnce' with data '"0x0017880103e58ad2"' of device 'RWL021' (0x0017880103e58ad2) of endpoint 1
zigbee2mqtt:warn 2019-12-13 9:45:56 AM Failed to configure hue/switch3 (0x0017880103e58ad2) ('undefined') (attempt #2)
zigbee2mqtt:warn 2019-12-13 9:45:56 AM This can be ignored if the device is working properly
zigbee2mqtt:error 2019-12-13 9:46:05 AM Cannot get the Node Descriptor of the Device: 0x0017880103e58ad2 (Error: Timed out after 10000 ms)
zigbee2mqtt:debug 2019-12-13 9:46:07 AM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.router' (0x00124b001bc744b1) of endpoint 8
zigbee2mqtt:info 2019-12-13 9:46:07 AM MQTT publish: topic 'home/zigbee/router/cc2530', payload '{"state":true,"led_state":false,"linkquality":80}'
zigbee2mqtt:debug 2019-12-13 9:46:08 AM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.router' (0x00124b001b7b2341) of endpoint 8
zigbee2mqtt:info 2019-12-13 9:46:08 AM MQTT publish: topic 'home/zigbee/router/cc2530_bed', payload '{"state":true,"led_state":false,"linkquality":33}'
zigbee2mqtt:error 2019-12-13 9:46:25 AM Cannot get the Node Descriptor of the Device: 0x0017880103e58ad2 (Error: Timed out after 10000 ms)
zigbee2mqtt:debug 2019-12-13 9:46:38 AM Received zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":73}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b47fe2) of endpoint 1
zigbee2mqtt:debug 2019-12-13 9:46:38 AM Received zigbee message of type 'devChange' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":73}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b47fe2) of endpoint 1
zigbee2mqtt:debug 2019-12-13 9:46:38 AM Received zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":1}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b47fe2) of endpoint 1
zigbee2mqtt:info 2019-12-13 9:46:39 AM MQTT publish: topic 'home/zigbee/pir/office', payload '{"illuminance":73,"linkquality":80,"occupancy":true,"battery":100,"voltage":3035}'
zigbee2mqtt:debug 2019-12-13 9:46:43 AM Received zigbee message of type 'endDeviceAnnce' with data '"0x0017880103e58ad2"' of device 'RWL021' (0x0017880103e58ad2) of endpoint 1
zigbee2mqtt:debug 2019-12-13 9:46:43 AM Configuring hue/switch3 (0x0017880103e58ad2) ...
```

Bastiencc, 2019-12-13:
I get this on old version. I updated to latest version and now it is way better !

davekr8, 2019-12-13:
Ok. Hope this will be working again in the new version. I really wanted to stop repairing, in the future.

davekr8, 2019-12-13:
Well, after switching to version 1.7.1 it getting much better, .... NOT !
```
Dez 13 15:00:21 raspberrypi3 npm[30717]: zigbee2mqtt:info  2019-12-13T14:00:21: MQTT publish: topic 'home/zigbee/bridge/config', payload '{"version":"1.7.1","commit":"b459c35","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}},"log_level":"debug","permit_join":true}'
Dez 13 15:00:31 raspberrypi3 npm[30717]: zigbee2mqtt:error 2019-12-13T14:00:31: Failed to configure 'hue/switch1', attempt 1 (Error: AREQ - ZDO - bindRsp after 10000ms
Dez 13 15:00:31 raspberrypi3 npm[30717]:     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Dez 13 15:00:31 raspberrypi3 npm[30717]:     at ontimeout (timers.js:436:11)
Dez 13 15:00:31 raspberrypi3 npm[30717]:     at tryOnTimeout (timers.js:300:5)
Dez 13 15:00:31 raspberrypi3 npm[30717]:     at listOnTimeout (timers.js:263:5)
Dez 13 15:00:31 raspberrypi3 npm[30717]:     at Timer.processTimers (timers.js:223:10))
Dez 13 15:00:31 raspberrypi3 npm[30717]: zigbee2mqtt:info  2019-12-13T14:00:31: Configuring 'hue/switch2'
Dez 13 15:00:38 raspberrypi3 npm[30717]: zigbee2mqtt:info  2019-12-13T14:00:38: Succesfully configured 'hue/switch2'
Dez 13 15:00:38 raspberrypi3 npm[30717]: zigbee2mqtt:info  2019-12-13T14:00:38: Configuring 'hue/switch4'
Dez 13 15:00:38 raspberrypi3 npm[30717]: zigbee2mqtt:debug 2019-12-13T14:00:38: Received Zigbee message from 'pir/office', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":59}' from endpoint 1 with groupID 0
Dez 13 15:00:38 raspberrypi3 npm[30717]: zigbee2mqtt:debug 2019-12-13T14:00:38: Received Zigbee message from 'pir/office', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
Dez 13 15:00:39 raspberrypi3 npm[30717]: zigbee2mqtt:info  2019-12-13T14:00:39: MQTT publish: topic 'home/zigbee/pir/office', payload '{"illuminance":59,"linkquality":75,"occupancy":true,"battery":100,"voltage":3035}'
Dez 13 15:00:48 raspberrypi3 npm[30717]: zigbee2mqtt:error 2019-12-13T14:00:48: Failed to configure 'hue/switch4', attempt 1 (Error: AREQ - ZDO - bindRsp after 10000ms
Dez 13 15:00:48 raspberrypi3 npm[30717]:     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Dez 13 15:00:48 raspberrypi3 npm[30717]:     at ontimeout (timers.js:436:11)
Dez 13 15:00:48 raspberrypi3 npm[30717]:     at tryOnTimeout (timers.js:300:5)
Dez 13 15:00:48 raspberrypi3 npm[30717]:     at listOnTimeout (timers.js:263:5)
Dez 13 15:00:48 raspberrypi3 npm[30717]:     at Timer.processTimers (timers.js:223:10))
Dez 13 15:00:48 raspberrypi3 npm[30717]: zigbee2mqtt:info  2019-12-13T14:00:48: Configuring 'hue/switch3'
Dez 13 15:00:58 raspberrypi3 npm[30717]: zigbee2mqtt:error 2019-12-13T14:00:58: Failed to configure 'hue/switch3', attempt 1 (Error: AREQ - ZDO - bindRsp after 10000ms
Dez 13 15:00:58 raspberrypi3 npm[30717]:     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Dez 13 15:00:58 raspberrypi3 npm[30717]:     at ontimeout (timers.js:436:11)
Dez 13 15:00:58 raspberrypi3 npm[30717]:     at tryOnTimeout (timers.js:300:5)
Dez 13 15:00:58 raspberrypi3 npm[30717]:     at listOnTimeout (timers.js:263:5)
Dez 13 15:00:58 raspberrypi3 npm[30717]:     at Timer.processTimers (timers.js:223:10))
```

davekr8, 2019-12-13:
Finally, for the specific switch:
```
Dez 13 15:17:30 raspberrypi3 npm[30717]: zigbee2mqtt:debug 2019-12-13T14:17:30: Device 'hue/switch3' announced itself
Dez 13 15:17:39 raspberrypi3 npm[30717]: zigbee2mqtt:debug 2019-12-13T14:17:39: Received Zigbee message from 'pir/office', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3035,"3":25,"4":17320,"5":56,"6":[0,17],"10":56191,"11":50,"100":0}}' from endpoint 1 with groupID 0
Dez 13 15:17:40 raspberrypi3 npm[30717]: zigbee2mqtt:info  2019-12-13T14:17:40: MQTT publish: topic 'home/zigbee/pir/office', payload '{"illuminance":50,"linkquality":78,"occupancy":true,"battery":100,"voltage":3035}'
Dez 13 15:17:50 raspberrypi3 npm[30717]: zigbee2mqtt:error 2019-12-13T14:17:50: Failed to interview 'hue/switch3', device has not successfully been paired
Dez 13 15:17:50 raspberrypi3 npm[30717]: zigbee2mqtt:info  2019-12-13T14:17:50: MQTT publish: topic 'home/zigbee/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"hue/switch3"}}'
Dez 13 15:17:50 raspberrypi3 npm[30717]: zigbee2mqtt:info  2019-12-13T14:17:50: Configuring 'hue/switch3'
Dez 13 15:18:00 raspberrypi3 npm[30717]: zigbee2mqtt:error 2019-12-13T14:18:00: Failed to configure 'hue/switch3', attempt 3 (Error: AREQ - ZDO - bindRsp after 10000ms
Dez 13 15:18:00 raspberrypi3 npm[30717]:     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
Dez 13 15:18:00 raspberrypi3 npm[30717]:     at ontimeout (timers.js:436:11)
Dez 13 15:18:00 raspberrypi3 npm[30717]:     at tryOnTimeout (timers.js:300:5)
Dez 13 15:18:00 raspberrypi3 npm[30717]:     at listOnTimeout (timers.js:263:5)
Dez 13 15:18:00 raspberrypi3 npm[30717]:     at Timer.processTimers (timers.js:223:10))
```
The others, already working before, seems to be settled, again.

Bastiencc, 2019-12-13:
Only one device is not working? everytime the same one ?

davekr8, 2019-12-13:
Yes, the one switch, which was not working when I started this issue, is still not working. I guess, removing the hue switch and pairing it again would be a workaround, but definitely not a solution...

davekr8, 2019-12-20:
Hmmm... I will start the re-pairingnoia, again, sadly.

davekr8, 2019-12-25:
Any resonance, here?!

Koenkk, 2019-12-26:
I have the same problem with my Hue Outdoor motion sensor, zigbee-herdsman does not discover the route again when the device switches parent. Hope to have a fix for it soon.

davekr8, 2019-12-26:
If you need any support from my setup, please let me know. I leave the switch in the current state.

Koenkk, 2019-12-27:
Thanks, will let you know once I know more

Koenkk, 2020-02-27:
Lets continue in #2693

closed 2020-02-27 by Koenkk