Koenkk/zigbee2mqtt #2520: npm start --- Cannot find module 'is-arrayish'

vollyimnetz, 2019-12-13:
# Bug Report
## What happened
Trying to start with `npm start` but got this error: https://hastebin.com/olulivavej.js

## What did you expect to happen
to get no error, or at least an qualified message whats wrong

## How to reproduce it (minimal and precise)
npm i
npm start

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: CC2531_DEFAULT_20190608.zip
Raspberrypi with nodejs v12.13.1

Koenkk, 2019-12-14:
Can you try `cd /opt/zigbee2mqtt && rm -rf node_modules && npm install`.

vollyimnetz, 2019-12-15:
Hi, thank you for investigating. Yeah, i tried reinstall before.
I tried once more - still didn't work. 
Here is the output of the install: https://hastebin.com/inerifacum.sql

Raspberry-Version
- Operating System: Raspbian GNU/Linux 9 (stretch)
- Kernel: Linux 4.19.66-v7+

Koenkk, 2019-12-15:
Can you try installing it manually? `cd /opt/zigbee2mqtt && npm install is-arrayish`.

vollyimnetz, 2019-12-15:
Thank you very much, that was a very good idea.

I tryed it and i got an output like
` cached data for is-arrayish@latest (sha512-eVRqCvVlZbuw3GrM63ovNSNAeA1K16kaR/LRY/92w0zxQ5/1YzwblUX652i4Xs9RwAGjW9d9y6X88t8OaAJfWQ==) seems to be corrupted. Refreshing cache.`
But after i installed is-arrayish there was the next node-module that failed. So obviously something strange happend to the npm cache.

Here is what helped me in the end:
`
cd /opt/zigbee2mqtt           #for change to correct directory
rm -rf node_modules          #get rid of all the already installed node-modules in the project
npm cache clean --force      #force the node cache to be cleared
npm i                                  #install all the zigbee2mqtt modules again
npm cache verify                 #just to check that the cache is correct this time
npm start                            #start zigbee2mqtt
`

Thank you for your support - even if its not a bug in zigbee2mqtt.

closed 2019-12-15 by vollyimnetz

myronlopes-tuks, 2023-08-17:
this helped me today. Thank you very much for sharing!