Koenkk/zigbee2mqtt #2521: Add Drayton Wiser TRV

arcanefoam, 2019-12-13:
# Current Status
Valve correctly reports local temp, occupied heating setpoint, pi heating demand, and boost (i.e. user request temp change at valve).

Allowed settings are occupied heating set point and keypad lock. 

### Errors
I am unable to change the occupied heating set point to remotely change the temperature setting. I either get a "mac transaction expired" message or no change happens. The valve is close by the router so I assume no link problems. Only thing that seems interesting is that the 513 "out" cluster is on a different endpoint; not sure if somehow set requests must be sent to that endpoint. 
E.g in the log (https://pastebin.com/eCGeTr83), I think the dataRequest should be to endopoint 2 not 1 (see device information below)?

# Device information
## Manufacturer
This is the Drayton Wiser TRV: https://wiser.draytoncontrols.co.uk/radiator-thermostat

## DB Information
```
{ 
  "id": 2,
  "type": "EndDevice",
  "ieeeAddr": "0x086bd7fffe7ed5e4",
  "nwkAddr": 63976,
  "manufId": 4190,
  "manufName": "Schneider Electric",
  "powerSource": "Battery",
  "modelId": "iTRV",
  "epList": [ 1,  2 ],
  "endpoints": {
    "1": {
    "profId": 260,
      "epId": 1,
      "devId": 769,
      "inClusterList": [ 0, 1, 3, 32, 513, 516, 2821, 65027 ],
      "outClusterList": [ 0, 25 ],
      "clusters": {
        ...
      },
      "binds": [
        {
          "cluster": 0,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b000be8a0ed",
          "endpointID": 1
        },
        {
          "cluster": 1,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b000be8a0ed",
          "endpointID": 1
        },
        {
          "cluster": 3,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b000be8a0ed",
          "endpointID": 1
        },
        {
          "cluster": 32,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b000be8a0ed",
          "endpointID": 1
        },
        {
          "cluster": 513,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b000be8a0ed",
          "endpointID": 1
        },
        {
          "cluster": 516,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b000be8a0ed",
          "endpointID": 1
        },
        {
          "cluster": 2821,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b000be8a0ed",
          "endpointID": 1
        }
      ]
    },
    "2": {
      "profId": 260,
      "epId": 2,
      "devId": 7,
      "inClusterList": [],
      "outClusterList": [ 513 ],
      "clusters": {},
      "binds": []
    }
  },
  ...
}
```
# Additions to Converters and Herdsman

## Current device configuration (devices.js)
```
{
        zigbeeModel: ['iTRV'],
        model: 'iTRV',
        vendor: 'Dryton Wiser',
        description: 'Smart heating thermostat',
        supports: 'temperature, battery, keypad lock, heating demand',
        fromZigbee: [
            fz.ignore_basic_report,
            fz.ignore_haDiagnostic,
            fz.ignore_genOta,
            fz.ignore_zclversion_read,
            fz.thermostat_att_report,
            fz.drayton_itrv_battery,
            fz.drayton_user_interface,
            fz.drayton_device_info,
        ],
        toZigbee: [
            tz.thermostat_occupied_heating_setpoint,
            tz.thermostat_keypad_lockout,
        ],
        meta: {configureKey: 2},
        configure: async (device, coordinatorEndpoint) => {
            let endpoint = device.getEndpoint(1);
            const binds = [
                'genBasic', 'genPowerCfg', 'genIdentify', 'genPollCtrl',
                'hvacThermostat', 'hvacUserInterfaceCfg', 'haDiagnostic'
            ];
            await bind(endpoint, coordinatorEndpoint, binds);
            await configureReporting.batteryVoltage(endpoint);
            await configureReporting.thermostatTemperature(endpoint);
            await configureReporting.thermostatOccupiedHeatingSetpoint(endpoint);
            await configureReporting.thermostatPIHeatingDemand(endpoint);
            const userInterfaceConfig = [
                {
                    attribute: 'keypadLockout',
                    minimumReportInterval: repInterval.MINUTE,
                    maximumReportInterval: repInterval.MAX,
                    reportableChange: 0,
                }
            ];
            await endpoint.configureReporting('hvacUserInterfaceCfg', userInterfaceConfig);
            const draytonDeviceConfig = [
                {
                    attribute: 'ALG',
                    minimumReportInterval: repInterval.MINUTE,
                    maximumReportInterval: repInterval.MAX,
                    reportableChange: 0,
                },
                {
                    attribute: 'ADC',
                    minimumReportInterval: repInterval.MINUTE,
                    maximumReportInterval: repInterval.MAX,
                    reportableChange: 0,
                },
                {
                    attribute: 'boost',
                    minimumReportInterval: repInterval.MINUTE,
                    maximumReportInterval: repInterval.MAX,
                    reportableChange: 0,
                }
            ];
            await endpoint.configureReporting('draytonDeviceInfo', draytonDeviceConfig);
        },
    },
```

## Added fromZigbe
```
    // Dryton Wiser
    drayton_device_info: {
        cluster: 'draytonDeviceInfo',
        type: 'attributeReport',
        convert: (model, msg, publish, options) => {
            const result = {};
            const data =  msg.data['deviceInfo'].split(',');
            if (data[0] == 'ALG') {
                // TODO What is ALG
                result['ALG'] = data.slice(1).join(',');
            }
            else if (data[0] == 'ADC') {
                // TODO What is ADC
                result['ADC'] = data.slice(1).join(',');
            }
            else if (data[0] == 'UI') {
                if (data[1] == 'BoostUp') {
                    result['boost'] = 1;
                }
                else if (data[1] == 'BoostDown') {
                    result['boost'] = -1;
                }
                else {
                    result['boost'] = 0;
                }
            }
           return result;
        },
    },
    drayton_itrv_battery: {
        cluster: 'genPowerCfg',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options) => {
            const result = {};
            if (typeof msg.data['batteryVoltage'] == 'number') {
                const battery = {max: 30, min: 22};
                const voltage = msg.data['batteryVoltage'] ;
                result.battery = toPercentage(voltage, battery.min, battery.max);
                result.voltage = voltage / 10;
            }
            if (typeof msg.data['batteryAlarmState'] == 'number') {
                const battLow = msg.data['batteryAlarmState'] ;
                if (battLow) {
                    results['battery_low'] = true;
                } else {
                    results['battery_low'] = false;
                }
            }
            return result;
        },
    },
    drayton_user_interface: {
        cluster: 'hvacUserInterfaceCfg',
        type: 'attributeReport',
        convert: (model, msg, publish, options) => {
            const result = {};
            if (typeof msg.data['keypadLockout'] == 'number') {
                const kLock = msg.data['keypadLockout'] ;
                if (kLock) {
                    result['keypad_lockout'] = true;
                } else {
                    result['keypad_lockout'] = false;
                }
            }
            return result;
        }
    },

    // Dryton ignores
    ignore_genOta: {
        cluster: 'genOta',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options) => null,
    },
    ignore_haDiagnostic: {
        cluster: 'haDiagnostic',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options) => null,
    },
    ignore_zclversion_read: {
        cluster: 'genBasic',
        type: 'read',
        convert: (model, msg, publish, options) => null,
    },
```

## Added to clusters
Required to get boost messages
```
    // Drayton Wiser
    draytonDeviceInfo: {
        ID: 0xFE03, // 65027
        attributes: {
	    deviceInfo: { ID: 32, type: dataType_1.default.charStr },
	},
        commands: {},
        commandsResponse: {}
    }
```

## Debug Info
zigbee2mqtt version: 1.7.1
CC253X firmware version: latest

sjorge, 2019-12-13:
Any reason why fz.thermostat_att_report is not a good starting point? That should handle things like local_temperature, system_node, set_point,...

arcanefoam, 2019-12-13:
fz.thermostat_att_report is there. What else do you have in mind?

sjorge, 2019-12-13:
🤦‍♂ I didn't find it so I searched for it, but I searched from tz.thermostat_att_report so I didn't find it. 

All looks good... kind of want to see if I can find one of those as I am not super happy with the eurotronics one I got as a test.

arcanefoam, 2019-12-13:
Yep, that is the one

sjorge, 2019-12-13:
Great, I may pick one up to compare next month.
Is the reporting of local_temp and such consistent with this one? (As in at a set interval not completely random with big gaps of hours of no updates)

arcanefoam, 2019-12-13:
Seems to be reporting constantly, too much I would think hehehe. Still playing with it as setting temp and keypad doesn't work always, some times it always replies with MAC transaction expires. Trying to figure out why it sometimes works and sometimes no.

sjorge, 2019-12-13:
@Koenkk we can raise the minimal we also the device to report right?

arcanefoam, 2019-12-13:
I think https://github.com/Koenkk/zigbee-herdsman/pull/10 is related to this. I am sometimes seeing a default response error. And in the log I see a lot of zcl messages.

Koenkk, 2019-12-14:
@arcanefoam can you try https://github.com/Koenkk/zigbee-herdsman-converters/issues/715#issuecomment-547073703

@sjorge changing the minimal reportable interval is indeed possible.

arcanefoam, 2019-12-16:
@Koenkk I have loaded the alternative firmware.  In the initial run I am getting an error: 

```
2019-12-16T17:17:31: Failed to configure 'DW_TRV_1', attempt 3 (Error: Don't know value type for 'undefined'
    at Object.IsDataTypeAnalogOrDiscrete  ...\zigbee2mqtt\node_modules\zigbee-herdsman\dist\zcl\utils.js:30:15)
```
Apparently the valve has an undefined type attribute. This is the log: https://pastebin.com/2Vc4Lijs
I "fixed" utils.js so it will graciously handle 'undefined':
```
// throw new Error(`Don't know value type for '${definition_1.DataType[dataType]}'`);
console.warn(`Don't know value type for '${definition_1.DataType[dataType]}'`);
return '';
```
Then ran again, it configured correctly.
Still getting erratic behaviour. Sometimes commands still fail: 
```
zigbee2mqtt:debug 2019-12-16T17:47:31: Received MQTT message on 'house/DW_TRV_1/set' with data '{                             
"keypad_lock": "unlock"                                                                                                       
}'                                                                                                                            
zigbee2mqtt:debug 2019-12-16T17:47:31: Publishing 'set' 'keypad_lock' to 'DW_TRV_1'                                           
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":52876,"destendpoint":1,"srcendpoint":1,"clusterid"
:516,"transid":17,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[16,10,2,1,0,48,0]}} +7s                     
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,17,36,1,140,206,1,1,4,2,17,0,30,7,16,10,2,1,0,48,0,81] +7s        
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +5s                                                      
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +1ms                                           
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [ ] - 100 +1ms                                        
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +7s                                             
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms                                                            
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,240,1,17,39] +6s                                               
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,240,1,17,39] +2ms                                    
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [�☺◄] - 39 +20ms                                    
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":240,"endpoint":1,"transid":17} +11s                
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +3ms                                                            
zigbee2mqtt:error 2019-12-16T17:47:37: Publish 'set' 'keypad_lock' to 'DW_TRV_1' failed: 'Error: Data request failed with erro
r: 'MAC transaction expired' (240)'                                                                                           
zigbee2mqtt:info  2019-12-16T17:47:37: MQTT publish: topic 'house/bridge/log', payload '{"type":"zigbee_publish_error","messag
e":"Publish 'set' 'keypad_lock' to 'DW_TRV_1' failed: 'Error: Data request failed with error: 'MAC transaction expired' (240)'
","meta":{"friendly_name":"DW_TRV_1"}}'                                                                                       
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,25,68,129,0,0,0,0,140,206,1,1,0,92,0,150,85,98,0,0,5,0,12,0,0,0,140,206,
29,53] +47s                                                                                                                   
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,0,0,140,206,1,1,0,92,0,150,85,98,0,0,5,0,12,0,0,
0,140,206,29,53] +2ms                                                                                                         
```

arcanefoam, 2019-12-16:
Here is another log. I think the valve is communicating a lot, but not sure if that is the intended rate:

https://pastebin.com/Rs0KKpMd

radix50, 2019-12-16:
Hi - I'm trying to decode the raw data from the iTRV for use directly with Hubitat..

Are you aware of any documentation on the functions of each endpoint and hot to set the setpoint and read the current temp etc?

I see data strings like this:
)ALG,f,1,215,236,-26,-13,171,35,2469,6,100
*ALG,10,1,215,236,-21,-10,101,40,2539,0,100

5ADC,1333,1338,2,236,-32768,2150,0,126,100,241,239,248
5ADC,1332,1335,2,236,-32768,2150,0,111,100,241,240,248

TRV1: Cluster 0B05: read attr - raw: BF4C010B050A01E0210000, dni: BF4C, endpoint: 01, cluster: 0B05, size: 0A, attrId: E001, encoding: 21, command: 0A, value: 0000

TRV1: Cluster 0B05: read attr - raw: BF4C010B05101C0120BC1D0128CB, dni: BF4C, endpoint: 01, cluster: 0B05, size: 10, attrId: 011C, encoding: 20, command: 0A, value: BC1D0128CB

Any pointers would be greatly appreciated!

arcanefoam, 2019-12-23:
@Koenkk  Seems to be working perfectly with version 1.8 (and the corresponding herdsman improvements)!  How do I open a bug to add it to the supported devices? I can write the wiki page too as I have figured out how to reset and put the valve into pairing mode.

@radix50 I could not find any documentation. The **hvacThermostat** cluster has all the information about the valve (i.e. temp, heating demand, etc), **hvacUserInterfaceCfg** for the keypad lock. I only had to add a new cluster with id 0xFE03 (65027) which provides de ALG and ADC messages you mention, but also provides the "BOOST" info to know if someone manually adjusted the valve, and the MOT (motor) state. I haven't taken the time to decode the information in the ALG and ADC messages as all information needed is provided in the **hvac** clusters. I think the ALG has information of the time/date since the valve has been reset/turned on. I would guess the ADC has the temp and valve information in "voltage" values.

closed 2019-12-23 by arcanefoam

messismore, 2019-12-26:
Awesome news @arcanefoam ! Can't wait for mine to arrive!

The repository for the docs are [over here,](https://github.com/Koenkk/zigbee2mqtt.io) if you did not see it already.

sjorge, 2019-12-26:
I also picked one up, not super happy with the eurotronic one so I hope this will be better!

arcanefoam, 2019-12-26:
I have been using the valve for three/four days know and its working fine. I still sometimes get time-outs when sending the temperature settings but eventually it gets through. Now I am starting to work in the integration with Home Assistant. I think I want to make a custom component that behaves somewhat like the Drayton Wiser mobile app.

I was wondering what was the difference between the in and out clusters. I have also ordered some aqara temp sensors and plan to do a test to see to does the valve temp compares with the room temp (with the aqara further away from the radiator). I would be interested in hooking them together as you mention.

sjorge, 2019-12-26:
@arcanefoam from my understanding, in/out cluster of the same type is like a slot to accept messages of the type.

So the msTemperature output of the aqara could go into the input of the trv... so in as I understand it, the aqara sensor could work like the little optional seperate sensor Drayton also sells. AFAIK that is also just a zigbee temp/humidity sensor.

If it does indeed work like that, that would be lovely as my aqara is placed more central so it is a better reflection of the overal room temp.

I am not sure which way the binding goes, but I think the source should be the sensor and the target the trv.

richdr, 2019-12-30:
@arcanefoam I have a few of these - are the changes in a PR somewhere?  Keen to give it a try!

sjorge, 2019-12-30:
I believe they got merged a while ago, 1.8.0+dev should have it I think.

~ sjorge

arcanefoam, 2019-12-30:
> @arcanefoam I have a few of these - are the changes in a PR somewhere? Keen to give it a try!

Not yet, unless someone else have added them (I have not made the PRs). I am fine-tuning some  final details and will make the PRs. Hopefully by the end of the day :).

arcanefoam, 2019-12-30:
@radix50 After looking at the MQTT messages for some time, this is what I can see:
Given (id in parenthesis)
```
ADC,  1333,  1338,   2,  236, -32768 , 2150,   0,  126,  100,  241,  239,  248
           (1)        (2)   (3)   (4)      (5)        (6)     (7)   (8)   (9)     (10)  (11)   (12)
```

1.  ?
2.  ?
3.  ?
4.  Current temperature (x10)
5. ?
6. Target temperature (setpoint - x100)
7. ?
8. Heating demand. Max value seen here seems 500. 
9. ?
10. ?
11. ?
12. ?

radix50, 2019-12-30:
Hi - thanks for that - 10/11/12 also look like temperatures (x10) , of course - I wonder if they are previous readings?

arcanefoam, 2019-12-30:
Dont think they are temps. As local temp gets closer to setpoint they change. But who knows? hehe

sjorge, 2019-12-30:
piHeatingDemand should be 0-255 (which is a valve position %, so 0=closed, 255=100% open)

So probably not that then if he upperl imit is 500

radix50, 2019-12-30:
Ah OK... it would be great if there was a common repository for devices and their inputs/outputs so they could be more easily implemented on different platforms instead of re-inventing the wheel each time!

arcanefoam, 2019-12-30:
That might be true for other models. The highest I've seen the piHeatingDemand fo up on this valve is 39. Which doing the inverse math of the fromZigbee, translates to a 100 value... so this one might be actually returning 0-100 value.

sjorge, 2019-12-30:
Most (all?) attribute should be in the Z-Stack ZCL docs, there is a pdf floating around somehwere but i can't find the link (on my phone atm).

IIRC there is at least one other TRV that has support that was also returning 0-100, so might be that some incorrectly implement it? If that the case, we can probably just have a different function for those to deal with it.

radix50, 2019-12-30:
Ahh thanks - Interesting...  I might finally get some useful devices working on my Hubitat...

sjorge, 2019-12-30:
I have found it very useful, generally most devices do as the zcl says.
attribute XYZ is range X-Y... ofcourse there are some odd balls in there, like already mentioned piHeatingDemand and piCoolingDemand seem to be 0-255 as % valve open, but also sometimes reported s 0-100... which I guess does technically fit in 0-255 and... is a valve open %, so I would not say either is wrong.

But overal it's a good starting point!

sjorge, 2020-01-02:
Also of interest, I found https://download.schneider-electric.com/files?p_enDocType=User+guide&p_File_Name=MEG5935_0000_HW_en.pdf&p_Doc_Ref=MEG5935_0000_HW_en which is the english manual online. As the one I got from amazon (not yet installed) only came with german instructions... should be useful for the pairing instructions near the bottom to create a support page.

arcanefoam, 2020-01-02:
@richdr Here are the PRs:
[Docs](https://github.com/Koenkk/zigbee2mqtt.io/pull/190)
[Converters](https://github.com/Koenkk/zigbee-herdsman-converters/pull/871)
[Herdsman](https://github.com/Koenkk/zigbee-herdsman/pull/119)

arcanefoam, 2020-01-16:
Have 5 of these babies at the house now. But having a lot of issues sending commands to them. I have to send the set temperature command multiple times to get a response. Any ideas? Even after I set the temperature it returns to the default (20) after some time. I think we are missing a command that let's the valve know it has been added to a "system". We would need someone with a sniffer and a Wiser hub to do some debugging.
Also, boost events are persistent. So we can no identify to events of the same type. Perhaps we need to create a command to send "Boost: Off" or similar?

Any help improving this work is appreciated

davidscottparker, 2020-01-16:
I only have 4 "babies", but have been using them in their standard configuration for about a year and they work really well. I managed to read the first on my zigbee2mqtt network, literally minutes ago, so will be able to assist a little.

I also have the room stat and the boiler controller as well, but not seen any mention of these, so I'm guessing that these have not yet been integrated with zigbee2mqtt yet?

sjorge, 2020-01-16:
The roomstat thing, seems interesting... but the website is very vague on it... like it can control multiple rooms but only works as a temp sensor for one?

davidscottparker, 2020-01-16:
The room stat controls multiple iTRVs, but for the SAME room, in my case, 3 radiators.

arcanefoam, 2020-01-16:
Yes, only the iTVRs as I didn't wanted to get the router/boiler control and
lock into their ecosystem.
I am writing a Home Assistant platform thst mimics the phone app, with a
separate switch for the boiler. Schedule and boiler control are working. If
the communication with the valves were stable it would make me a happy
panda and we could provide a really nice project to the community

davidscottparker, 2020-01-16:
I am connecting via nodeRED and as you, am now only interested in direct control of the iTRVs

arcanefoam, 2020-01-19:
For some unknown reason the babies are behaving better now. Tried reading/setting "system state" and"running state" from hvacThermostat to see if the boost info modified any of these values, no luck. I also tried adding a command to change the boost value, but not sure if "UI,BoostOff" is the message the valve is expecting to receive to clear the flag. From my analysis, it now seems the valves want to return to their "default" state (set point to 20) after 2 hours and 13 minutes. You can see the spikes in the following image. Schedule is running smoothly, boiler activated as needed, house at wanted temps, looking good so far.

[image]

sjorge, 2020-01-19:
I think according to the manual that is normal, boost + or - is temporary for 1hour.

From the manual
```
Changing the room temperature Is it too cool? +2 °C (1 h)Start a temperature pulse. Increase the room temperature by 2 °C for 1 hour.Is it too warm? -2 °C (1 h)Start a temperature pulse. Decrease the room tempera-ture by 2 °C for 1 hour
```

I need to swap my TRV back to this one, currently have the eurotronic one back because I did a few PR's for that.

I want to play with lowering the max reporting interval, Eurotronic had the suggested value at 10 minutes, after changing it to that, the unit has been much more stable and I have less timeouts talking to it. We could also try setting a reporting interval for the boost value... not sure it will do much though but it is worth experimenting with.

> boiler activated as needed

What are you using to control the boiler?

arcanefoam, 2020-01-19:
I am not boosting valve via +/- twist, they just act up.

I am using an AC switch, my boiler uses a live (230V) signal for control,
so I used one of these:
https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.co.uk%2Fulk%2Fitm%2F303114657380

It also provides a really powerful router and all my valves connect to it
instead of the USB router. I also left my mechanic thermostat wired in
parallel so I have a fall back if the pc/software fails.

sand3rk, 2020-01-25:
I get this error in Zigbeemqtt log:
`zigbee2mqtt:error 2020-01-25 11:41:37: Failed to configure '0x086bd7fffedfee28', attempt 1 (Error: Cluster with key 'draytonDeviceInfo' does not exist
    at Object.getCluster (/zigbee2mqtt-1.9.0/node_modules/zigbee-herdsman/dist/zcl/utils.js:59:15)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.9.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:215:39)
    at Generator.next (<anonymous>)
    at /zigbee2mqtt-1.9.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/zigbee2mqtt-1.9.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:4:12)
    at Endpoint.configureReporting (/zigbee2mqtt-1.9.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:213:16)
    at Object.configure (/zigbee2mqtt-1.9.0/node_modules/zigbee-herdsman-converters/devices.js:7044:28)
    at async DeviceConfigure.configure (/zigbee2mqtt-1.9.0/lib/extension/deviceConfigure.js:99:13))`

sjorge, 2020-01-25:
Are both herdsman and herdsman-conveters up to date? Those were renamed to the ‘wiser’ prefix.

~ sjorge

arcanefoam, 2020-01-25:
You might need the latest develop versions. Or copy the cluster.js from the
herdsman repo to your local install into the node_modules/zigbee_herdsman

sjorge, 2020-01-25:
Interesting, I tried to join mine again (Eberle, but same unit) and I'm not having any luck either.
I'm not seeing an error though, it's just not joining.

sjorge, 2020-01-25:
I managed to get mine to rejoin, I hit the same error and have a solution.
Will open a PR in a few minutes.

sjorge, 2020-01-25:
@sand3rk can you try with the changes from https://github.com/Koenkk/zigbee-herdsman-converters/pull/939 applied to devices.js... although I am now hitting a different error but I've never seen that one before.

sand3rk, 2020-01-25:
> 
> You might need the latest develop versions. Or copy the cluster.js from the herdsman repo to your local install into the node_modules/zigbee_herdsman On Sat, 25 Jan 2020, 11:53 Jorge Schrauwen, <notifications@github.com> wrote:
> […](#)

I don't see a cluster.js file in the repo. 
I just have zigbee2mqtt installed through hassio and get the error. How do I install herdsman or herdsman-converters? I don't see a devices.js anywhere, just have /share/zigbee2mqtt before the config folder.

sjorge, 2020-01-25:
It should be under `node_modules/zigbee-herdsman-converters`

sand3rk, 2020-01-25:
I only have the log folder under /share/zigbee2mqtt.

sand3rk, 2020-01-25:
I have to follow this guide i guess: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#hassio-addon

Where can I find the lastest dev version of cluster.js?

sjorge, 2020-01-25:
> I also tried adding a command to change the boost value, but not sure if "UI,BoostOff" is the message the valve is expecting to receive to clear the flag.

I swapped back to the wiser trv to check the error earlier today. I hit the error, after fixing it still fails to configure the `wiserDeviceInfo` endpoint for reporting. I do get updates to boostUp and boostDown (maybe we should rename those to be up,down,none ?) but it never goes to None for me... probably because the reporting for that endpoint failed to configure.

I also found no way on the TRV to turn off boost it is either up or down...  kind of sucks.

> I only have the log folder under /share/zigbee2mqtt
I'm not familiar with how home assitant plugins work, but it seems you can just drop it into the path you mentioned: https://github.com/danielwelch/hassio-zigbee2mqtt/blob/master/zigbee2mqtt-edge/run.sh#L38

There seems to be some code that copies it to the right location.

sand3rk, 2020-01-25:
>https://raw.githubusercontent.com/Koenkk/zigbee-herdsman-converters/8afbb59b7c71c3d8488de7c735645c3e92ed40e8/devices.js
>
> The one from dev is still broken, I fixed the error... but I got a new one now on configuring one of the reporting endpoints. (But at least the device shows up now)

Thanks, is this link the fixed one?

sand3rk, 2020-01-25:
I now get the follow error:

`zigbee2mqtt:error 2020-01-25 17:32:41: Failed to configure '0x086bd7fffedfee28', attempt 1 (Error: AREQ - ZDO - bindRsp after 10000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.9.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7))`

But the error from before is gone

sjorge, 2020-01-25:
Yes, it should get you past the error you are having. But you might hit another one after I haven't been able to fix yet.

sjorge, 2020-01-25:
@sand3rk did you remove and rejoin the device? It looks like zigbee2mqtt could not reach it to do the configure step.

sand3rk, 2020-01-25:
@sjorge , no not yet. I will try that.

arcanefoam, 2020-01-25:
My valves do not report any configuration errors. I might get the timeout
on one attempt, but then it succeeeds.

Regarding the boost I have tried adding toZigbee functions to send data to
the deviceInfo and tried sending BoostOff, BoostNone and others but still
no luck.
I also tried sending system_state but it does not change the valves
behavior.

I still feel that some other configuration/command is needed to tell the
valve that it is part of a system. Looking at my Home Assistant graphs it
seems that every two hours that I get a occupied_heating_setpoint spike to
20 degrees the valve opens (pi_heating_demand =100) so it kind of ovverides
the setpoint I am sending it.
It seems it might be something in the deviceInfo cluster but with no
information from the manufacturer we are blind. Unless someone has a wiser
hub and does some sniffing.

arcanefoam, 2020-01-25:
I am also getting timeouts occasionally when sending setpoint changes. I
really don't understand much of the zigbee protocol or how herdsman work or
else I would do some more debugging.

arcanefoam, 2020-01-25:
This are the setpoint spikes and the heating demand.

sand3rk, 2020-01-26:
I re-paired, but still got this error:
`zigbee2mqtt:error 2020-01-26 13:52:03: Failed to configure '0x086bd7fffedfee28', attempt 3 (Error: Cluster with key 'draytonDeviceInfo' does not exist
    at Object.getCluster (/zigbee2mqtt-1.9.0/node_modules/zigbee-herdsman/dist/zcl/utils.js:59:15)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.9.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:215:39)
    at Generator.next (<anonymous>)
    at /zigbee2mqtt-1.9.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/zigbee2mqtt-1.9.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:4:12)
    at Endpoint.configureReporting (/zigbee2mqtt-1.9.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:213:16)
    at Object.configure (/zigbee2mqtt-1.9.0/node_modules/zigbee-herdsman-converters/devices.js:7044:28)
    at async DeviceConfigure.configure (/zigbee2mqtt-1.9.0/lib/extension/deviceConfigure.js:99:13))`

But climate card is still grey, it does not respond to the card, but it does not say: No converter available.

[12 more replies]