Koenkk/zigbee2mqtt #2526: socat dependency for zigbee2mqtt service

mr2c12, 2019-12-14:
Socat service should be required and started before zigbee2mqtt.
This should be the "improved" version:

> [Unit]
> Description=zigbee2mqtt
> Wants=socat-vusb.service
> After=network.target socat-vusb.service
> 
> [Service]
> ExecStart=/usr/bin/npm start
> WorkingDirectory=/opt/zigbee2mqtt
> StandardOutput=inherit
> StandardError=inherit
> Restart=always
> User=pi
> 
> [Install]
> WantedBy=multi-user.target

Please note te "Wants" and "After" changes.

tavalin, 2019-12-14:
Excuse my ignorance but afaik z2m doesn't have a dependency on socat so how was this work for those that don't have socat installed?

mr2c12, 2019-12-14:
You are correct, z2m does not directly depends on socat.
But, if you use CC2530 sniffer connected to an esp8266, socat is required to connect to the serial port. In this case (mine :) ), if socat does not start *before* z2m, you get an "unable to connect to serial port" error.

Koenkk, 2019-12-15:
Thanks, added to https://www.zigbee2mqtt.io/information/connecting_cc2530.html

closed 2019-12-15 by Koenkk