Koenkk/zigbee2mqtt #2527: Device is listed as joined, but doesnt work and not in config.yaml

GMFalka, 2019-12-14:
Hi!

Got a Tradfri dimmer, that worked for a while, then not.. Now i cant repair it to the stick.

Tried removing/force removing it from the config or resetting the device but didnt work. 

I can directly pair the dimmer to a tradfri bulb the "IKEA way" and it works. So the dimmer is functional.

The log states the dimmer as:
```
info  2019-12-14 17:57:51: Currently 14 devices are joined:
info  2019-12-14 17:57:51: 0x000b57fffe8ff31d (0x000b57fffe8ff31d): ICTC-G-1 - IKEA TRADFRI wireless dimmer (EndDevice)
```
Thats probably why i cant repair it to the stick.
Tried removing the database.db, but still couldnt pair with zigbee2mqtt. Theres no log about the pairing even in debug mode.

Any help would be appreciated

Koenkk, 2019-12-15:
Duplicate of/possible solution: https://github.com/Koenkk/zigbee2mqtt/issues/2343#issuecomment-565146154

closed 2019-12-15 by Koenkk

GMFalka, 2019-12-17:
> Duplicate of/possible solution: [#2343 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/2343#issuecomment-565146154)

Thank you @Koenkk!! Reflashing the stick solved it.

I actually bought a few extra IKEA puck and bulb, i could not pair any of them so. Now it works! :)