Koenkk/zigbee2mqtt #2528: More info about group in HA

myradon, 2019-12-14:
Koen,

I think zigbee2mqtt is awesome. As a z2m and HA newbie it's quite a steep learning curve. It took me 3 days to figure out why I can't add a z2m-group to HA. Please could you add a simple text "it's not a HA group" with a [simple example maybe](https://community.home-assistant.io/t/error-configuring-new-mqtt-group-in-ha-zigbee2mqtt-group/154647/5). Because configuration naming and terminology of both platforms are much alike, I got lost. 

Cheers and keep up the good work!

Koenkk, 2019-12-15:
I guess you mean https://www.zigbee2mqtt.io/integration/home_assistant.html#Groups , what exactly do you want me to change there?

myradon, 2019-12-15:
I mean it's a z2m-group is not a group in HA/Domoticz/OpenHAB etc. As a newbie I started fiddling in HA with groups. Took me quite some time to figure this out. I believe when you emphasize this in your docs it will be clearer to newcomers.

Koenkk, 2019-12-16:
So what should I put there? If you provide me with the correct text I will put it there.

myradon, 2019-12-16:
Groups are not auto-discovered. Grouped devices in zigbee2mqtt will be exposed as a single device in your home automation system.  The following example will show you how to add a group of lights to Home Assistant:

`
light:
- platform: mqtt
  schema: json
  name: [MY_GROUP]
  command_topic: "zigbee2mqtt/[GROUP_FRIENDLY_NAME]/set"
  state_topic: "zigbee2mqtt/[GROUP_FRIENDLY_NAME]"

  // Modify according to features supported by all devices in group
  color_temp: true
  brightness: true
  rgb: true
` 
@Koenkk how about the sensor-aspect of grouping? What sensor payload does Z2M use for a group of identical devices?

Koenkk, 2019-12-17:
Thanks, updated.

Groups are only able to control a group of devices, to receive values (sensor) the entities itself are used.

myradon, 2019-12-17:
Niets te danken!

Koenkk, 2019-12-17:
:smile: I will close this one

closed 2019-12-17 by Koenkk