Koenkk/zigbee2mqtt #2531: Trying to implement a own device, need help with general device mode.

rs-development, 2019-12-14:
No Issue but a request for further guidelines.

I am trying to program my own devices. So far I am able to connect to the Zigbee network and zigbee2mqtt recognises the device.

I am trying to implement a simple on-off switch.

Are there any DIY projects out there? I I would not like to reinvent the wheel.
Is there any possibility to restart the zigbee2mqtt interview?
Where do I have to send requests?

I am very thankful for every help!

rs-development, 2019-12-15:
So after a long night, I managed to imitate a existing device. It is also added to HomeAssistant.

I tried to send a SET request to my device via the zigbee2mqtt/0xf4ce36358b01d6c3/set topic but it fails with:

    Received MQTT message on 'zigbee2mqtt/0xf4ce36358b01d6c3/set' with data '{
      "state": "ON",
      "brightness": 255, 
    }
    '
    debug 2019-12-15T13:12:37: Publishing 'set' 'state' to '0xf4ce36358b01d6c3'
    error 2019-12-15T13:12:38: Publish 'set' 'state' to '0xf4ce36358b01d6c3' failed: 'Error: Data request failed with error: 'No network route' (205)'
    info  2019-12-15T13:12:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0xf4ce36358b01d6c3' failed: 'Error: Data request failed with error: 'No network route' (205)'","meta":{"friendly_name":"0xf4ce36358b01d6c3"}}'

Is this a problem with my device attributes?

Koenkk, 2019-12-15:
Are you sure the device is online? Btw there are many diy devices already supported, see https://www.zigbee2mqtt.io/information/supported_devices.html#custom-devices-diy

closed 2020-02-13 by Koenkk