Koenkk/zigbee2mqtt #2532: Running in docker on Synology does not see zigbee stick

cnoork, 2019-12-14:
# Bug Report
## What happened
I try to run zigbee2mqtt on my Synology NAS with DSM 6.2.2
Created a symlink for the device to /dev/ttyZigbee, the script for adding a line in allow was not possible because the file structure is different on Synology, but found it in "/volume1/@docker/containers/" and placed the content "c 166:0 rwm" there. In the docker-compose I have "devices: - /dev/ttyZigbee:/dev/ttyZigbee". Tested the device path on HASS config and see the device there. But unfortunatly the zigbee2mqtt container crashes and restart only with the message 

- zigbee2mqtt:error 2019-12-14T21:49:24: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyZigbee'
- npm ERR! code ELIFECYCLE
- npm ERR! errno 1"

## What did you expect to happen
That the container starts

## How to reproduce it (minimal and precise)
Start/reinstall the container

## Debug Info
zigbee2mqtt version: 1.7.1 (commit #b459c35)
CC253X firmware version: CC2531_DEFAULT_20190608

cnoork, 2019-12-16:
I have tested the container and stick with the same config files on a Raspberry pi 4 and there it was running directly. So it looks like the trouble is in combination with the Synology NAS DS918+.  I have already tested by adding extra rights to the physical host:
  environment:
      - UID=0
      - GID=0

I have also updated to the latest release zigbee2mqtt@1.8.0. But I think it is with passing the usb stick to the docker container (accepting/rights). Unfortunately I cant connect to the shell to check that because it then already restarting. I know my zwave stick is working in a docker container and is passed through at the same way on my NAS, even tested the zigbee stick with the homeassistant container and see the stick in the dev list. My NAS has a x86_64 architecture and did not check if the downloaded image is correct. Just to think of possibilities where I can look.

cnoork, 2019-12-22:
does someone have some tips how I can debug this issue, would really like to run the container on the Synology.

Is there a possibility to start the container without starting the zigbee daemon?

Koenkk, 2019-12-23:
@cnoork sorry but I don't have any experience with Synology so can't help you further. 

> Is there a possibility to start the container without starting the zigbee daemon?

This is not possible, maybe you can check the steps from https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start

cnoork, 2019-12-26:
Hi koenkk, thnx for your reaction, I will check the link and when I have progress I will off course add my results. Currently I have the workaround with raspberry pi but would like to get it to my NAS eventually.

closed 2020-03-03 by stale[bot]

cgarciaq, 2020-12-01:
I have the same problem with docker and synology.
Maybe the issue is that we don't have read or write permission to /dev/ttyACM0 ?

$ lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-3         0451:16a8:0009 02  2.00   12MBit/s 50mA 2IFs (Texas Instruments TI CC2531 USB CDC __0X00124B001949CAC2)
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F008498AADA5A77)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub

$ ll /dev/ttyACM0 
crw------- 1 root root 166, 0 Dec  1 12:11 /dev/ttyACM0

alekamos, 2021-02-06:
If nothing showed up under /dev/ttyACM0 or /dev/ttyUSB0 it may be something related to usb driver. 
I solved manually update the driver following [this](https://zwaveguide.com/synology-z-wave-usb-stick-usage-guide/).
Anyway i don't think is a zigbee2mqtt issue, more dsm and usbkey problem.