Koenkk/zigbee2mqtt #2534: TypeError: Cannot read property 'type' of undefined - with 3 Philips/OSRAM plugs

0neV1s10n, 2019-12-15:
Hi,

Recently I noticed errors regarding these 3 devices:

AB3257001NJ - OSRAM Smart+ plug (Router)
046677552343 - Philips Hue smart plug bluetooth (Router)
9290012607 - Philips Hue motion sensor (EndDevice)

Running Zigbee2MQTT v 1.7.1.
Plugs can still be switched on/off.

Here are the logs extractx:

`error 2019-12-14T22:01:08: Failed to configure 'AB3257001NJ - OSRAM Smart+ plug (Router)', attempt 1 (TypeError: shepherd.find is not a function
    at Object.configure (/zigbee2mqtt-1.7.1/node_modules/zigbee-herdsman-converters/devices.js:1580:37)
    at DeviceConfigure.configure (/zigbee2mqtt-1.7.1/lib/extension/deviceConfigure.js:66:32)
    at DeviceConfigure.onZigbeeStarted (/zigbee2mqtt-1.7.1/lib/extension/deviceConfigure.js:41:28)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.7.1/lib/controller.js:320:44))
...
info  2019-12-14T22:01:08: Configuring '046677552343 - Philips Hue smart plug bluetooth (Router)'
error 2019-12-14T22:01:08: Failed to configure '046677552343 - Philips Hue smart plug bluetooth (Router)', attempt 1 (ReferenceError: bind is not defined
    at Object.configure (/zigbee2mqtt-1.7.1/node_modules/zigbee-herdsman-converters/devices.js:1213:13)
    at DeviceConfigure.configure (/zigbee2mqtt-1.7.1/lib/extension/deviceConfigure.js:66:32)
    at DeviceConfigure.onZigbeeStarted (/zigbee2mqtt-1.7.1/lib/extension/deviceConfigure.js:41:28))
...
info  2019-12-14T22:01:08: Configuring '9290012607 - Philips Hue motion sensor (EndDevice)'
error 2019-12-14T22:01:08: Failed to configure '9290012607 - Philips Hue motion sensor (EndDevice)', attempt 1 (TypeError: shepherd.find is not a function
    at Object.configure (/zigbee2mqtt-1.7.1/node_modules/zigbee-herdsman-converters/devices.js:1133:37)
    at DeviceConfigure.configure (/zigbee2mqtt-1.7.1/lib/extension/deviceConfigure.js:66:32)
    at DeviceConfigure.onZigbeeStarted (/zigbee2mqtt-1.7.1/lib/extension/deviceConfigure.js:41:28))

zigbee2mqtt:error 2019-12-15T07:35:11: Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at mappedDevice.fromZigbee.filter (/zigbee2mqtt-1.7.1/lib/extension/deviceReceive.js:83:42)
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeEvent (/zigbee2mqtt-1.7.1/lib/extension/deviceReceive.js:82:52)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.7.1/lib/controller.js:320:44)
    at Controller.onZigbeeEvent (/zigbee2mqtt-1.7.1/lib/controller.js:228:14)
    at Zigbee.emit (events.js:198:13)
    at Controller.herdsman.on (/zigbee2mqtt-1.7.1/lib/zigbee.js:57:52)
    at Controller.emit (events.js:198:13)
    at Controller.<anonymous> (/zigbee2mqtt-1.7.1/node_modules/zigbee-herdsman/dist/controller/controller.js:415:22)
    at Generator.next (<anonymous>))`

...

`error 2019-12-14T22:22:46: Failed to configure '0x7cb03eaa0a018841', attempt 3 (TypeError: shepherd.find is not a function
    at Object.configure (/zigbee2mqtt-1.7.1/node_modules/zigbee-herdsman-converters/devices.js:1580:37)
    at DeviceConfigure.configure (/zigbee2mqtt-1.7.1/lib/extension/deviceConfigure.js:66:32)
    at DeviceConfigure.onZigbeeEvent (/zigbee2mqtt-1.7.1/lib/extension/deviceConfigure.js:49:18)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.7.1/lib/controller.js:320:44)
    at Controller.onZigbeeEvent (/zigbee2mqtt-1.7.1/lib/controller.js:228:14)
    at Zigbee.emit (events.js:198:13)
    at Controller.herdsman.on (/zigbee2mqtt-1.7.1/lib/zigbee.js:57:52)
    at Controller.emit (events.js:198:13)
    at Controller.<anonymous> (/zigbee2mqtt-1.7.1/node_modules/zigbee-herdsman/dist/controller/controller.js:415:22)
    at Generator.next (<anonymous>))
error 2019-12-14T22:28:39: Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at mappedDevice.fromZigbee.filter (/zigbee2mqtt-1.7.1/lib/extension/deviceReceive.js:83:42)
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeEvent (/zigbee2mqtt-1.7.1/lib/extension/deviceReceive.js:82:52)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.7.1/lib/controller.js:320:44)
    at Controller.onZigbeeEvent (/zigbee2mqtt-1.7.1/lib/controller.js:228:14)
    at Zigbee.emit (events.js:198:13)
    at Controller.herdsman.on (/zigbee2mqtt-1.7.1/lib/zigbee.js:57:52)
    at Controller.emit (events.js:198:13)
    at Controller.<anonymous> (/zigbee2mqtt-1.7.1/node_modules/zigbee-herdsman/dist/controller/controller.js:415:22)
    at Generator.next (<anonymous>))
info  2019-12-14T22:30:31: Configuring '0x0017880104b49be4'
error 2019-12-14T22:30:31: Failed to configure '0x0017880104b49be4', attempt 2 (TypeError: shepherd.find is not a function
    at Object.configure (/zigbee2mqtt-1.7.1/node_modules/zigbee-herdsman-converters/devices.js:1133:37)
    at DeviceConfigure.configure (/zigbee2mqtt-1.7.1/lib/extension/deviceConfigure.js:66:32)
    at DeviceConfigure.onZigbeeEvent (/zigbee2mqtt-1.7.1/lib/extension/deviceConfigure.js:49:18)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.7.1/lib/controller.js:320:44))`

Any idea what could cause these issues ?

Many thanks for your support.

Koenkk, 2019-12-15:
Do you have a custom zigbee-shepherd-converters mapped? (when wanted to add support for a new device). If so please remove this as it seems your version is outdated and not compatible.

0neV1s10n, 2019-12-15:
Hey,
Indeed, I thought the device.js would be replaced with zigbee2mqtt updates, but one must actually remove the file in order to use the default one.
Thanks.

closed 2019-12-16 by Koenkk