Koenkk/zigbee2mqtt #2535: UnhandledPromiseRejectionWarning and no messages received after that

mr2c12, 2019-12-15:
#1875 Bug Report
## What happened
The zigbee2mqtt stop sending messages two times in the last 24 hours. In the service status I get the error:

> dic 15 08:49:06 raspberrypi npm[19379]: zigbee2mqtt:error 2019-12-15T07:49:06: Publish 'set' 'state' to '0x000d6ffffe150371' failed: 'Error: SRSP - AF - dataRequest after 6000ms'
> dic 15 08:49:06 raspberrypi npm[19379]: zigbee2mqtt:info  2019-12-15T07:49:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x000d6ffffe150371' failed: 'Error: SRSP - AF - dataR
> dic 15 08:50:37 raspberrypi npm[19379]: (node:19394) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
> dic 15 08:50:37 raspberrypi npm[19379]:     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
> dic 15 08:50:37 raspberrypi npm[19379]:     at ontimeout (timers.js:436:11)
> dic 15 08:50:37 raspberrypi npm[19379]:     at tryOnTimeout (timers.js:300:5)
> dic 15 08:50:37 raspberrypi npm[19379]:     at listOnTimeout (timers.js:263:5)
> dic 15 08:50:37 raspberrypi npm[19379]:     at Timer.processTimers (timers.js:223:10)
> dic 15 08:50:37 raspberrypi npm[19379]: (node:19394) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an a
> dic 15 08:50:37 raspberrypi npm[19379]: (node:19394) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are 

And thereafter, no messages are sent. My zigbee devices are isolated.

I suspect there is something wrong with Socat (I am using CC2530 connected to an ESP8266). When this situation occurs, I have to restart both socat and zigbee2mqtt.

By the way, is there a method to "ping" zigbee2mqtt? I would like to implement some rule in Home Assistant, so that it can restart the necessary services on its own.

Koenkk, 2019-12-15:
Can you try connecting the CC2530 directly to your system running zigbee2mqtt (by it's nature the setup with the ESP8266 is much more prone to instability issues due to wifi connection issues).

mr2c12, 2019-12-15:
Yeah, sure. Is the "GPIO" connection as stable as the one via usb? (I would prefer to connect the CC2530 to the GPIO pins)

Koenkk, 2019-12-15:
Yes

mr2c12, 2019-12-18:
Still hadn't time to rewire the cc2530 to the rpi GPIO, but I made some tests..

Setting the queue message limit to 1 does not improve the situation.
Anyway, it seems that this problem occurs more frequently after I pair one device. Once I paired all my devices and reboot the coordinator/socat/zigbee2mqtt, the whole system has been up and running for 48 hours.

I will update the issue as soon as I have some news.

mr2c12, 2020-01-29:
Once switched to a wired connection, the problem disapperared.

closed 2020-01-29 by mr2c12