Koenkk/zigbee2mqtt #2536: Need a consult with coordinator hardware and firmware and other

iStitch07, 2019-12-15:
Hello
I need a consultation about coordinator hardware and firmware.

Preambula:
Now I have coordinator on cc2531 and 24 devices, include 12 routers (8 - IKEA LED1650R5, 1 - Xiaomi Smart Plug, 2 - IKEA LED1649C5 and 1 - CC2530 router)

My Zigbee Network map now looks like this:
[image]

Problem: after I add yesterday 8 ikea leds, I sometime randomly receive errors like this:

```
zigbee2mqtt:error 2019-12-14T18:58:36: Publish 'set' 'state' to 'Living Room Main Light 8' failed: 'Error: Data request failed with error: 'No network route' (205)',

zigbee2mqtt:info  2019-12-15T12:57:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'Living Room Main Light 5' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'","meta":{"friendly_name":"Living Room Main Light 5"}}'
```

Question:
As I understand, I need to change my coordinator firmware to source_routing?
And what better use for coordinator (now I use cc2531 without antenna, but waiting parcel with both)?

cc2531 like this
[image]

or cc2530 like this
[image]

And procedure: 
before flashing I need to delete all my connected devices? 
If yes, how? Just send friendly name to zigbee2mqtt/bridge/config/remove? it will enough?

after flashing, I need to connect to coordinator first my cc2530 router, after all my others routers (bulbs, plugs) and only after this connect my sensors?

and about in one month I will receive 2 additional cc2530 routers. Can I pair its after all others or better way before wait its and connect all cc2530 firstly?

Thanks for help )

Koenkk, 2019-12-15:
> As I understand, I need to change my coordinator firmware to source_routing?

Yes

If you plan to buy a new coordinator anyway, I would recommend the CC26X2R1 (much more future proof) (https://www.zigbee2mqtt.io/information/supported_adapters.html)

Good luck!

iStitch07, 2019-12-15:
Okay, I ordered CC26X2R1, but I not know how it connect to raspberry :)

Koenkk, 2019-12-15:
You can easily connect it via USB (cable is included)

closed 2019-12-15 by iStitch07