Koenkk/zigbee2mqtt #2537: Brightness step command

osresearch, 2019-12-15:
It seems there should be a way to adjust the current brightness of a lamp based on the definition of `genLevelCntrl` / `stepWithOnOff` in [zigbee-herdsman/src/zcl/definition/cluster.ts](https://github.com/Koenkk/zigbee-herdsman/blob/958889fce3ed6274b2b855b1fae23f0379db1eee/src/zcl/definition/cluster.ts#L543), although I don't see a way to generate the command from the mqtt messages to the zigbee2mqtt server. Am I missing it somewhere?

Koenkk, 2019-12-15:
It's indeed not supported yet.

kluszczyn, 2020-01-02:
@Koenkk, do I assume correctly, that with this implementation it will be possible to increase/decrease brightness without providing new brightness value/without knowing current brightness? This would be awesome!

Koenkk, 2020-01-05:
Yes you are right

closed 2020-03-12 by stale[bot]