Koenkk/zigbee2mqtt #2538: unable to start docker container zigbee2mqtt 1.8.0

mpuff, 2019-12-15:
Hello,

I tried to start the Docker container (zigbee2mqtt 1.8.0) and wanted to test the tcp serial connection to a CC2531 stick.

When i start the container with the config  i get these error:

>Using /app/data as data directory
> zigbee2mqtt@1.8.0 start /app
> node index.js
zigbee2mqtt:info  2019-12-15 21:10:27: Logging to console and directory: '/app/data/log/2019-12-15.21-10-27'
zigbee2mqtt:info  2019-12-15 21:10:27: Starting zigbee2mqtt version 1.8.0 (commit #da4d26a)
zigbee2mqtt:info  2019-12-15 21:10:27: Starting zigbee-herdsman...
zigbee2mqtt:error 2019-12-15 21:10:34: Error while starting zigbee-herdsman
zigbee2mqtt:error 2019-12-15 21:10:34: Failed to start zigbee
zigbee2mqtt:error 2019-12-15 21:10:34: Exiting...
zigbee2mqtt:error 2019-12-15 21:10:34: Error: SRSP - SYS - version after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.8.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.8.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-12-15T20_10_34_935Z-debug.log`

in the log file i have these informations:

`info  2019-12-15 21:10:27: Logging to console and directory: '/app/data/log/2019-12-15.21-10-27'
info  2019-12-15 21:10:27: Starting zigbee2mqtt version 1.8.0 (commit #da4d26a)
info  2019-12-15 21:10:27: Starting zigbee-herdsman...
error 2019-12-15 21:10:34: Error while starting zigbee-herdsman
`

These are the config file:

homeassistant: false

permit_join: false

mqtt:
  base_topic: zigbee2mqtt_test
  server: 'mqtt://192.168.178.10'
  include_device_information: true

serial:
  port: tcp://192.168.178.29:20108

mpuff, 2019-12-16:
i have vorget the '' in the config by the serial configuration.
serial:
port: 'tcp://192.168.178.29:20108'

now the container is starting!

closed 2019-12-16 by mpuff

SAOPP, 2021-11-14:
Hey guys!

I faced with the next:

1. I got rpi2 with cc2531 and configuring ser2net
2. Got this config and port listening also on my rpi2:
```
connection: &con01
  accepter: tcp,ipv4,20108
  connector: serialdev,/dev/ttyACM0,115200n81,local
  options:
    kickolduser: true
```
`tcp        0      0 0.0.0.0:20108           0.0.0.0:*               LISTEN      4084/ser2net`
3. Got VM with one local z2m instance is running, and trying to deploy another z2m instance for testing this tcp connection to my rpi2
4. This is my compose for deploy my first local z2m instance:
```
  z2m-2531-pve:
    image: koenkk/zigbee2mqtt:latest-dev
    container_name: z2m-2531-pve
    hostname: z2m-2531-pve
    network_mode: host
    privileged: true
#    stdin_open: true
#    tty: true
    restart: unless-stopped
    volumes:
      - /home/docker_storage/z2m-2531:/app/data
      - /run/udev:/run/udev:ro
    devices:
      - /dev/ttyACM0:/dev/ttyACM0
    environment:
      - TZ=Europe/Kiev
```
5. And in the end, my compose for second z2m-remote to my rpi2:
```
  zigbee2mqtt-hall:
    image: koenkk/zigbee2mqtt:1.18.3
    container_name: zigbee2mqtt-hall
    hostname: z2m-hall-remote
    network_mode: host
    privileged: true
#    stdin_open: true
#    tty: true
    restart: unless-stopped
    volumes:
      - /home/docker_storage/z2m_hall_remote:/app/data
      - /run/udev:/run/udev:ro
#    devices:
#      - /dev/ttyACM0:/dev/ttyACM0
    environment:
      - TZ=Europe/Kiev
```
and z2m confoguration os port:
```
serial:
  # Location of CC2531 USB sniffer
  port: 'tcp://192.168.0.23:20108'
```
The question is, what I need to use in device section in my remote z2m instance? 'Cuz right now I got 'socket error'... My first local z2m instance working with `/dev/ttyACM0` in my VM. Or, what I do wrong? Thanks.