Koenkk/zigbee2mqtt #2539: 8718696449691 - Philips Hue White Single bulb B22 (Router) ==> Not supported (Router)

0neV1s10n, 2019-12-15:
Hi again ;-)
I recently added 2 Phlipes Hue bulbs, bought in a duo box.
One is recognized juste fine, the other not:

```
zigbee2mqtt:info  2019-12-15T21:07:00: 0x0017880104cac47d (0x0017880104cac47d): 8718696449691 - Philips Hue White Single bulb B22 (Router)
zigbee2mqtt:info  2019-12-15T21:07:00: 0x0017880104cac47f (0x0017880104cac47f): Not supported (Router)
```

Any idea ?
Thanks.

Kryzek, 2019-12-16:
What is model for non-supported device? For supported it is '8718696449691' - if this is different for another device it might be another revision of the same bulb and have different modelID - hence "different device".

0neV1s10n, 2019-12-16:
Hi Kryzek,
The non-supported device has Model:9290011370B on it.
Does it ring a bell ? I have to check the supported one. But I would be surpirised if one single box contains 2 different models. I will check that.

Koenkk, 2019-12-16:
Can you share your `database.db`?

0neV1s10n, 2019-12-16:
Supported:

`{"id":9,"type":"Router","ieeeAddr":"0x0017880104cac47d","nwkAddr":53353,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"LWB010","epList":[11,242],"endpoints":{"11":{"profId":49246,"epId":11,"devId":256,"inClusterList":[0,3,4,5,6,8,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LWB010","manufacturerName":"Philips","powerSource":1,"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20170908","swBuildId":"1.29.0_r21169"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":2,"stackVersion":1,"zclVersion":1,"interviewCompleted":false,"meta":{}}
`

Non-supported:
`{"id":10,"type":"Router","ieeeAddr":"0x0017880104cac47f","nwkAddr":29599,"manufId":4107,"epList":[11,242],"endpoints":{"11":{"profId":49246,"epId":11,"devId":256,"inClusterList":[0,3,4,5,6,8,4096],"outClusterList":[25],"clusters":{},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{}}`

Koenkk, 2019-12-16:
`"interviewCompleted":false` -> the device has not successfully been interviewed, try factory resetting the device to retrigger the interview. You can reset it via touchlink: https://www.zigbee2mqtt.io/information/touchlink

0neV1s10n, 2019-12-17:
Hey,

Other devices also have `"interviewCompleted":false` but indeed, after 3 resets, the device appears properly in database:

```
{"id":9,"type":"Router","ieeeAddr":"0x0017880104cac47d","nwkAddr":53353,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"LWB010","epList":[11,242],"endpoints":{"11":{"profId":49246,"epId":11,"devId":256,"inClusterList":[0,3,4,5,6,8,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LWB010","manufacturerName":"Philips","powerSource":1,"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20170908","swBuildId":"1.29.0_r21169"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":2,"stackVersion":1,"zclVersion":1,"interviewCompleted":false,"meta":{}}
{"id":10,"type":"Router","ieeeAddr":"0x0017880104cac47f","nwkAddr":30800,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"LWB010","epList":[11,242],"endpoints":{"11":{"profId":49246,"epId":11,"devId":256,"inClusterList":[0,3,4,5,6,8,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LWB010","manufacturerName":"Philips","powerSource":1,"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20170908","swBuildId":"1.29.0_r21169"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20170908","swBuildId":"1.29.0_r21169","zclVersion":1,"interviewCompleted":true,"meta":{}}
```

Issue may be closed.

Thanks a lot !

closed 2019-12-18 by Koenkk