Koenkk/zigbee2mqtt #2540: Gledopto GL-C-008 white endpoint not working anymore since re-sync with master

dasoho, 2019-12-16:
# Bug Report
## What happened
I was on an earlier version of z2m (probably around Sep-2019) and got my GL-C-008 working properly. There are two different sub-versions of this RGB-CCT device available, one which just has one endpoint and one which has two endpoint to control RGB and CCT independently. I have the latter one and I could issue the following MQTT message
``
zigbee2mqtt/LEDController2/set {"brightness":100, "color_temp":200}  
``
to set brightness and color temp of the RGB endpiont. Of course, setting the rgb value using the appropriate keys worked as well on the RGB endpoint. These commands were affecting the RGB capable LED's on the strip. Independently of the RGB LED's I was able to set the CCT LED's with the following MQTT message:
``
zigbee2mqtt/LEDController2/white/set {"brightness":150, "color_temp":250}  
`` 
So, I was able to set both LED types independently and thus, getting the max. possible brightness out of the strip.
Yesterday, I synced on the latest version on master and addressing of the /white endpoint does not work anymore. Instead, I get the following assertion in the log:
``
error 2019-12-15T19:32:00: Failed to call 'EntityPublish' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Postfix 'white' is given but device defines no endpoints
    at Zigbee.resolveEntity (/opt/zigbee2mqtt/lib/zigbee.js:173:25)
    at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/entityPublish.js:55:36)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:320:44)
    at Controller.onMQTTMessage (/opt/zigbee2mqtt/lib/controller.js:239:14)
    at MQTT.emit (events.js:198:13)
    at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.js:86:14)
    at MqttClient.emit (events.js:198:13)
    at MqttClient._handlePublish (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:1162:12)
    at MqttClient._handlePacket (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:351:12)
    at work (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:283:12))
``
Can anyone help pointing me into the right direction where to start looking at to get the desired behavior back?

Many thanks in advance.

## What did you expect to happen
I would like to control RGB and CCT channels of the LED strip independently.

## How to reproduce it (minimal and precise)
see above

## Debug Info
zigbee2mqtt version: master branch as of Dec-15th
CC253X firmware version: 1.6.0

Koenkk, 2019-12-16:
It can now be controlled via e.g. `zigbee2mqtt/LEDController2/set` with payload  `{"white_value":100}`

dasoho, 2019-12-16:
Yes, I already saw that but I have still two issues with that compared to the previous solution with an earlier commit:

1. I have a GL-C-008 which actually has two white channels (warm&cold (CCT)) besides rgb. With `{"white_value":100}` I just can control the cold white LEDs, the warm white ones stay off. Is there a possibility to control warm white as well?

1.  If I use `{"white_value":100}` the rgb LED's turn implicitely off, I'd like to use them simultanously.

As said before, both mentioned issues were not present in an earlier version when I was able to control the white channel via the seperate path like:
`` 
zigbee2mqtt/LEDController2/white/set {"brightness":150, "color_temp":250}
``
What was the reason for changing the previous behavior? For me, the previous solution was perfectly fitting because it simply supported all features the device provides.

Koenkk, 2019-12-16:
@dasoho previously the problem was exactly the opposite (white leds stayed on), I've made the behaviour the same as when the controller is connected to a Philips Hue bulb (as this is the reference for Gledopto). 

If there is any interest in also allowing to control these separately, please leave a thumbs up on the first post (as this is quite some work).

dasoho, 2019-12-16:
Actually the Gledopto GL-C-008 comes in two sub-types, although they have the same model name. (confusing!?)
One, they call "1ID" controller and this one doesn't allow to control rgb and CCT independently and the other they call "2ID" controller and this one allows it explicitely. I do not have any of these specific bridges but from what I can read is that the latter one appears as two different devices in these kind of manufacturer specific control apps and the first one behaves more as a Hue bulb.
I do not exactly know how we can differenciate these devices but I guess it is in the number of endpoints they support.
Not sure if it is allowed to mention here but if you search for "GL-C-008" on A..express you can see that they have these 1ID and 2ID controller and it would be nice if the behavior in z2m simply depends on the device.

Koenkk, 2019-12-16:
I think all controller should be able to do this as they are separate channels. For future reference, issues linked to this change: #1731 #1711

dasoho, 2019-12-16:
But when buying a GL-C-008 you have to decide between 1id and 2id version, so I am pretty sure the behavior is somehow determined by the firmware. In the product description it is differentiated as follows:

**1ID Controller vs. 2ID Controller**

The **1ID** controller shows up as one light in the Philips Hue app. This means that it can be controlled as being one light just in the same way as a regular Philips Hue light strip. This is convenient when using it with voice control (Alexa/Google Home). The disadvantage is that you cannot turn on both the colored LEDs and the White LEDs at the same time.

The **2ID** controller shows up as **TWO** lights in the Philips Hue app. This is different from a regular Philips Hue light strip. The advantage of this is that you can turn on both the colored AND the white LEDs at the same time, allowing for more light output and control of saturation. The disadvantage is that it can be inconvenient when using voice control (Alexa/Google Home) as you need to control two sepparate lights.

nixent, 2019-12-17:
Issues reported for GL-C-008 are somewhat inconsistent because some people describe behavior for  1ID version while others test same scenarios with 2ID version and vice versa. Gledopto could simply choose to call their devices GL-C-008A and GL-C-008B to avoid all this confusion. 
Gledopto representative wrote on [Reddit](https://www.reddit.com/r/Gledopto/comments/crp1bs/glc008_vs_glc007/):
> 2ID version, consists of RGB ID(GL-C-007) and dual white ID(GL-C-006)

I have 2ID version and it is indeed recognized as GL-C-007

dasoho, 2019-12-17:
@nixent : As you said you are owning a 2ID version, may I ask you if you managed to control the rgb and dual white endpoints independently with the current z2m master?

nixent, 2019-12-17:
@dasoho No, no luck so far.
My findings so far is that before [Merge remote-tracking branch 'origin/master' into zigbee-herdsman-con](https://github.com/Koenkk/zigbee-herdsman-converters/commit/0754f58e52ef03d50da44cca302bb0d2f9431072#diff-6c9a6acf22f90d1c6e524d9f3c5c1745) converters for GL-C-007 were implemented in [devices.js,](https://github.com/Koenkk/zigbee-herdsman-converters/blob/a2bf25718ddfd728754b9bc1e64ad25b6d1d5461/devices.js#L2578) after the merge, it was moved to [toZigbee.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/0754f58e52ef03d50da44cca302bb0d2f9431072/devices.js#L2643). These changes switched logic from 2ID to 1ID implementation. Rolling back to any version of converters before Oct 4th should make 2ID devices to work. Didn't try it yet, just analysed the code.

I think root cause of the issue is that 2ID GL-C-008 reports its model as GL-C-007 and then converter cannot distinguish between real GL-C-007 and 2ID GL-C-008. Because of that it just runs converters for GL-C-007 which results in behavior you have reported.

@kirovilya maybe you can provide valuable input here, since you have implemented 2ID converter

Koenkk, 2019-12-17:
I will close this issue in favour of #2335 (as it is older). See https://github.com/Koenkk/zigbee2mqtt/issues/2335#issuecomment-566701129 on how to enable separate control of the leds (experimental).

closed 2019-12-17 by Koenkk