Koenkk/zigbee2mqtt #2543: No converter available for 'ZNCZ02LM' with cluster 'genTime' and type 'read' and data '["time"]'

kimet83, 2019-12-16:
# Bug Report
## What happened
warn log reported

warn  2019-12-17 05:48:35: No converter available for 'ZNCZ02LM' with cluster 'genTime' and type 'read' and data '["time"]'

and if networkmap updating, reported error

error 2019-12-17 05:58:50: Failed to execute LQI for 'livingroom_plug'

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.8.0
CC253X firmware version: 20190608

kitopopo, 2019-12-16:
hi,

I have updated to 1.8.0 version. Everything seems to work fine for now, but I also get the following warn:

No converter available for 'ZNCZ02LM' with cluster 'genTime' and type 'read' and data '["time"]'

I have generated the network map and the map is generated correctly and without errors but in the zigbemqtt log I can see the warning commented above. It seems that ZNCZ02LM devices are my 2 xiaomi plug routers.

Thank you for your work, we await the meaning of this error.

thanks in advanced

juckee, 2019-12-17:
Me too

zigbee2mqtt:warn  2019-12-17 12:09:10: No converter available for 'ZNCZ02LM' with cluster 'genTime' and type 'read' and data '["time"]' | stdout
zigbee2mqtt:warn  2019-12-17 12:09:19: No converter available for 'LLKZMK11LM' with cluster 'genTime' and type 'read' and data '["time"]' | stdout

IronButterfly, 2019-12-17:
The same issue:

zigbee2mqtt:warn  2019-12-17 16:28:30: No converter available for 'QBKG04LM' with cluster 'genTime' and type 'read' and data '["time"]'

BLiNdGuN, 2019-12-17:
Same here... and also with Xiaomi's LLKZMK11LM and ZNCZ02LM

zigbee2mqtt:warn  2019-12-16 21:49:13: No converter available for 'ZNCZ02LM' with cluster 'genTime' and type 'read' and data '["time"]'
zigbee2mqtt:warn  2019-12-16 21:49:26: No converter available for 'LLKZMK11LM' with cluster 'genTime' and type 'read' and data '["time"]'

1igeza, 2019-12-17:
The same issue:
`Dec 17 15:23:54 zigbee2mqtt npm[16180]: zigbee2mqtt:debug 2019-12-17 15:23:54: Received Zigbee message from 'Wlacznik dol', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
Dec 17 15:23:54 zigbee2mqtt npm[16180]: zigbee2mqtt:warn  2019-12-17 15:23:54: No converter available for 'QBKG03LM' with cluster 'genTime' and type 'read' and data '["time"]'
`

Koenkk, 2019-12-17:
Thanks for reporting all.

A `No converter available` message indicates that a certain Zigbee message from a device will not be converted to MQTT. For this message it doesn't matter actually as we don't want to report it to MQTT (does not contain interesting information).

In the latest Zigbee2mqtt dev it will be logged as debug instead of warn.

closed 2019-12-17 by Koenkk