Koenkk/zigbee2mqtt #2544: Support for EcoSmart/LDS Remode ZBT-CCTSwitch-D0001

verjus, 2019-12-17:
## EcoSmart/LDS ZBT-CCTSwitch-D0001

### Device Information:

EcoSmart Switch is a  4 button remote from an EcoSmart Kit sold by Home Depot in the U.S. See [FCC ID](https://fccid.io/2AB2Q6ARCZABZH)

### Important Note/Disclaimer:
I am generating this information from an older version of Z2M (with Shepherd). I am aware that this is not ideal and will update to the latest version (Herdsman) ASAP.

### Current Status

Main Issue is that button 2 (Brightness) does not generate a message within zigbee2mqtt. It doesn't seem to pick up CID 8 and groupId 16387. Can someone help me with this? I am stuck at this point.

Pairing OK.

Button 1: (On/Off) works.
Button 2: (Brightness) does not generate zigbee2mqtt command.
Button 3: (ColorTemp) works
Button 4: (Programmable) appears to work but not fully understood.

### Current Device.js

    {
        zigbeeModel: ['ZBT-CCTSwitch-D0001'],
        model: '6ARCZABZH',
        vendor: 'LDS/EcoSmart',
        description: 'Remote Control',
        fromZigbee: [
            fz.genOnOff_cmdOn, fz.genOnOff_cmdOff, fz.generic_battery, fz.tint404011_move_to_color_temp, fz.cmd_move_to_level_with_onoff   
        ], // We will add this later
        toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches). 
    },

### DB Information
{
  "id": 15,
  "type": "EndDevice",
  "ieeeAddr": "0xccccccfffec15cba",
  "nwkAddr": 14437,
  "manufId": 4456,
  "manufName": "LDS",
  "powerSource": "Battery",
  "modelId": "ZBT-CCTSwitch-D0001",
  "epList": [
    1
  ],
  "zclVersion": 3,
  "appVersion": 2,
  "stackVersion": 6,
  "hwVersion": 1,
  "swBuildId": "2.00",
  "status": "offline",
  "joinTime": null,
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 2048,
      "inClusterList": [
        0,
        1,
        3,
        4096,
        64769
      ],
      "outClusterList": [
        3,
        4,
        6,
        8,
[... 292 lines cut ...]
+1s

  zigbee-herdsman:unpi:parser --- parseNext [] +9ms

  zigbee-herdsman:znp:AREQ <-- AF - incomingMsg -
{"groupid":16387,"clusterid":768,"srcaddr":14437,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":65,"securityuse":0,"timestamp":1656585,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[1,57,10,29,1,3,0,0,0]}}
+12s
  zigbee-herdsman:unpi:parser --- parseNext [] +10ms

### BUTTON 4.

man:znp:AREQ <-- AF - incomingMsg -
{"groupid":16387,"clusterid":8,"srcaddr":14437,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":70,"securityuse":0,"timestamp":1742485,"transseqnumber":0,"len":6,"data":{"type":"Buffer","data":[1,60,4,254,0,0]}}
+2s
  zigbee-herdsman:unpi:parser --- parseNext [] +8ms

  zigbee-herdsman:znp:AREQ <-- AF - incomingMsg -
{"groupid":16387,"clusterid":768,"srcaddr":14437,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":70,"securityuse":0,"timestamp":1742500,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[1,61,10,222,0,3,0,0,0]}}
+18ms
  zigbee-herdsman:unpi:parser --- parseNext [] +9ms

### Additional Information:
[Here](https://github.com/dmulcahey/zha-device-handlers/issues/209)
and 
[Here](https://community.hubitat.com/t/zigbee-3-0-button-controller-8-88-usd-at-home-depot-includes-cct-bulb/29016/36)

qosmio, 2019-12-17:
I also bought these from the Home Depot sale this weekend. I'm currently working on getting this converter finalized. Note that currently button #2 needs to be held down for brightness up/down. Your fz command needs to be **fz.CTR_U_brightness_updown_hold** and **fz.CTR_U_brightness_updown_release**. Single press only seems to work when paired directly with the bulb. There is a manufacturer specific code in the incluster that I'm assuming has something to do with it.

verjus, 2019-12-17:
Looks like rapid progress is being made [here](https://github.com/mbafford/zigbee-herdsman-converters/commit/a5d2b92a2d59607aad8b3466e918b99312156522) by @mbafford

mbafford, 2019-12-17:
Yeah, technically, it works perfectly for my use with these two pull requests:

https://github.com/Koenkk/zigbee-herdsman-converters/compare/master...mbafford:ZBT-CCTSwitch-D0001?expand=1

https://github.com/mbafford/zigbee-herdsman/commit/91d7e811b953432022571d159581c0c528ac1047

I just don't know zigbee2mqtt well enough (just started yesterday) to know if the code meets the standards and requirements of the project. If someone who knows better can give a code review and let me know if anything needs changing, I'll happily submit a PR.

Thanks @verjus for the mention.

Koenkk, 2019-12-17:
Will be supported in the dev branch in a few hours, thanks!

closed 2019-12-17 by Koenkk

mbafford, 2019-12-17:
Thanks, @Koenkk!

BTW, for anyone using this remote, @brg468 brought up a good point. If you long press the button4 (scene) the remote changes the behavior of button4 to try to match the remote's "current scene", and this driver won't work properly for button4. 

This may be fixable, but the short term fix is to not long push the 4th button - and to reset the remote if that has happened and the remote is no longer working correctly.

https://github.com/dmulcahey/zha-device-handlers/pull/211#issuecomment-566729130

Neo201069, 2020-02-19:
These are showing 182% battery life what am I doing wrong?

qosmio, 2020-02-19:
Could you provide a log dump?

Neo201069, 2020-02-19:
./run.sh: line 17: [Info] Configuration backup found in
/share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No
such file or directory
[Info] Socat is DISABLED and not started
2020-02-19T01:27:23: PM2 log: Launching in no daemon mode
2020-02-19T01:27:23: PM2 log: App [npm:0] starting in -fork mode-
2020-02-19T01:27:23: PM2 log: App [npm:0] online
> zigbee2mqtt@1.10.0 start /zigbee2mqtt-1.10.0
> node index.jszigbee2mqtt:info  2020-02-19 01:27:26: Logging to console and directory: '/share/zigbee2mqtt/log/2020-02-19.01-27-26' filename: log.txtzigbee2mqtt:info  2020-02-19 01:27:27: Starting zigbee2mqtt version 1.10.0 (commit #unknown)zigbee2mqtt:info  2020-02-19 01:27:27: Starting zigbee-herdsman...zigbee2mqtt:info  2020-02-19 01:27:28: zigbee-herdsman startedzigbee2mqtt:info  2020-02-19 01:27:28: Coordinator firmware version: '{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}}'zigbee2mqtt:info  2020-02-19 01:27:28: Currently 1 devices are joined:zigbee2mqtt:info  2020-02-19 01:27:28: Livingroom Remote2 (0xccccccfffebee7d5): 6ARCZABZH - Leedarson 4-Key Remote Controller (EndDevice)zigbee2mqtt:info  2020-02-19 01:27:28: Zigbee: disabling joining new devices.zigbee2mqtt:info  2020-02-19 01:27:28: Connecting to MQTT server at mqtt://core-mosquittozigbee2mqtt:info  2020-02-19 01:27:28: Connected to MQTT serverzigbee2mqtt:info  2020-02-19 01:27:28: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'zigbee2mqtt:info  2020-02-19 01:27:28: MQTT publish: topic 'zigbee2mqtt/Livingroom Remote2', payload '{"linkquality":126,"brightness":64,"transition":0.3,"color_temp":181,"battery":182,"rate":127}'zigbee2mqtt:info  2020-02-19 01:27:28: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.10.0","commit":"unknown","coordinator":{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}},"log_level":"info","permit_join":false}'zigbee2mqtt:info  2020-02-19 01:27:28: MQTT publish: topic 'homeassistant/sensor/0xccccccfffebee7d5/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/Livingroom Remote2","json_attributes_topic":"zigbee2mqtt/Livingroom Remote2","name":"Livingroom Remote2_battery","unique_id":"0xccccccfffebee7d5_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0xccccccfffebee7d5"],"name":"Livingroom Remote2","sw_version":"Zigbee2mqtt 1.10.0","model":"4-Key Remote Controller (6ARCZABZH)","manufacturer":"Leedarson"},"availability_topic":"zigbee2mqtt/bridge/state"}'zigbee2mqtt:info  2020-02-19 01:27:28: MQTT publish: topic 'homeassistant/sensor/0xccccccfffebee7d5/action/config', payload '{"icon":"mdi:gesture-double-tap","value_template":"{{ value_json.action }}","state_topic":"zigbee2mqtt/Livingroom Remote2","json_attributes_topic":"zigbee2mqtt/Livingroom Remote2","name":"Livingroom Remote2_action","unique_id":"0xccccccfffebee7d5_action_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0xccccccfffebee7d5"],"name":"Livingroom Remote2","sw_version":"Zigbee2mqtt 1.10.0","model":"4-Key Remote Controller (6ARCZABZH)","manufacturer":"Leedarson"},"availability_topic":"zigbee2mqtt/bridge/state"}'zigbee2mqtt:info  2020-02-19 01:27:28: MQTT publish: topic 'homeassistant/sensor/0xccccccfffebee7d5/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/Livingroom Remote2","json_attributes_topic":"zigbee2mqtt/Livingroom Remote2","name":"Livingroom Remote2_linkquality","unique_id":"0xccccccfffebee7d5_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0xccccccfffebee7d5"],"name":"Livingroom Remote2","sw_version":"Zigbee2mqtt 1.10.0","model":"4-Key Remote Controller (6ARCZABZH)","manufacturer":"Leedarson"},"availability_topic":"zigbee2mqtt/bridge/state"}'zigbee2mqtt:info  2020-02-19 01:28:13: MQTT publish: topic 'zigbee2mqtt/Livingroom Remote2', payload '{"linkquality":54,"brightness":64,"transition":0.3,"color_temp":181,"battery":182,"rate":127,"click":"power","action":"on"}'zigbee2mqtt:info  2020-02-19 01:28:13: MQTT publish: topic 'zigbee2mqtt/Livingroom Remote2', payload '{"linkquality":54,"brightness":64,"transition":0.3,"color_temp":181,"battery":182,"rate":127,"click":""}'zigbee2mqtt:info  2020-02-19 01:28:13: MQTT publish: topic 'zigbee2mqtt/Livingroom Remote2', payload '{"linkquality":54,"brightness":64,"transition":0.3,"color_temp":181,"battery":182,"rate":127,"action":""}'zigbee2mqtt:info  2020-02-19 01:28:15: MQTT publish: topic 'zigbee2mqtt/Livingroom Remote2', payload '{"linkquality":108,"brightness":13,"transition":0.3,"color_temp":181,"battery":182,"rate":127,"click":"brightness","action":"brightness_up"}'zigbee2mqtt:info  2020-02-19 01:28:15: MQTT publish: topic 'zigbee2mqtt/Livingroom Remote2', payload '{"linkquality":108,"brightness":13,"transition":0.3,"color_temp":181,"battery":182,"rate":127,"click":""}'zigbee2mqtt:info  2020-02-19 01:28:15: MQTT publish: topic 'zigbee2mqtt/Livingroom Remote2', payload '{"linkquality":108,"brightness":13,"transition":0.3,"color_temp":181,"battery":182,"rate":127,"action":""}'zigbee2mqtt:info  2020-02-19 01:28:17: MQTT publish: topic 'zigbee2mqtt/Livingroom Remote2', payload '{"linkquality":93,"brightness":13,"transition":0.3,"color_temp":222,"battery":182,"rate":127,"click":"colortemp","act [... cut ...] :0.3,"color_temp":153,"battery":182,"rate":127,"click":"memory","action":"recall"}'zigbee2mqtt:info  2020-02-19 01:28:18: MQTT publish: topic 'zigbee2mqtt/Livingroom Remote2', payload '{"linkquality":87,"brightness":254,"transition":0.3,"color_temp":153,"battery":182,"rate":127,"click":""}'zigbee2mqtt:info  2020-02-19 01:28:18: MQTT publish: topic 'zigbee2mqtt/Livingroom Remote2', payload '{"linkquality":87,"brightness":254,"transition":0.3,"color_temp":153,"battery":182,"rate":127,"action":""}'