Koenkk/zigbee2mqtt #2545: Can't pair OSRAM Classic E27 Multicolor (ACO3647)

sjorge, 2019-12-17:
# Bug Report
## What happened
I'm trying to pair an OSRAM ACO3647 with little success, I know the device is not on the supported list but it should not be to hard to add support. However I can't seem to successfully start the pairing.

When turning on the bulb it cycles colors to settle on white... but zigbee2mqtt doesn't see a new device. I also did multiple off/on 5 sec cycles but it never gets to the 3x blink to indicate it is reset (I assume because it is already unpaired?)

## What did you expect to happen
Be able to pair the bulb.

## How to reproduce it (minimal and precise)
n/a

## Debug Info
zigbee2mqtt version: 1.8.0+dev
CC253X firmware version: `{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}`

sjorge, 2019-12-17:
Using touchlink reset also has no effect

wixoff, 2019-12-17:
I'm familiar with the Osram bulbs in general but not this specific model.  If this bulb is similar to the others, then the R-G-B cycling at power-on indicates it is reset and unpaired; you won't get the three blinks until it does pair.

So I think your bulb is in the expected new-from-box state, but I don't know why it won't pair.

(For the record, I have had ongoing issues with Osram/Lightify bulbs, North America versions.  They are hard to pair successfully -- though my problems are different -- and after they pair they frequently lose connectivity after a while and have lengthy periods of unavailability.  I hope you have better luck!)

Koenkk, 2019-12-17:
Could it be that it only pairs on a certain channel? (channel 15 is for example quite common).

sjorge, 2019-12-17:
Changing the channel will require repairing all devices right? Is there a way to make sure that the newer bulbs need to be on a specific channel?

Also could it be that it requires zigbee 3.0 ? The package mentions only zigbee 3.0, but I thought most devices should be backwards compatible?

Koenkk, 2019-12-18:
All devices should be backwards compatible, perhaps you can try in a separate test setup?

sjorge, 2019-12-18:
Only have one ti stick and a nRF one I use for sniffing, found some reddit posts that the newer bulbs marked zigbee 3.0 don't work on Hue bridges either, so it may be something to do with the bulb :(

sjorge, 2019-12-18:
I made some minor progress using touchlink! Looks like the bulb listens to ZLL touchlink commands but it has to be SUPER close to the coordinator, anything further than 0.5cm from the base of the bulb doesn't seem to work... it does throw timeouts.

```
zigbee2mqtt:info  2019-12-18 11:34:36: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2019-12-18 11:34:36: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.8.0-dev","commit":"960f9e4","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}'
zigbee2mqtt:info  2019-12-18 11:34:45: MQTT publish: topic 'zigbee2mqtt/bedroom/motion', payload '{"illuminance":0,"linkquality":68,"occupancy":true,"battery":100,"voltage":3015}'
zigbee2mqtt:info  2019-12-18 11:35:49: MQTT publish: topic 'zigbee2mqtt/masterbedroom/repeater', payload '{"linkquality":55}'
zigbee2mqtt:info  2019-12-18 11:35:51: Starting touchlink factory reset...
zigbee2mqtt:info  2019-12-18 11:35:56: MQTT publish: topic 'zigbee2mqtt/bedroom/radiator', payload '{"local_temperature_calibration":0,"linkquality":36,"occupied_heating_setpoint":20,"local_temperature":19.5,"unoccupied_heating_setpoint":16,"pi_heating_demand":12,"eurotronic_error_status":0,"current_heating_setpoint":20,"eurotronic_host_flags":{"mirror_display":false,"boost":false,"window_open":false,"child_protection":true},"system_mode":"auto","eurotronic_system_mode":129,"battery":100}'
zigbee2mqtt:info  2019-12-18 11:36:04: Device '0xf0d1b8000010b893' joined
zigbee2mqtt:info  2019-12-18 11:36:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0xf0d1b8000010b893"}}'
zigbee2mqtt:info  2019-12-18 11:36:04: Starting interview of '0xf0d1b8000010b893'
zigbee2mqtt:info  2019-12-18 11:36:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xf0d1b8000010b893"}}'
zigbee2mqtt:info  2019-12-18 11:36:04: MQTT publish: topic 'zigbee2mqtt/0xf0d1b8000010b893/availability', payload 'online'
zigbee2mqtt:info  2019-12-18 11:36:15: MQTT publish: topic 'zigbee2mqtt/bedroom/motion', payload '{"illuminance":0,"linkquality":68,"occupancy":false,"battery":100,"voltage":3015}'
zigbee2mqtt:error 2019-12-18 11:36:15: Failed to call 'BridgeConfig' 'onMQTTMessage' (Error: SREQ '--> AF - interPanCtl - {"cmd":1,"data":[24]}' failed with status '186' (expected '0')
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:245:27)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58))
zigbee2mqtt:error 2019-12-18 11:36:25: Failed to setup reporting for '0xf0d1b8000010b893' - Error: AREQ - ZDO - bindRsp after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:425:11)
    at tryOnTimeout (timers.js:289:5)
    at listOnTimeout (timers.js:252:5)
    at Timer.processTimers (timers.js:212:10)
zigbee2mqtt:info  2019-12-18 11:36:31: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp/bulb/availability', payload 'offline'
zigbee2mqtt:error 2019-12-18 11:36:31: Failed to ping 'bedroom/desk_lamp/bulb'
(node:61872) UnhandledPromiseRejectionWarning: Error: Timeout - 27321 - 11 - 72 - 0 - 1 after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:425:11)
    at tryOnTimeout (timers.js:289:5)
    at listOnTimeout (timers.js:252:5)
    at Timer.processTimers (timers.js:212:10)
(node:61872) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:61872) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
zigbee2mqtt:error 2019-12-18 11:36:35: Failed to interview '0xf0d1b8000010b893', device has not successfully been paired
zigbee2mqtt:info  2019-12-18 11:36:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0xf0d1b8000010b893"}}'
(node:61872) UnhandledPromiseRejectionWarning: Error: Timeout - 47486 - 1 - 71 - 0 - 1 after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:425:11)
    at tryOnTimeout (timers.js:289:5)
    at listOnTimeout (timers.js:252:5)
    at Timer.processTimers (timers.js:212:10)
(node:61872) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 2)
```

It never completes the interview :(

```
{"id":14,"type":"Router","ieeeAddr":"0xf0d1b8000010b893","nwkAddr":47486,"manufId":4489,"manufName":"LEDVANCE","powerSource":"Mains (single phase)","modelId":"CLA60 RGBW Z3","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,2821,4096,64527],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"CLA60 RGBW Z3","manufacturerName":"LEDVANCE","powerSource":1}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{},"lastSeen":1576668975289}
```

But I ran out of time for now, will poke it some more later.

sjorge, 2019-12-18:
Looks like the actually coordinator stick crashed, after a stop, unplug, start it seems to be back... ofcourse not fully interviewed

sjorge, 2019-12-18:
Another go at touchlink reset with the coordinator super close worked fine. I'll close this for now.

closed 2019-12-18 by sjorge