Koenkk/zigbee2mqtt #2547: Can't add eleventh device (Ikea Tradfri 806lm bulb) via configuration.yaml

FabienTregan, 2019-12-17:
# Bug Report

I can't add my ikea tradfri bulb to a group, the outlet are working.

## What works

I have a configuration with 5 ikea tradfri remote, cc2531 on a raspberry, and 5 ikea tradfri outlets. Things worked fine. The five outlets are in group 1 and answer correctly.

I then added an Ikea tradfri 806lm bulb, and it paired well :
```
mosquitto_pub -t "zigbee2mqtt/bridge/config/devices/get" -m "x"
```
```
{"ieeeAddr":"0xccccccfffe3c7387","type":"Router","networkAddress":53073,"model":"LED1836G9","friendly_name":"lum/salon/n6","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI bulb E27 WW 806lm","hardwareVersion":2,"softwareBuildID":"2.1.022","dateCode":"20181203","lastSeen":1576581916586}
```

I tryed to unplug four of the outlets just to make sure I was not over the device limit for the cc2531, in the [full report](https://pastebin.com/pcMV7ZJu) we can see that the cc2531 sees the 5 remote, the remaining plug, and the bulb.

The bulb switches on/off fine when I post to `zigbee2mqtt/lum/salon/n6`.

## What do not work

I can't add this bulb to the group.

In my [`configuration.yaml`](https://pastebin.com/gXNKge4w) file I have it added to the group like other devices but it does not answer to messages on `zigbee2mqtt/grp/salon` .

If I try to list the group memberships fo the outlet and the bulb:
```bash
$ mosquitto_pub -t "zigbee2mqtt/bridge/device/0x086bd7fffe218aca/get_group_membership" -m "x"
$ mosquitto_pub -t "zigbee2mqtt/bridge/device/0xccccccfffe3c7387/get_group_membership" -m "x"```
 I get the expected answer from the outlet, but no group is found for the bulb:
```
zigbee2mqtt/bridge/device/0x086bd7fffe218aca/get_group_membership x
zigbee2mqtt/lum/salon/n1 {"state":"ON","linkquality":57,"brightness":0,"group_list":[1],"group_capacity":255}
zigbee2mqtt/bridge/device/0xccccccfffe3c7387/get_group_membership x
zigbee2mqtt/lum/salon/n6 {"state":"OFF","brightness":0,"linkquality":107,"group_list":[],"group_capacity":255}
```
(note: I get no answer if trying to use frendly_name, see #2545 

if I try to ad them both a another group:

```
zigbee2mqtt/bridge/config/add_group {"friendly_name": "test_group", "id": 2}
zigbee2mqtt/bridge/group/test_group/add /lum/salon/n1
zigbee2mqtt/bridge/group/test_group/add /lum/salon/n6
zigbee2mqtt/bridge/device/0x086bd7fffe218aca/get_group_membership x
zigbee2mqtt/lum/salon/n1 {"state":"ON","linkquality":57,"brightness":0,"group_list":[1],"group_capacity":255}
zigbee2mqtt/bridge/group/test_group/add 0x086bd7fffe218aca
zigbee2mqtt/bridge/log {"type":"device_group_add","message":{"friendly_name":"lum/salon/n1","group":"test_group"}}
```

On the othe end,if I try to add to the group using command line and ieeeAddr instead of device fiendly name, it works:

```
zigbee2mqtt/bridge/device/0x086bd7fffe218aca/get_group_membership x
zigbee2mqtt/lum/salon/n1 {"state":"ON","linkquality":57,"brightness":0,"group_list":[1,2],"group_capacity":255}
``zigbee2mqtt/bridge/group/test_group/add 0xccccccfffe3c7387
zigbee2mqtt/bridge/log {"type":"device_group_add","message":{"friendly_name":"lum/salon/n6","group":"test_group"}}
zigbee2mqtt/bridge/device/0xccccccfffe3c7387/get_group_membership x
zigbee2mqtt/lum/salon/n6 {"state":"OFF","brightness":0,"linkquality":107,"group_list":[2],"group_capacity":255}
```

So the problem seems to be related to #2546 but there are still two strange things:
 - it also applies to configuration.yaml where I use ieeeAddr
 - with confguration.yaml, it applies to my bulb but not to the plugs.

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

FabienTregan, 2019-12-17:
Further investigation : I can't add the bulb to group 1 with command line either:
```
zigbee2mqtt/bridge/config/add_group {"friendly_name": "test_group2", "id": 3}

zigbee2mqtt/bridge/group/salon/add 0xccccccfffe3c7387
zigbee2mqtt/bridge/device/0xccccccfffe3c7387/get_group_membership x
zigbee2mqtt/lum/salon/n6 {"state":"OFF","brightness":0,"linkquality":107,"group_list":[2],"group_capacity":255}

zigbee2mqtt/bridge/device/0xccccccfffe3c7387/get_group_membership x
zigbee2mqtt/lum/salon/n6 {"state":"OFF","brightness":0,"linkquality":107,"group_list":[2],"group_capacity":255}

zigbee2mqtt/bridge/group/test_group2/add 0xccccccfffe3c7387
zigbee2mqtt/bridge/log {"type":"device_group_add","message":{"friendly_name":"lum/salon/n6","group":"test_group2"}}

zigbee2mqtt/bridge/device/0xccccccfffe3c7387/get_group_membership x
zigbee2mqtt/lum/salon/n6 {"state":"OFF","brightness":0,"linkquality":107,"group_list":[2,3],"group_capacity":255}

zigbee2mqtt/bridge/group/salon/add 0xccccccfffe3c7387
zigbee2mqtt/bridge/device/0xccccccfffe3c7387/get_group_membership x
zigbee2mqtt/lum/salon/n6 {"state":"OFF","brightness":0,"linkquality":107,"group_list":[2,3],"group_capacity":255}
```
Apparently I can add to group 2 and 3 but not to 1. Is it "full" ? Is the a problem with the group friendly name ? It the problem with the bulb or with Z2M ?

FabienTregan, 2019-12-17:
Apparently, adding / removing to/from group do not work using mqtt is M2Z thinks the ieeeAddr is not / is already in the group.

By manually removing the device from the `configuration.yaml` file and restarting Z2M, I could then join the group with:
```
zigbee2mqtt/bridge/group/salon/add 0xccccccfffe3c7387
```

Koenkk, 2019-12-17:
Can you provide the log of https://github.com/Koenkk/zigbee2mqtt/issues/2547#issuecomment-566524561

closed 2020-02-22 by stale[bot]