Koenkk/zigbee2mqtt #2554: Can Xiaomi/Mijia/Aquara connect to generic/ikea router?

mr2c12, 2019-12-18:
Mijia cube and Mijia door sensors, in my case, can connect only to coordinator (cc2530).
Even if I repair them really close to:
- ikea bulb (w/router)
- ikea rage extender

They always go to the coordinator.
This is a big problem, since I need them far from the coordinator, and the link quality is just "0". They works but..

Should those xiaomi devices be able to connect to an IKEA router?
What if I use another cc2530 as a router close to them? Is it possible to connect them to that router?

ginkage, 2019-12-20:
If you use a cc2530 router, they will connect to it (tested with Aqara and Mijia temperature sensor).

mr2c12, 2019-12-20:
Great. Unfortunately I can confirm that Mijia devices are unable to connect to Ikea router.

ginkage, 2019-12-20:
On the upside, the cc2530 router can connect to IKEA ones, and use those to route the signal from your sensors to the rest of the network, so it's probably what you should do anyway.

zhapwey, 2019-12-20:
I have been struggling with this today. Although I am sure that I have connected an aquara motion sensor to an ikea bulb at some point and it worked, I don't remember how.

mr2c12, 2019-12-21:
@ginkage do you think that
coordinator -> ikea router -> cc2530/cc2531 router -> mijia device
will work?

ginkage, 2019-12-21:
@mr2c12, yes, I think it will work.

zhapwey, 2019-12-21:
And do you know if a temperature sensor would be connected to a Xiaomi plug or to one of its rooter?

ginkage, 2019-12-22:
@zhapwey No idea: I don't have any of those. I should probably get one, just for the sake of testing...

zhapwey, 2019-12-23:
I have been testing today, the xiaomi devices connect to the cc2530 rooter without problems, in case there is a coordinator, ikea rooter, cc2530 rooter it works without problem too.
That is if you know if there is any way to prevent cc2530 repeaters from sending as many messages because as you have a lot of equipment it is a lot of traffic that generates.

mr2c12, 2020-01-07:
To answer my own initial question: yes, xiaomi devi es can connect to Cc2530 router.

closed 2020-01-07 by mr2c12

mr2c12, 2020-01-29:
Useful info: xiaomi devices can connect to ikea bulb too.