Koenkk/zigbee2mqtt #2555: Unknown Device

Haui74, 2019-12-18:
Can you please add support for this 4 channel RGBW controller.

Thanks a lot!!

[image]
[image]
[image]

Haui74, 2020-01-25:
Can you ?help me please

Koenkk, 2020-01-26:
See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Haui74, 2020-02-11:
Can you please add the device, its to complicated to me. Im not a programmer ;) Thank you very much!! :)

closed 2020-04-18 by stale[bot]