Koenkk/zigbee2mqtt #2556: Move zigbee2mqtt to another pc and keep all the paired devices

mr2c12, 2019-12-18:
Is it possible to move a fully configured zigbee2mqtt installation to another host in the same network, using the same sniffer, without reconnecting every device?
Should I just make a fresh installation on the destination, than copy&past the whole content of the zigbee2mqtt directory? Than grant the necessary permissions, stop the "old" zigbee2mqtt and start the "new" one?
Should it be enough?

My goal is to:

1. keep the device paired
2. keep any zigbee2mqtt configuration

pepp86, 2019-12-18:
Yes. Just copy you data directory to the new machine.

Koenkk, 2019-12-19:
@pepp86 is right, just copy your `data` folder and start :smile:

closed 2019-12-19 by Koenkk

jeppper, 2020-08-25:
I have just configured a new ssd drive for zigbee2mqtt and copied my existing data folder from my old drive.
All but one light works now. It is a Gledopto RGBW controller. I have tried to reset it and pair it again but no luck.
I can fine pair it to my Hue Bridge and I am also able to pair other products to zigbee2mqtt.
I have also noticed that my state.json file looks strange now. Normally it is a fine list but now it only consists of one line of text.
Anyone has an idea what to do?

Kimovitzh, 2021-05-07:
I tried this between two Raspberry's, only Philips motionsensors migrated. IKEA Trådfri component didn't.
Is that normal?