Koenkk/zigbee2mqtt #2557: Can´t pair Ledvance Smart+ Plug AC10691

ststrell12345, 2019-12-18:
# Bug Report

## What happened
Hello,
I try to get my Ledvance(former OSRAM) Smart+ Plugs connected to Zigbee2MQTT running on HassIO with CC2531. No matter what I do, they just don´t show up in Z2M.

What I tried so far:
- Add them to Z2M 1.8 with Zigbee Version 1.2 and channel 11
- Flashing Zigbee 3.0 to CC2531 because Ledvance states, that they use Zigbee 3.0 (As I read later, the devices should be compatible with old versions of Zigbee, so that was probably not necessary?!)
- Add them to Z2M 1.8 with Zigbee Version 3 and channel 11
- Add them to Z2M 1.8 edge with Zigbee Version 3 and tried every channel from 11 to 25

I also tried another plug in case one is broken, same result...
Pairing of other devices like Tradfri or Aqara is possible without any effort.

## What did you expect to happen
Pairing the devices

## How to reproduce it (minimal and precise)
Try to pair, nothing happens
## Debug Info
zigbee2mqtt version: 1.8 dev
CC2531 firmware version: :{"type":"zStack30x","meta":{"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425}

sjorge, 2019-12-18:
Might be the same issue I am having here: https://github.com/Koenkk/zigbee2mqtt/issues/2545

Koenkk, 2019-12-19:
Can you try resetting the device through https://www.zigbee2mqtt.io/information/touchlink

ststrell12345, 2019-12-19:
Hi, thanks for your reply!
Is there a difference between the reset via Touchlink or the normal one, when pressing on/off 10s?
I already tried resetting it with the on/off button multiple times.

coxtor, 2019-12-28:
Hi similar to the issue #2545 I got the issue fixed by @sjorge "Looks like the actually coordinator stick crashed, after a stop, unplug, start it seems to be back... ofcourse not fully interviewed"

ststrell12345, 2020-01-02:
Hi, sorry for the late reply, I just got home from Christmas holidays.
I tried resetting it via Touchlink, but no matter what I do, it just does not work.
I tried it with the coordinator stick directly touching the plug at different places, but it only states "Failed to factory reset device through Touchlink" in the logs.
Anything else I can do? Otherwise I will return them....
Btw: Happy new year! :)

ststrell12345, 2020-01-12:
Tried it again a couple of times with a 'out of the box' plug and the dev version. Did not work either. Sent them back today. Anyway thanks for our support!

closed 2020-01-12 by ststrell12345