Koenkk/zigbee2mqtt #2558: YRD226HA2619 State Reporting

arduino43, 2019-12-18:
Hello,

I have a YRD226HA2619 that is not reporting any states, but lock/unlock is working.

* No battery state reported
* Lock/Unlock state not reported
* No signal state is reported

CC2531
zigbee2mqtt v 1.8 (issue in v 1.7 as well)

I noticed there was a PR in April for this lock, but since moving to herdsmen it appears to no longer work.

Koenkk, 2019-12-19:
This was a bug in 1.7 setting it up incorrectly, when repairing the device with 1.8 it should work. so:

- Remove the device https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove
- Pair it.

arduino43, 2019-12-27:
This seems to still be an issue after update.

moistgun, 2020-01-03:
I have this lock working with 1.8 in the same manner as @arduino43
I am able to publish lock and unlock commands, but the state is not updating, and for me, is timing out.

This is after removing and repairing as mentioned above.

```
zigbee2mqtt:error 2020-01-02 21:00:29: Publish 'set' 'state' to 'mydevice' failed: 'Error: Timeout - 34930 - 1 - 28 - 257 - 11 after 15000ms'
zigbee2mqtt:info  2020-01-02 21:00:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'mydevice' failed: 'Error: Timeout - 34930 - 1 - 28 - 257 - 11 after 15000ms'","meta":{"friendly_name":"mydevice"}}'
```

arduino43, 2020-01-03:
I was able to get this working, and the state is reporting correctly, after upgrading to the C2650. I think it may have been a signal issue for me.

Koenkk, 2020-01-07:
I found the root cause of this issue, when updating to the latest dev you should see the zigbee2mqtt configures the device (`Succesfully configured [FRIENDLY_NAME]`), after this it should work.

moistgun, 2020-01-07:
I'm still getting the same errors using the zigbee2mqtt-edge hassio addon, but I can test whenever that gets updated.

I don't recall ever getting an option to upgrade that addon though so I was under the impression it was always up to date with the dev branch of the zb2mqtt project.

moistgun, 2020-01-08:
Sorry about hopping all over these issues. 

I figured out how to add custom code to the addon, and did what you said here >
https://github.com/Koenkk/zigbee2mqtt/issues/2635#issuecomment-571545670 and am getting states back from my lock, and no timeouts so far.

Koenkk, 2020-01-08:
Great, I will close this issue then because that config is already present in the latest edge addon and will be in the next zigbee2mqtt release. Feel free to re-open if the problem reappears.

closed 2020-01-08 by Koenkk