Koenkk/zigbee2mqtt #2559: "interview_failed" when pairing Xiaomi MFKZQ01LM cube

beodegit, 2019-12-18:
# Bug Report
## What happened

Pairing of a Xiaomi Cube MFKZQ01LM fails with "interview_failed". Pairing of another new device is succesful.

## What I tried
-clear entries of new device in devices.yaml & database.db
-upgrade 1.7.1 to 1.8.0-dev

## What did you expect to happen

to pair with "interview succesful", as the device is on the supported devices list ([MFKZQ01LM]( https://www.zigbee2mqtt.io/devices/MFKZQ01LM.html))

## Debug Info

zigbee2mqtt version: 1.7.1  / 1.8.0-dev
CC2531 firmware version: 20190608

part of regular logfile below, Complete  DEBUG=* npm start log on [pastebin](https://pastebin.com/QbHKN9Xi) (Look for device 0x00158d0002a6f82e)
.

info  2019-12-17 18:11:51: Device '0x00158d0002a6f82e' joined
info  2019-12-17 18:11:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00158d0002a6f82e"}}'
info  2019-12-17 18:11:51: Starting interview of '0x00158d0002a6f82e'
info  2019-12-17 18:11:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d0002a6f82e"}}'
error 2019-12-17 18:13:11: Failed to interview '0x00158d0002a6f82e', device has not successfully been paired
info  2019-12-17 18:13:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00158d0002a6f82e"}}'
info  2019-12-17 18:51:06: MQTT publish: topic 'zigbee2mqtt/ikea-wall-zolder', payload '{"state":"OFF","linkquality":26}'
info  2019-12-17 18:58:03: MQTT publish: topic 'zigbee2mqtt/ikea-wall-zolder2', payload '{"state":"OFF","linkquality":23}'
info  2019-12-17 19:51:06: MQTT publish: topic 'zigbee2mqtt/ikea-wall-zolder', payload '{"state":"OFF","linkquality":23}'
info  2019-12-17 19:58:03: MQTT publish: topic 'zigbee2mqtt/ikea-wall-zolder2', payload '{"state":"OFF","linkquality":23}'
info  2019-12-17 20:29:38: Device '0x000b3cfffefcca55' joined
info  2019-12-17 20:29:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000b3cfffefcca55"}}'
info  2019-12-17 20:29:38: Starting interview of '0x000b3cfffefcca55'
info  2019-12-17 20:29:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000b3cfffefcca55"}}'
info  2019-12-17 20:30:29: Successfully interviewed '0x000b3cfffefcca55', device has successfully been paired
info  2019-12-17 20:30:29: Device '0x000b3cfffefcca55' is supported, identified as: IKEA TRADFRI control outlet (E1603/E1702)
info  2019-12-17 20:30:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x000b3cfffefcca55","model":"E1603/E1702","vendor":"IKEA","description":"TRADFRI control outlet","supported":true}}'
info  2019-12-17 20:30:29: Configuring '0x000b3cfffefcca55'
info  2019-12-17 20:30:29: Succesfully configured '0x000b3cfffefcca55'
info  2019-12-17 20:31:49: Starting interview of '0x00158d0002a6f82e'
info  2019-12-17 20:31:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d0002a6f82e"}}'
error 2019-12-17 20:33:09: Failed to interview '0x00158d0002a6f82e', device has not successfully been paired
info  2019-12-17 20:33:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00158d0002a6f82e"}}'

Koenkk, 2019-12-19:
Did you try repairing it once more? (by resetting it)

beodegit, 2019-12-19:
I tried repairing it with short press and long (>5s) press and both seem to trigger repairing the same way, but after 6 times 'node descriptor request failed' and 2 times 'active endpoints request failed' pairing stops.

mr2c12, 2019-12-20:
I confirm that pairing the cube is a bit tricky.
What is working for me is pressing the reset Button every 2-3 sec AFTER you start the repair process.
You should:
- hold reset for 5 sec
- the pairing should start
- now quick press the reset every 2-3

This should get it done. Or at least, it works in my case.

beodegit, 2019-12-20:
Amazing, that worked in my case too:

info  2019-12-20 07:59:24: Successfully interviewed '0x00158d0002a6f82e', device has successfully been paired

@Koenkk can you maybe add this info to the [Cube's](https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/MFKZQ01LM.md) device page?

Thanks!

jiri-jirus, 2020-01-25:
I have to confirm, was not able to pair the cube in 1.9.0 the quick press trick after reset worked for me.

Koenkk, 2020-01-25:
I think you will need to hold the reset button a little bit longer, until the blinking goes off, on zigbee2mqtt 1.9.0 dev I've managed to successfully pair it with this. Quick press trick shouldn't be needed.

jiri-jirus, 2020-01-26:
good to know!

kristianrl, 2020-02-05:
I am encountering a similar issue, @Koenkk :

It seemed like I was close to connecting correctly, but after a few seconds, the Aquara Cube disconnected:

```
Device '0x00EXAMPLE' joined
MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00EXAMPLE"}}'
Starting interview of '0x00EXAMPLE'
MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00EXAMPLE"}}'
MQTT publish: topic 'homeassistant/sensor/0x00EXAMPLE/action/config', payload '{"icon":"mdi:gesture-double-tap","value_template":"{{ value_json.action }}","state_topic":"zigbee2mqtt/0x00EXAMPLE","json_attributes_topic":"zigbee2mqtt/0x00EXAMPLE","name":"0x00EXAMPLE_action","unique_id":"0x00EXAMPLE_action_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00EXAMPLE"],"name":"0x00EXAMPLE","sw_version":"Zigbee2mqtt 1.8.0","model":"Mi/Aqara smart home cube (MFKZQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
MQTT publish: topic 'homeassistant/sensor/0x00EXAMPLE/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x00EXAMPLE","json_attributes_topic":"zigbee2mqtt/0x00EXAMPLE","name":"0x00EXAMPLE_battery","unique_id":"0x00EXAMPLE_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00EXAMPLE"],"name":"0x00EXAMPLE","sw_version":"Zigbee2mqtt 1.8.0","model":"Mi/Aqara smart home cube (MFKZQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
MQTT publish: topic 'homeassistant/sensor/0x00EXAMPLE/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00EXAMPLE","json_attributes_topic":"zigbee2mqtt/0x00EXAMPLE","name":"0x00EXAMPLE_linkquality","unique_id":"0x00EXAMPLE_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00EXAMPLE"],"name":"0x00EXAMPLE","sw_version":"Zigbee2mqtt 1.8.0","model":"Mi/Aqara smart home cube (MFKZQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
MQTT publish: topic 'zigbee2mqtt/0x00EXAMPLE', payload '{"battery":68,"voltage":2945,"linkquality":57}'
Device '0x00EXAMPLE' left the network
MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x00EXAMPLE"}}'
```

Later, it did not pair succesfully:

```
Starting interview of '0x00EXAMPLE'
MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00EXAMPLE"}}'
Failed to interview '0x00EXAMPLE', device has not successfully been paired
MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00EXAMPLE"}}'
```

I can no longer get it to produce anything from the Cube in the log (I have tried to long-press as written in the documentation, as well as @mr2c12's suggestion https://github.com/Koenkk/zigbee2mqtt/issues/2559#issuecomment-567811985 ). Other zigbee devices work.

Koenkk, 2020-02-05:
@kristianrl https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair (especially battery; try replacing or disconnecting it for some time).

kristianrl, 2020-02-11:
> @kristianrl https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair (especially battery; try replacing or disconnecting it for some time).

I had already tried those suggestions (as well as restarting the Docker instance), but it seemed like a physical restart of the server running Docker solved the issue.

zorruno, 2020-02-15:
I can add a 'metoo' to this.  Possibly because of low battery (reports 74 now) but after a rebuild of my server I could not get this device to pair (was paired fine previous to that).  I can now report pressing reset for 5 secs (the LED flashes) then waiting 2 seconds and re-pressing the button briefly allowed it to pair.  I didn't try with non dev version.

errors:
zigbee2mqtt:error 2020-02-15 13:54:06: Failed to interview
device has not successfully been paired

versions:
Version of Zigbee2Mqtt: 1.10.0-dev
Coordinator version: 20190608

closed 2020-04-23 by stale[bot]

LaborEtArs, 2021-01-08:
Another 'me too'!
zigbee2mqtt version 1.17.0 (? from mqtt zigbee2mqtt/bridge/info/version)
The cube successfully joined only after pressing long (~5sec) and than pressing short every few seconds. Joined very quickly then.

lepinkainen, 2021-01-15:
One more 'me too', except I was unable to pair with the long press until light goes out + short presses every few seconds trick.

```
zigbee2mqtt    | Zigbee2MQTT:info  2021-01-15 23:48:07: Starting interview of 'Aqara Cube'
zigbee2mqtt    | Zigbee2MQTT:info  2021-01-15 23:48:07: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Aqara Cube","ieee_address":"0x00158d0004aafc38","status":"started"},"type":"device_interview"}'
zigbee2mqtt    | Zigbee2MQTT:info  2021-01-15 23:48:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"Aqara Cube"},"type":"pairing"}'
zigbee2mqtt    | Zigbee2MQTT:error 2021-01-15 23:50:26: Failed to interview 'Aqara Cube', device has not successfully been paired
zigbee2mqtt    | Zigbee2MQTT:info  2021-01-15 23:50:26: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Aqara Cube","ieee_address":"0x00158d0004aafc38","status":"failed"},"type":"device_interview"}'

```

Using Zigbee2MQTT version 1.17.0 (commit #07cdc9d) via Docker