Koenkk/zigbee2mqtt #2561: OSRAM AA70155 - 'MAC no ack' (233)'

vencakratky, 2019-12-18:
# Bug Report
## What happened
Pairing device in the Room, where is Coordinator CC2531, all was functionall. After moving to next room, where is router Innr SP120 (linkquality 90) and sending "turn_on" command i become  'Error: Data request failed with error: 'MAC no ack' (233)' 
## What did you expect to happen
Leave Bulb in next Room - with router Innr SP120. This router via Hassio working well and linkquality with Bulb is 90
## What i tried
Remove device with MQTT explorer and  topic: "zigbee2mqtt/bridge/config/force_remove"
payload: "0x841826000000e61c", new pairing in Room with router Innr SP120, but problem the same: Error: Data request failed with error: 'MAC no ack' (233)' . If i put Bulb back to the Room with coordinator, working good and without error.
## Debug Info
There is config:
https://hastebin.com/vofenazuje.json 

Debug Log:
https://hastebin.com/ahitebeqox.sql
[image]

zigbee2mqtt version: 1.8.0.
CC2531 firmware version: 20190608

lolorc, 2019-12-23:
also having this issue with a Osram plug:
```
Dec 23 16:38:13 zigbee2mqtt:error 2019-12-23 16:38:13: Publish 'set' 'state' to 'prise_02' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'
Dec 23 16:38:13 zigbee2mqtt:debug 2019-12-23 16:38:13: Error: Data request failed with error: 'MAC no ack' (233)
Dec 23 16:38:13     at ZStackAdapter.<anonymous> (/home/pi/zigbee2mqtt.dev/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:578:27)
Dec 23 16:38:13     at Generator.next (<anonymous>)
Dec 23 16:38:13     at fulfilled (/home/pi/zigbee2mqtt.dev/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:5:58)
```

the device was working ok yesterday, afair I haven't made any change to my network. (running latest dev)
re-pairing the device doesn't change anything. My other Osram plugs still work ok.

vencakratky, 2019-12-23:
I am waiting for CC-Debugger Programmer (Order on AliExpress) , after i want try the second firmware Z-Stack_Home1.2 - source routing for CC2531, maybe it help, but I'm not sure

Pockrandt, 2019-12-27:
Source routing definitely helps. 
Also check the revision of your stick.
My 1.0 "CC25XX" became wonky with 6 nearby routers.
Comparison:
https://photos.app.goo.gl/ThpVK1SkuDT3Ti3T8
(Revision ist at the top of the USB port)

vencakratky, 2019-12-27:
> 
> 
> Source routing definitely helps.
> Also check the revision of your stick.
> My 1.0 "CC25XX" became wonky with 6 nearby routers.
> Comparison:
> https://photos.app.goo.gl/ThpVK1SkuDT3Ti3T8
> (Revision ist at the top of the USB port)

And if I have Revision 1.0, will "source routing" not cause problems? Unfortunately, I have no experience yet.  At worst, I buy a new CC2531 with revision 2.0.

Pockrandt, 2019-12-27:
> And if I have Revision 1.0, will "source routing" not cause problems? Unfortunately, I have no experience yet. At worst, I buy a new CC2531 with revision 2.0.

source routing helps the more routers you have.
My 1.0 with 6 routers (ikea/osram bulbs & osram plugs) in short distance (like 5m² room) some device wouldn't react every other day for the first command. 

btw this is my personal experience, dunno if others expect the same behaviour.

mabi, 2020-01-21:
> Source routing definitely helps.
> Also check the revision of your stick.
> My 1.0 "CC25XX" became wonky with 6 nearby routers.
> Comparison:
> https://photos.app.goo.gl/ThpVK1SkuDT3Ti3T8
> (Revision ist at the top of the USB port)

where did you find the 2.4 revision? Thanks

vencakratky, 2020-01-21:
I close this Issue. After source routing really it´s working.

closed 2020-01-21 by vencakratky

Pockrandt, 2020-01-21:
> > Source routing definitely helps.
> > Also check the revision of your stick.
> > My 1.0 "CC25XX" became wonky with 6 nearby routers.
> > Comparison:
> > https://photos.app.goo.gl/ThpVK1SkuDT3Ti3T8
> > (Revision ist at the top of the USB port)
> 
> where did you find the 2.4 revision? Thanks

ali express  from "diymore Official Store" 2.2.19 which was the cheapest option with reviews at the time
https://de.aliexpress.com/item/32852226435.html?spm=a2g0s.9042311.0.0.52874c4dkgqr9k

btw looks like they now also sell them with antenna, might be a try worth