Koenkk/zigbee2mqtt #2563: Feature request 'genTime' and type 'read' and data '["time"]'

edillmann, 2019-12-19:
# Bug Report

This is not a bug but a feature request 

## What happened

Xiaomi devices are asking to get current time, should'nt it be wise to respond to this inquiries ?

No converter available for 'QBKG03LM' with cluster 'genTime' and type 'read' and data '["time"]'

## What did you expect to happen

should'nt it be wise to respond to these requests ?

## How to reproduce it (minimal and precise)

Add a Xiaomi switch QBKG03LM

## Debug Info
zigbee2mqtt version: 1.8.0
CC253X firmware version: zStacck30x :20190425

Koenkk, 2019-12-19:
This is already done, see https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/controller.ts#L509

closed 2019-12-19 by Koenkk

NewFolk, 2019-12-21:
Just update to 1.8.0 And got a lot of message on log:
> No converter available for 'ZNCZ02LM' with cluster 'genTime' and type 'read' and data '["time"]'
> No converter available for 'ZNCZ02LM' with cluster 'genTime' and type 'read' and data '["time"]'
> No converter available for 'QBKG04LM' with cluster 'genTime' and type 'read' and data '["time"]'
> No converter available for 'QBKG04LM' with cluster 'genTime' and type 'read' and data '["time"]'
> No converter available for 'QBKG04LM' with cluster 'genTime' and type 'read' and data '["time"]'
> 

Is it the same?
It is very noisy. will you release bug fix version?

qm3ster, 2020-02-16:
The lines have moved, it's this: https://github.com/Koenkk/zigbee-herdsman/blob/7650ad1bff19ca738fc7a02d5ad9153916935dc9/src/controller/controller.ts#L525-L550

treemonkey83, 2023-12-23:
i have noticed the same thing happening for Woolley plugs https://www.zigbee2mqtt.io/devices/BSD29.html#woolley-bsd29
the "type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0" messages are relentless and spamming the coordinator.