Koenkk/zigbee2mqtt #2564: Xiaomi QBKG04LM Wall Switch Won't Decouple

tcherokee, 2019-12-19:
# Bug Report
I am trying to get my Xiaomi QBKG04LM Wall Switch (No Neutral) to go into decoupled mode. If I try to get or set the operation mode it doesn't work. It worked perfectly for my Xiaomi QBKG03LM Double button wall switch though.

## What happened

Here is what happens when I try and set or get the operation mode

### Getting
I send the following payload
```
{
    "operation_mode": {
        "button": "single"
    }
}
```
to `zigbee2mqtt/Master Bedroom Light Switch/get`

This is the error I got in my logs

```
[34mzigbee2mqtt:debug[39m 2019-12-19 05:06:07: Received MQTT message on 'zigbee2mqtt/Master Bedroom Light Switch/get' with data '{"operation_mode":{"button":"single"}}'
[34mzigbee2mqtt:debug[39m 2019-12-19 05:06:07: Publishing get 'get' 'operation_mode' to 'Master Bedroom Light Switch'
[31mzigbee2mqtt:error[39m 2019-12-19 05:06:17: Publish 'get' 'operation_mode' to 'Master Bedroom Light Switch' failed: 'Error: Timeout - 64089 - 2 - 21 - 0 - 1 after 10000ms'
[34mzigbee2mqtt:debug[39m 2019-12-19 05:06:17: Error: Timeout - 64089 - 2 - 21 - 0 - 1 after 10000ms
at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
at ontimeout (timers.js:436:11)
at tryOnTimeout (timers.js:300:5)
at listOnTimeout (timers.js:263:5)
at Timer.processTimers (timers.js:223:10)
[32mzigbee2mqtt:info [39m 2019-12-19 05:06:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'get' 'operation_mode' to 'Master Bedroom Light Switch' failed: 'Error: Timeout - 64089 - 2 - 21 - 0 - 1 after 10000ms'","meta":{"friendly_name":"Master Bedroom Light Switch"}}'
```

### Setting
Trying to set the switch button to decoupled mode. I send the following payload
```
{
    "operation_mode": {
        "button": "single",
        "state": "decoupled"
    }
}
```
to 
to `zigbee2mqtt/Master Bedroom Light Switch/set`

This is the error I get in my logs
```
[34mzigbee2mqtt:debug[39m 2019-12-19 05:15:05: Received Zigbee message from 'Master Bedroom Light Switch', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3300,"3":26,"5":22,"8":9750,"10":0,"100":0,"101":0,"110":0,"111":255,"153":0,"155":0}}' from endpoint 1 with groupID 0
[34mzigbee2mqtt:debug[39m 2019-12-19 05:15:07: Received MQTT message on 'zigbee2mqtt/Master Bedroom Light Switch/set' with data '{"operation_mode":{"button":"single","state":"decoupled"}}'
[34mzigbee2mqtt:debug[39m 2019-12-19 05:15:07: Publishing 'set' 'operation_mode' to 'Master Bedroom Light Switch'
[34mzigbee2mqtt:debug[39m 2019-12-19 05:15:15: Received Zigbee message from 'Master Bedroom Light Switch', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
[34mzigbee2mqtt:debug[39m 2019-12-19 05:15:15: No converter available for 'QBKG04LM' with cluster 'genTime' and type 'read' and data '["time"]'
[31mzigbee2mqtt:error[39m 2019-12-19 05:15:17: Publish 'set' 'operation_mode' to 'Master Bedroom Light Switch' failed: 'Error: Timeout - 64089 - 2 - 22 - 0 - 4 after 10000ms'

[34mzigbee2mqtt:debug[39m 2019-12-19 05:15:17: Error: Timeout - 64089 - 2 - 22 - 0 - 4 after 10000ms

at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
at ontimeout (timers.js:436:11)
at tryOnTimeout (timers.js:300:5)
at listOnTimeout (timers.js:263:5)
at Timer.processTimers (timers.js:223:10)
[32mzigbee2mqtt:info [39m 2019-12-19 05:15:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'operation_mode' to 'Master Bedroom Light Switch' failed: 'Error: Timeout - 64089 - 2 - 22 - 0 - 4 after 10000ms'","meta":{"friendly_name":"Master Bedroom Light Switch"}}'
```

## What did you expect to happen

I expected the switch to go into decoupled mode.
I think it could be related to this [bug](https://github.com/Koenkk/zigbee2mqtt/issues/2130)

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.8.0-dev 
CC253X firmware version: using the CC1352P_2 Co-ordinator with the 20191106 firmware

Koenkk, 2019-12-23:
Can you try using `zigbee2mqtt/Master Bedroom Light Switch/set`?

tcherokee, 2019-12-24:
> Can you try using `zigbee2mqtt/Master Bedroom Light Switch/set`?

I did try using this. I commented on my results in the setting section in my comment above. This was the error I got

`[32mzigbee2mqtt:info [39m 2019-12-19 05:15:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'operation_mode' to 'Master Bedroom Light Switch' failed: 'Error: Timeout - 64089 - 2 - 22 - 0 - 4 after 10000ms'","meta":{"friendly_name":"Master Bedroom Light Switch"}}'`

Did I perhaps get the payload wrong?

Koenkk, 2019-12-26:
No, it means that somehow the device cannot be reached.

tcherokee, 2020-01-01:
Hmm that is a bit weird as the Switch in the bedroom (which doesn't work) is closer to the co-ordinator than the double switch in the living room (which does work). Plus other commands seem to work , light switching on and off the lights. I will try jiggling things around to see if it makes a difference.

CODeRUS, 2020-01-14:
same for me, timeout error. ON/OFF works just fine

CODeRUS, 2020-01-15:
this works:
topic: "zigbee2mqtt/0x00158d0001333b64/system/set"
payload:
```
{
  "operation_mode": {
    "button": "single",
    "state": "decoupled"
  }
}
```

tcherokee, 2020-01-19:
This works perfectly, thanks.

How did you figure out what the right topic was?

CODeRUS, 2020-01-19:
from here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L531

CODeRUS, 2020-02-19:
@Koenkk ^ please update docs :)

Koenkk, 2020-02-20:
@CODeRUS thanks for pinging me again, updated the docs, thanks!

Assuming this can be closed now.

closed 2020-02-20 by Koenkk