Koenkk/zigbee2mqtt #2567: Need help to add a new Thermostat from Lupus Electronics

rsys-dev, 2019-12-19:
Hi,

I tried to add a new Thermostat from Lupus Electronics. I paired it with my zigbee network and added this to devices.js
`
> {
>         zigbeeModel: ['TRV_00.00.03.09TC'],
>         model: '12053',
>         vendor: 'Lupus',
>         description: 'LUPUSEC Radiator Valve Thermostat V.2018',
>         supports: 'get/set valve',
>         fromZigbee: [],
>         toZigbee: [],
>     },
`

I also got this from "zigbee2mqtt/bridge/log"

`
> {
> "ieeeAddr":"0x1234567890abcdef1",
> "type":"EndDevice",
> "networkAddress":xxxxx,
> "model":"12053",
> "friendly_name":"Thermostat",
> "manufacturerID":0,
> "manufacturerName":"ClimaxTechnology",
> "powerSource":"Battery",
> "modelID":"TRV_00.00.03.09TC",
> "hardwareVersion":0,
> "dateCode":"20180308",
> "lastSeen":1576784211757
> }
`

But i have no idea how to find valid commands and write propper converters.

Has anybody an idea how to proceed

Kind Regards
Manuel

sjorge, 2019-12-19:
You can try by adding the following to fromZigbee:
- fz.thermostat_att_report
- fz.battery_percentage_remaining (might need a different one depending on the battery type)

toZigbee is probably more guess work but you could try some basics
- tz.thermostat_occupied_heating_setpoint
- tz.thermostat_unoccupied_heating_setpoint

rsys-dev, 2019-12-20:
Hi sjorge,

thanks for your help. Now I get some response from the Thermostat in the log file (temperature, heating_setpoint etc).
I will play around with these informations the next days.

Kind Regards
Manuel

rsys-dev, 2020-01-02:
The thermostat seems to work now (Set Temperature). I am trying to find out if there are other settings and send a pull request.

closed 2020-03-10 by stale[bot]

coxtor, 2023-10-06:
has this been integrated yet ?