Koenkk/zigbee2mqtt #2569: Need help to add Feller Smart Light Control EdizioDue

kalsan, 2019-12-20:
I'm attempting to support the new device Feller EdizioDue 1903fh, a product of the "Friends of Hue" program by Swiss manifacturer Feller. The cool thing about it is that it's piezo-electric, i.e. no batteries needed. According to the product page, a Hue v2 bridge is required. I fail to connect the device to zigbee2mqtt, the pairing does not seem to take place, the logs show nothing.

Here's a link to the product page: https://www.feller.ch/de/Produktangebot/Funktaster/Zusatzinformationen

Here's a video of how they pair it to an actual Hue bridge: https://www.youtube.com/watch?v=ox1YBWgCxoo&t=385s

When I set permit_join to true and attempt to connect as in the video, nothing happens. I'm new to zigbee, but I read that any zigbee/lightlink device should work with zigbee2mqtt.

Which further steps can I take to analyze and ideally solve the problem?

kalsan, 2019-12-20:
After investigating a little more, I found out that the Feller switch uses Zigbee Green Power. According to #211, Z-Stack 3.0 is required for Green Power, which is still experimental according to #1445. I couldn't quite figure out from these issues whether Green Power already works with Z-Stack 3.0 or whether it's just possible to potentially make it work. I think I will give it a try, though.

closed 2019-12-20 by kalsan

yvansandoz, 2022-02-08:
Did you find a workaround to connect your Feller EdizioDue with Zigbee2MQTT?

UnconventionalMindset, 2022-12-06:
If anyone needs to know how, I managed with this guide: https://www.zigbee2mqtt.io/devices/PTM_215Z.html

kalsan, 2024-01-04:
Indeed! Bought a Feller and it worked immediately after resetting using the procedure described in the PTM 215Z guide :+1: