Koenkk/zigbee2mqtt #2571: Switch problem with 1.8.0 only

dzungpv, 2019-12-20:
I use my custom switch, Xiaomi switch with Z2M without any issue until version 1.8.0.
The problem:
- I have use Home Assistant automation to sync state of two switches.
- I tested with 2 switches with my custom hardware DNCKATSW001 and  Xiaomi switch: QBKG03LM
- They work normal like this: when **switch.id2_switch** ON then **switch.id1_switch** ON, when **switch.id2_switch** OFF then **switch.id1_switch** OFF and the same with  **switch.id1_switch** states with the automation.
But version 1.8.0 make 2 switches change state endlessly 
- The problem never occur until version 1.8.0, i restore version 1.7.1 and problem solve
- Home Assistant version: 0.102.x, 0.103.x, I don't know if it occur with older version

So it may relate to delay of the on/off command or multi process in version 1.8.0.
Automation here so anyone can produce the bug with any switch in HA (I am guest):

```
  #automation for BR3 light button 1 sync
  - id: control_light_bedroom3
    alias: Control light bedroom 3.1
    trigger:
      platform: state
      entity_id:
        - switch.id1_switch
    condition:
      - condition: template
        value_template: '{{ states.switch["id2_switch"].state != trigger.to_state.state }}'
    action:
      - service_template:
          switch.turn_{{trigger.to_state.state}}
        data:
          entity_id:
            - switch.id2_switch
   #automation for BR3 light button 2 sync
  - id: control_light_bedroom3_2
    alias: Control light bedroom 3.2
    trigger:
      platform: state
      entity_id:
        - switch.id2_switch
    condition:
      - condition: template
        value_template: '{{ states.switch["id1_switch"].state != trigger.to_state.state }}'
    action:
      - service_template:
          switch.turn_{{trigger.to_state.state}}
        data:
          entity_id:
            - switch.id1_switch
```
Queue setting:

```
  "queue": {
    "delay": 50,
    "simultaneously": 5
  }
```

nldroid, 2019-12-20:
Not to discard the bug but what's the use case?

Does the state of the switch matter? Shouldn't you check the state of the item that is switched?

I can imagine that a state change triggers the other switch state change and that will trigger again and again and again..

dzungpv, 2019-12-20:
@nldroid It use for stair switch like this: https://www.electricaltechnology.org/2012/11/stair-case-wiring-circuit-diagram-or.html , but with single switch, I can turn on/off 1 switch from many position.

dzungpv, 2019-12-22:
@Koenkk after debug version 1.7.1 and version 1.8.0 i have found the problem:
On version 1.7.1 event processing faster than in 1.8.0, the delay in 1.8.0 cause my problem.
Two test with same devices, same setting running on CC3530.

On 1.7.1:

```
zigbee2mqtt:debug 2019-12-22T05:25:13: Received MQTT message on 'zigbee2mqtt/0x00124b000d7ff551/left/set' with data 'ON'
zigbee2mqtt:debug 2019-12-22T05:25:13: Publishing 'set' 'state' to '0x00124b000d7ff551'
zigbee2mqtt:debug 2019-12-22T05:25:13: Received Zigbee message from '0x00124b000d7ff551', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2019-12-22T05:25:13: MQTT publish: topic 'zigbee2mqtt/0x00124b000d7ff551', payload '{"state_right":"OFF","linkquality":106,"state_left":"OFF","button_left":"hold"}'
zigbee2mqtt:debug 2019-12-22T05:25:13: Received Zigbee message from '0x00124b000d7ff551', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2019-12-22T05:25:13: MQTT publish: topic 'zigbee2mqtt/0x00124b000d7ff551', payload '{"state_right":"OFF","linkquality":106,"state_left":"ON","button_left":"release"}'
zigbee2mqtt:info  2019-12-22T05:25:13: MQTT publish: topic 'zigbee2mqtt/0x00124b000d7ff551', payload '{"state_right":"OFF","linkquality":106,"state_left":"ON"}'
```

Version 1.8.0
```
zigbee2mqtt:debug 2019-12-22 12:02:04: Received MQTT message on 'zigbee2mqtt/0x00xxxb000d7ff551/left/set' with data 'ON'
zigbee2mqtt:debug 2019-12-22 12:02:04: Publishing 'set' 'state' to '0x00xxxb000d7ff551'
zigbee2mqtt:debug 2019-12-22 12:02:05: Received Zigbee message from '0x00xxxb000d7ff551', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2019-12-22 12:02:05: MQTT publish: topic 'zigbee2mqtt/0x00xxxb000d7ff551', payload '{"state_right":"OFF","linkquality":106,"state_left":"ON","button_left":"release"}'
zigbee2mqtt:info  2019-12-22 12:02:05: MQTT publish: topic 'zigbee2mqtt/0x00xxxb000d7ff551', payload '{"state_right":"OFF","linkquality":106,"state_left":"ON"}'
zigbee2mqtt:debug 2019-12-22 12:02:13: Received Zigbee message from '0x00xxxb000d7ff551', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 2 with groupID 0
zigbee2mqtt:info  2019-12-22 12:02:13: MQTT publish: topic 'zigbee2mqtt/0x00xxxb000d7ff551', payload '{"state_right":"OFF","linkquality":106,"state_left":"ON","button_right":"hold"}'
```

So like logs above version 1.8.0 take 7 seconds to finish ON command, but version 1.7.1 take 1 seconds, so it cause the problems.

Koenkk, 2019-12-23:
I'm not sure where this delay comes from, can you verify using sniffing or debug some more to see where it comes from?

dzungpv, 2020-01-10:
@Koenkk I write the firmware for those device, it is may cause by duplicate ZCL report update on/off state of the switch, send by zcl_SendReportCmd after each time device change state.

I send the capture file, from packet 596, i send Off command from HA  after that HA auto send Off command to mirror switch. At 834 i send On command.
[switch.zip](https://github.com/Koenkk/zigbee2mqtt/files/4046492/switch.zip)

Koenkk, 2020-01-12:
@dzungpv do you also have this zigbee2mqtt debug log of that? I don't see anything strange in the sniff logs, could you also try explaining the problem a bit more, I still have problems understanding what the exact problem is.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

dzungpv, 2020-01-17:
@Koenkk It is randomly, so i am still investigate it.

dzungpv, 2020-01-18:
Maybe it cause by double publish with same state, so i temporary fix the problem by delay automation in HA, so close it now.

closed 2020-01-18 by dzungpv