Koenkk/zigbee2mqtt #2573: upgrade from 1.6 not working

zp-markusp, 2019-12-20:
hi,

I have currently running on a CC2531 with latest coordinator firmware which works more or less quite stable. After upgrading to any version > 1.6 it's not starting anymore. I tried Versions 1.7, 1.7.1, 1.8.
I even don't get any logs at all.
Current environment is raspberry pi zero and I'm running it with docker:

    docker run \
        -it \
        --name=zigbee2mqtt \
        -v /opt/zigbee2mqtt/data:/app/data \
        --device=/dev/ttyACM0 \
        -e TZ=Europe/Amsterdam \
        -v /run/udev:/run/udev:ro \
        --privileged=true \
        koenkk/zigbee2mqtt:1.7.0

Log directory stays empty, even no folder is created as in V1.6. Also state and db file was not modified during start.

log output (/var/log/debug)
https://hastebin.com/uyomofivic.coffeescript

Any ideas what could be the issue?

Thanks, Markus

zp-markusp, 2019-12-20:
looks like it's related to arm architecture of the pi zero device.

for 1.6 it was working with koenkk/zigbee2mqtt:arm32v6 but the current installation is reporting the arch as 'linux/arm' for which no build is available...
Any possible workarounds?

Koenkk, 2019-12-23:
Can you add the following parameter to force the image? `--platform arm/v6`

zp-markusp, 2019-12-24:
Hi,

I tried this, but doesn't work too... same result.
My workaround is to run it locally without docker. Works but the other solution would be preferred ;-)

closed 2020-02-29 by stale[bot]