Koenkk/zigbee2mqtt #2575: New install on Raspbian (pi4) failing with "Error while starting zigbee-herdsman"

gliquorish, 2019-12-20:
# Bug Report
## What happened
Starting Zigbee2MQTT always fails at "Starting-Herdsman"

## What did you expect to happen
Expect Zigbee2MQTT to start

## How to reproduce it (minimal and precise)
This is a fresh install of Raspbian with all patches and updates applied running on a Pi4-4G.
Fresh install of Z2M following instructions on this site.
CC2531 appears in /dev as ttyAMC0.  If this is not connected, starting Z2M fails sooner, so I think this is working.  I am very new to this, so have never had an older version installed.

lrwxrwxrwx 1 root root 13 Dec 20 22:41 usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0001599AEE-if00 -> ../../ttyACM0

Output from "DEBUG=* npm start" @ https://hastebin.com/qoxufitevi.coffeescript

## Debug Info
zigbee2mqtt version: 1.8.0
CC253X firmware version: CC2531_DEFAULT_20190608

Any help would be gratefully received.
Regards
Gavin

didiht, 2019-12-21:
Could it be that there exists a Zigbee network with same channel/panId already?

You can try to change channel and/or panId (e.g channel: 20, pan_id: 0x1a63).

Koenkk, 2019-12-23:
Please check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start

gliquorish, 2019-12-23:
All of the checks pass and the device name is correct.  I have tried this now on OpenHasbian, Raspbian and Windows and I have the same error on each.
I am beginning to believe that the CC25321 is faulty.  It always shows as a TI device on Raspbian and a Serial Device on Windows but I have no other way of testing the unit.
Is there a guide to testing the device without a full suite of software, e.g. a command that will return RAW data from the device, maybe return a firmware version or serial number etc.?

Using Wireshark with the USB sniffer driver didn't appear to return anything.  I use Wireshark frequently for work, so understand the network sniffing but have never done anything with USB at this level.

The CC2531 was flashed using the Arduino method and I believe this is working, certainly, there is an error thrown if I try flashing without it connected as detailed in the guide.

I do have a new CC2531 on order, but I'm not expecting it before Christmas.

Regards
Gavin

gliquorish, 2019-12-30:
New CC2531 arrived today, flashed as before and Herdsman now starts.
On with configuring the rest of the system.

Closing ticket.

closed 2019-12-30 by gliquorish