Koenkk/zigbee2mqtt #2577: Zigbee2mqtt requires node version 10

zmbcgn, 2019-12-20:
# Bug Report
## What happened
'npm start' fails due to node version requirement

pi@zigpi:~/zigbee2mqtt/data $ npm start

> zigbee2mqtt@1.6.0 start /home/pi/zigbee2mqtt
> node index.js

                Zigbee2mqtt requires node version 10, you are running v12.14.0!

/home/pi/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/bindings/bindings.js:121
        throw e;
        ^

Error: The module '/home/pi/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 64. This version of Node.js requires
NODE_MODULE_VERSION 72. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

## What did you expect to happen
Howto on https://www.zigbee2mqtt.io/ says node version should be 12.x or 10.x after install. So I assume that should work.

## How to reproduce it (minimal and precise)
reinstall and re-run npm start

## Debug Info
zigbee2mqtt version: 1.6.0 
CC253X firmware version: unknown

airdrummingfool, 2019-12-20:
This was fixed in https://github.com/Koenkk/zigbee2mqtt/commit/c2be95777f3e91edb31bfe1d532ac39c7f935d81 and should not happen on version 1.8.0.

zmbcgn, 2019-12-21:
indeed. thanks for the heads up.

closed 2019-12-21 by zmbcgn